PX4/PX4-Autopilot #18096: Bridge chat to Matrix

vranki, 2021-08-18:
Current chat platform (Slack) is a bit outdated solution for an open source project. Matrix (https://matrix.org) offers 
the same functionality and a lot more in completely FOSS manner. Major FOSS organizations such as Mozilla, KDE and 
Gnome have migrated to Matrix and it is becoming the standard for communication replacing incompatible IRC, Gitter, Slack and
Discord for projects.

**Describe problem solved by the proposed feature**
Slack is a communication platform silo which requires using their closed source client on centralized server owned by single company. Matrix offers same (better) functionality but is decentralized and has FOSS software stack. 

**Describe your preferred solution**
Matrix supports bridging to Slack, so the current workspace can be seamlessly bridged without any action needed
from current users. Those who prefer Slack can keep using it, and those wishing to use Matrix can participate via it.

In practice - 
* Preferably someone with admin access in Slack side makes rooms on Matrix and bridges them. I can help with on Matrix side if needed.
* On Matrix side a "PX4" space is created where all related rooms are added. This is about the same as a Slack workspace and offers a single point to link to. Users can then join rooms they are interested in.
* Update documentation to point to both Matrix and Slack

**Describe possible alternatives**
There are not any realistic alternatives to Matrix. Not bridging is one, but that would force contributors to use Slack or stay out of it.

**Additional context**
Matrix supports bridging to multiple messengers, so in future it would be possible to bridge the rooms also to Discord, Telegram, IRC etc for those who prefer them.  

Aerospace space has some related rooms already: https://matrix.to/#/!czJErpTOMRvDmDStSw:hacklab.fi?via=hacklab.fi&via=matrix.org&via=kabelsalat.ch

mrpollo, 2021-08-18:
Thanks @vranki we are aware of the issue with Slack and are planning to migrate to matrix, we already have an instance running, you are welcome to join us, https://matrix.to/#/!pDUITTrJSoXkOZZjRU:chat.lfx.linuxfoundation.org?via=chat.lfx.linuxfoundation.org

vranki, 2021-09-02:
This room seems to be awfully quiet. Is there any timeline for the actual migration?