PX4/PX4-Autopilot #18118: Unable to roslaunch px4 posix_sitl.launch

HatedFate, 2021-08-20:
I'd installed PX4, Gazebo, and ROS according to the document (Not sure if I missed any steps), and I managed to get other stuff running like:
`sudo make px4_sitl gazebo`
But when I tried to run ROS, Gazebo, and PX4 together using:
`roslaunch px4 posix_sitl.launch`
it returns:
`RLException: [posix_sitl.launch] is neither a launch file in package [px4] nor is [px4] a launch file name
The traceback for the exception was written to the log file
`
I had tried using other people's solution by typing:
`source catkin_ws/devel/setup.sh`
But this result in the same error and I had also tried to use hoping it would work:
`source catkin_ws/devel_isolated/setup.sh`
But it result in the same error

Jaeyoung-Lim, 2021-08-20:
@HatedFate So, you should not put PX4 inside your workspace.

Basically, you are missing some path configurations such as
```
cd <PX4-Autopilot_clone>
DONT_RUN=1 make px4_sitl_default gazebo
source ~/catkin_ws/devel/setup.bash    # (optional)
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 posix_sitl.launch
```

You can use http://docs.px4.io/master/en/simulation/ros_interface.html for more detailed instructions

HatedFate, 2021-08-20:
I ran the commands and it work, but it is returning a new error:

`ERROR [px4] is_server_running: failed to create lock file: /tmp/px4_lock-0, reason=Permission denied
`
I tried the command in another terminal:
`sudo make clean`

I rerun the roslaunch command, but now it is returning another error:
`ERROR: cannot launch node of type [px4/px4]: Cannot locate node of type [px4] in package [px4]. Make sure file exists in package path and permission is set to executable (chmod +x)
`
`[FATAL] [1629483351.301168, 0.000000]: Error: specified file /home/kannachan/px4/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf does not exist
`
`[vehicle_spawn_kannachan_desktop_6829_3207880129139104892-5] process has died [pid 6867, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -sdf -file /home/kannachan/px4/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf -model iris -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_kannachan_desktop_6829_3207880129139104892 __log:=/home/kannachan/.ros/log/a5aeac6e-01e2-11ec-9a63-3c58c292baa8/vehicle_spawn_kannachan_desktop_6829_3207880129139104892-5.log].
log file: /home/kannachan/.ros/log/a5aeac6e-01e2-11ec-9a63-3c58c292baa8/vehicle_spawn_kannachan_desktop_6829_3207880129139104892-5*.log
`

HatedFate, 2021-08-20:
The way I installed gazebo, ros, and px4 was more messed than the way I look, so I reinstalled everything according to https://docs.px4.io/master/en/simulation/ros_interface.html. It appears that there’s a new issue with gazebo process dying for some reason. #18119

closed 2021-08-20 by HatedFate