PX4/PX4-Autopilot #18119: gzserver preventing roslaunch px4 posix_sitl.launch

HatedFate, 2021-08-20:
I downloaded PX4, ROS, and Gazebo 9 according to the documentation provided in:
https://docs.px4.io/master/en/simulation/ros_interface.html

After installing, exporting, and sourcing everything, I'd ran:

`roslaunch px4 posix_sitl.launch
`
But this returns an error:

`[gazebo-3] process has died [pid 3361, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/kannachan/Developer/PX4-Autopilot/Tools/sitl_gazebo/worlds/empty.world __name:=gazebo __log:=/home/kannachan/.ros/log/b091d7c0-01fe-11ec-8747-3c58c292baa8/gazebo-3.log].
log file: /home/kannachan/.ros/log/b091d7c0-01fe-11ec-8747-3c58c292baa8/gazebo-3*.log
[gazebo_gui-4] process has died [pid 3373, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/kannachan/.ros/log/b091d7c0-01fe-11ec-8747-3c58c292baa8/gazebo_gui-4.log].
log file: /home/kannachan/.ros/log/b091d7c0-01fe-11ec-8747-3c58c292baa8/gazebo_gui-4*.log
`
I searched for some answers online ( https://answers.gazebosim.org//question/4153/gazebo-crashes-immediately-using-roslaunch-after-installing-gazebo-ros-packages/ )and was told that I should run:
 
`killall gzserver` 

But there is no processes named gzserver.

codingCoffee, 2021-11-26:
@HatedFate were you able to resolve this issue by any chance?

HatedFate, 2021-12-13:
> @HatedFate were you able to resolve this issue by any chance? 

Sorry for the late response. Yes, the issue did resolved on its own, but I am not entirely sure what happen so it's probably not that big of a deal.

closed 2021-12-14 by HatedFate