PX4/PX4-Autopilot #18121: IridiumSBD Driver and Parameters Missing

SAR-mango, 2021-08-21:
**Description**
The Iridium SBD section simply does not appear in Parameters in QGC. Searching for any relevant parameters (ISBD_CONFIG, for example) yields no results. Attempts to start the driver through the shell lead to `nsh: iridiumsbd: command not found`.

I have this issue with master, v1.12.1, and v1.12.0. I flashed to a Pixhawk 4 (fmu-v5) board and ensured all parameters were firmware default.

**Modifying boards/px4/fmu-v5/default.cmake as described below appears to fix the problem.** I was just messing around and did not expect this to work, and now that it does work, I have no idea why.

Line 52 currently contains:
`telemetry # all available telemetry drivers`
Below line 52 I just added:
`telemetry/iridiumsbd`
After making this change, building, and flashing, the Iridium SBD section appears, the parameters are present, and the driver is functional (confirmed by trying `iridiumsbd status` in the shell).

**To Reproduce**
1. Flash master, v1.12.1, or v1.12.0 to a Pixhawk 4 board. Reset all parameters to firmware default if desired.
2. Open the list of parameters in QGC and search for the Iridium SBD section, it won't be present.
3. Search for relevant parameters such as ISBD_CONFIG, they will not be found.
4. Open the shell (MAVLink Console) in QGC and try iridiumsbd commands such as 'iridiumsbd status'. The commands will not be found.

**Expected behavior**
The Iridium SBD section and its parameters should appear in QGC with fmu-v5_default firmware. The driver should also be enabled and working, and commands in the shell should be recognized.

**Log Files and Screenshots**
No flight tests were performed as they are irrelevant to this issue, so there are no logs to provide.

**Not Working (v1.12.1)**
[image]
[image]
[image]
[image]
[image]
[image]

**Working (after firmware modification)**
[image]
[image]

**Drone**
Irrelevant, but I have successfully set up PX4 with a large fixed-wing glider. Working on an ultra long range project (solar plane) where satellite might me necessary. I currently took all the avionics out and they are just on the bench, so a photo of the setup isn't really helpful anyway.

**Additional Context**
Screenshots show when I was testing with Mac, but this problem is present on Windows QGC as well.

dagar, 2021-08-22:
The `iridiumsbd` driver was disabled for now due to a number of outstanding issues. I believe in certain situations it was crashing on startup and getting users stuck in a boot loop.

https://github.com/PX4/PX4-Autopilot/pull/17837

SAR-mango, 2021-08-22:
Right, sorry, I just realized that about an hour ago and was about to close this issue.

I have HW to test with and I have time as well, school starts in two weeks. I enabled the driver and configured it with TELEM2. It did not crash or anything, but the driver was not communicating with the modem. Switched to TELEM1 and it worked perfectly. Should I make a new issue?

ryanjAA, 2021-08-25:
@SAR-mango do you have a slack account in the px4 workspace? DM me at @ryanjohnston

SAR-mango, 2021-08-25:
Sorry... not in the PX4 workspace. The invite link seems not to be working.

julianoes, 2023-02-14:
The IridiumSBD driver can be enabled using KConfig. See https://github.com/PX4/PX4-Autopilot/pull/21105#issuecomment-1429042093 for how it's done.

For instance for Pixhawk 6X the command is: `make px4_fmu-v6x_default boardconfig`, then you select *drivers* -> *Telemetry* -> *iridiumsbd*

closed 2023-02-14 by julianoes