PX4/PX4-Autopilot #18125: Camera plugin topic not added on multiple drones

saadnec, 2021-08-23:
I am trying to add downward_depth_camera to my drones in multi_uav_mavros_sitl_sdf.launch file , to do that i copied the lines from the camera jinja file and pasted them into iris.sdf.jinja file. The camera is showing in gazebo but the topics shwoing when i run **rostopic list** are showing globally not individually of every single drone the topics are
"""
/camera/rgb/image_raw
/camera/rgb/image_raw/compressed
/camera/rgb/image_raw/compressed/parameter_descriptions
/camera/rgb/image_raw/compressed/parameter_updates
/camera/rgb/image_raw/compressedDepth
/camera/rgb/image_raw/compressedDepth/parameter_descriptions
/camera/rgb/image_raw/compressedDepth/parameter_updates
/camera/rgb/image_raw/theora
/camera/rgb/image_raw/theora/parameter_descriptions
/camera/rgb/image_raw/theora/parameter_updates
""""
Instead of
"""
uav0/camera/rgb/......

uav1/camera/rgb/......
""" 
etc
What am i doing wrong?

Jaeyoung-Lim, 2021-08-23:
@saadnec Which camera plugin are you using? It is most likely that the ros camera plugin is not configured correctly to have muliple cameras

saadnec, 2021-08-23:
> 
> 
> @saadnec Which camera plugin are you using? It is most likely that the ros camera plugin is not configured correctly to have muliple cameras

depth_camera

Jaeyoung-Lim, 2021-08-23:
@saadnec What exact plugin are you using? Is this the depth_camera model? or the depth_camera gazebo plugin? or the depth camera sensor?

saadnec, 2021-08-23:
> @saadnec What exact plugin are you using? Are you talking about the [depth_camera](https://github.com/PX4/PX4-SITL_gazebo/blob/master/models/depth_camera/depth_camera.sdf.jinja) model? or the [depth camera sensor](https://github.com/PX4/PX4-SITL_gazebo/blob/7fda4d311a9daff4bec4f2fe83e63fde0b8f04b0/models/depth_camera/depth_camera.sdf.jinja#L27)? or the [gazebo depth camera plugin](https://github.com/PX4/PX4-SITL_gazebo/blob/7fda4d311a9daff4bec4f2fe83e63fde0b8f04b0/models/depth_camera/depth_camera.sdf.jinja#L73)?
> 
> If you are including the same depth camera model in all of your vehicles(e.g. `<include>model://depth_camera</include>`, they will be all publishing to the same topic.
> 
> The camera namespace is defined in https://github.com/PX4/PX4-SITL_gazebo/blob/7fda4d311a9daff4bec4f2fe83e63fde0b8f04b0/models/depth_camera/depth_camera.sdf.jinja#L74

So you are saying that i need to make changes to multi_uav_mavros_sitl_sdf.launch file ? if yes then what changes do i have to make to it ? ,because multi_uav_mavros_sitl_sdf.launch is calling single_vehicle_spawn_sdf.launch for each vehicle.

Jaeyoung-Lim, 2021-08-23:
@saadnec No, you need to change your sdf file of your model to not include the depth camera model, and have them directly as part as your templain. 

To explain in detail, you need to have different [camera names](https://github.com/PX4/PX4-SITL_gazebo/blob/7fda4d311a9daff4bec4f2fe83e63fde0b8f04b0/models/depth_camera/depth_camera.sdf.jinja#L74) for each vehicle. If you just use `<include>model://depth_camera</include>` then it will simply use https://github.com/PX4/PX4-SITL_gazebo/blob/7fda4d311a9daff4bec4f2fe83e63fde0b8f04b0/models/depth_camera/depth_camera.sdf.jinja, which all has the same camera name

saadnec, 2021-08-23:
> @saadnec No, you need to change your sdf file of your model to not include the depth camera model, and have them directly as part as your templain.
>
> To explain in detail, you need to have different [...]

Thank you very much ,sorted this issue with your help :)

closed 2021-08-23 by saadnec