PX4/PX4-Autopilot #18131: Cannot send mavlink message to rotate gimbal using ROS message

tawfiq1200, 2021-08-23:
I can rotate the gimbal of typhoon480 using the following code that I found in pymavlink [documenation](https://www.ardusub.com/developers/pymavlink.html):

```
connection = mavutil.mavlink_connection('udpin:0.0.0.0:14540')
connection.wait_heartbeat()
connection.mav.command_long_send(connection.target_system, connection.target_component, mavutil.mavlink.MAV_CMD_DO_MOUNT_CONTROL, 1, pitch, roll, yaw, 0, 0, 0, mavutil.mavlink.MAV_MOUNT_MODE_MAVLINK_TARGETING)
connection.close()
```

However, I am working on a ROS based project and I am setting up UDP connection from the ROS launch file like this:
`<arg name="fcu_url" default="udp-b://:14540@14548"/>`

So I do need to make a UDP connection from the python code.

I am creating a heartbeat connection like this and I am trying to send the mavlink message as a ROS message like this:

```
hb_mav_msg = mavutil.mavlink.MAVLink_heartbeat_message(mavutil.mavlink.MAV_TYPE_ONBOARD_CONTROLLER, 0, 0, 0, 0, 0)

gimbal_mav_msg=mavutil.mavlink.mav.command_long_send(mavutil.mavlink.target_system, mavutil.mavlink.target_component, mavutil.mavlink.MAV_CMD_DO_MOUNT_CONTROL, 1, pitch, roll, yaw, 0, 0, 0, mavutil.mavlink.MAV_MOUNT_MODE_MAVLINK_TARGETING)
gimbal_ros_msg = mavlink.convert_to_rosmsg(gimbal_mav_msg)
pub.publish(gimbal_ros_msg)
```
It gives the following error:
`
AttributeError: module 'pymavlink.dialects.v10.ardupilotmega' has no attribute 'mav'`

dayjaby, 2021-09-01:
Sorry for the late answer. Did you consider using mavros? I have working examples for doing pretty much the same by using mavros: https://gist.github.com/dayjaby/cbe9f027a7d48638c776ca8512058b29

tawfiq1200, 2021-09-01:
@dayjaby Thanks, did you mean running this script should rotate the gimbal of typhoon480 simulated drone? I rosrun this script but the gimbal did not move.

tawfiq1200, 2021-09-01:
@dayjaby Here are the scripts that I ran, they did not rotate the gimbal:

test_gimbal.py:

```
#!/usr/bin/env python3
import rospy
from gimbal import MavlinkPublicationTest

def main():
    rospy.init_node("test")
    test=MavlinkPublicationTest()
    test.gimbal_rotate()

if __name__ == "__main__":
    main()
```

gimbal.py:

```
#!/usr/bin/env python3
import rospy
from mavros import mavlink
from mavros_msgs.msg import Mavlink
from pymavlink.dialects.v20 import common

class MavlinkPublicationTest:

    def __init__(self):

        self.mav_pub = rospy.Publisher("/mavlink/to", Mavlink, queue_size=1)
        self.mav = common.MAVLink(None)

    def gimbal_rotate(self):
        
        print("I am rotating gimbal")
        msg = common.MAVLink_command_long_message(
                target_system=0,
                target_component=0,
                command=205,
                confirmation=0,
                param1=-90, # pitch
                param2=45, # roll
                param3=0, # yaw
                param4=0,
                param5=0,
                param6=0,
                param7=2
            )
        msg.pack(self.mav) # you can add ,force_mavlink1=True here if you need mavlink version 1
        self.mav_pub.publish(mavlink.convert_to_rosmsg(msg))
```
The code executes, prints the message but the typhoon480's gimbal does not rotate.

tawfiq1200, 2021-09-02:
Does it need a heartbeat message to be sent? If so, then what type, for connection, I am using this from roslaunch file:
`<arg name="fcu_url" default="udp-b://:14540@14548"/>`

tawfiq1200, 2021-09-03:
@dayjaby  Since you said it is a working code, I think you were to move the simulated gimbal with it? However, I need to know what communication protocol you were using, could you please provide me your roslaunch file?