PX4/PX4-Autopilot #18142: make px4_sitl jmavsim/gazebo doesn't work

NicolassFelipe, 2021-08-24:
Hello everyone,
I'm starting with the simulation of Px4, I have a virtual machine with ubuntu 20.04 LTS (build-essentials and updates installed). I have followed the steps of the px4 installation guide (https://www.youtube.com/watch?v=OtValQdAdrU - https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html#gazebo- jmavsim-and-nuttx-pixhawk-targets)
running the commands in the terminal ( i know ther is an existing issue about this but its almost empty so i post this again)

git clone https://github.com/PX4/Firmware.git --recursive
cd Firmware
bash ./Tools/setup/ubuntu.sh
sudo reboot now
cd Firmware
make px4_sitl jmavsim

commander takeoff
commander land

The problem occurs when entering "make px4_sitl jmavsim" or "make px4_sitl gazebo" i get the following (this copypaste is made with jmavsim):

nico@nico-VirtualBox:~/Firmware$ make px4_sitl jmavsim
[0/1] cd /home/nico/Firmware/build/px4_sitl_defa...mware /home/nico/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/nico/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /home/nico/Firmware
build_path: /home/nico/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/nico/Firmware/build/px4_sitl_default/bin/px4" "/home/nico/Firmware/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/nico/Firmware"/test_data
Creating symlink /home/nico/Firmware/build/px4_sitl_default/etc -> /home/nico/Firmware/build/px4_sitl_default/tmp/rootf

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
INFO  [parameters] BSON document size 470 bytes, decoded 470 bytes
[param] Loaded: eeprom/parameters_10016
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Buildfile: /home/nico/Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 1 second
Options parsed, starting Sim.
Starting GUI...
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
libEGL warning: DRI2: failed to authenticate
INFO  [simulator] Simulator connected on TCP port 4560.
Init MAVLink
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2021-08-24/23_33_00.ulg
INFO  [logger] Opened full log file: ./log/2021-08-24/23_33_00.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative

After this if i enter "commander takeoff" i get:

pxh> commander takeoff
pxh> INFO  [commander] Armed by internal command
INFO  [navigator] Using minimum takeoff altitude: 2.50 m
INFO  [commander] Takeoff detected
WARN  [commander] Failsafe enabled: no datalink
INFO  [commander] Failsafe mode activated
INFO  [tone_alarm] battery warning (fast)
INFO  [navigator] RTL HOME activated
INFO  [navigator] RTL: landing at home position.	
INFO  [navigator] RTL: climb to 489 m (1 m above destination)	
INFO  [commander] Failsafe mode deactivated
INFO  [navigator] RTL: return at 489 m (1 m above destination)	
INFO  [navigator] RTL: land at destination	
INFO  [commander] Landing detected
INFO  [commander] Disarmed by landing
INFO  [logger] closed logfile, bytes written: 7246978

I do not know if it is supposed to do something else or different, I have rebooted, updated, even deleted and reset the virtual machine image ... I appreciate any help. In addition, my intention is to do HITL simulation later, my university has provided me with a px4 mini for it, but I don't want to skip this part, I don't even know if I can but I want to simulate it in SITL first

potaito, 2021-08-25:
I think it would be better to post these things on Slack, since it's not really an issue in PX4 that you are reporting.

If you look through the output after you attempted a takeoff via console, you'll see that it rejects the takeoff because there is no datalink:
```
WARN [commander] Failsafe enabled: no datalink
INFO [commander] Failsafe mode activated
INFO [tone_alarm] battery warning (fast)
```

Download QGroundControl and open it. It should automatically connect to the simulation on your local machine (I hope the virtual machine passes everything through). Then use QGC to perform a takeoff. It will mostly likely complain that you don't have an RC, so either connect a radio / gamepad to the computer running QGC, or enable virtual joysticks in QGC, or find and disable the parameter that is responsible for checking the RC link. 

I'm sorry if my instructions in the other issue to run `commander takeoff` confused you. This is not the intended workflow for a new user. I was just trying to debug the issue.

NicolassFelipe, 2021-08-25:
@potaito 

Thank you very much, i will try what you said and if it doesnt work ill post this on Slack. I thought it might be useful to try with QGC but as the guides mention it on "next steps" i thought this part should work without it....  thks again!

NicolassFelipe, 2021-08-25:
@potaito Thank you very much, i will try what yoy said. Your instructions where fine on the other issue btw, the other person gave a little less details so its ok.

NicolassFelipe, 2021-08-25:
@potaito @potaito Thank you very much, i will try what yoy said. Your instructions where fine on the other issue btw, the other person gave a little less details so its ok.

Im trying to get into px4stack from here : https://px4.io/community/community-overview/ but link doesnt seem to work.

SEUZTh, 2021-09-05:
Hi, I met the same problem. Have you solved it?

NicolassFelipe, 2021-09-05:
@SEUZTh  Hi Yes. I did what Potaito commented above, opening QGroundControl first allows to simulate well. It doesn't appear in the Px4 guides but it makes sense considering the diagram shown at http://docs.px4.io/master/en/simulation/.

SEUZTh, 2021-09-05:
> @SEUZTh Hi Yes. I did what Potaito commented above, opening QGroundControl first allows to simulate well. It doesn't appear in the Px4 guides but it makes sense considering the diagram shown at http://docs.px4.io/master/en/simulation/.

Thanks for your reply.
I am using Ubuntu 20.04 arm64 on MacBook M1. But QT haven't supported arm64. So I can't use QGC.
Maybe I need to build QGC which can run on arm64.

richtong, 2021-09-06:
> @NicolassFelipe Not sure where you got https://slack.px4.io/, but the correct link should be **https://px4.slack.com**

I issued a PR on that doc error, px4.slack.com does work :-) and that other link slack.px4.io is brokn

hamishwillee, 2021-09-06:
I've asked @mrpollo to please look at the link on px4.io. I can fix broken links on the docs.

NicolassFelipe, 2021-09-06:
> 
> 
> > @NicolassFelipe Not sure where you got https://slack.px4.io/, but the correct link should be **https://px4.slack.com**
> 
> I issued a PR on that doc error, px4.slack.com does work :-) and that other link slack.px4.io is brokn

Thx! i tried to enter the px4 slack, but it says i don't have an account on this workspace, i have a registered mail account and i can use another slack... can u help me with that?

Jaeyoung-Lim, 2021-09-07:
@NicolassFelipe https://slack.com/intl/en-ch/help/articles/212675257-Join-a-Slack-workspace

richtong, 2021-09-09:
I think you can close this issue the fix is to run qgroundcontrol so you don't get the data link message (it needs something to connect to for takeoff).

Jaeyoung-Lim, 2021-09-09:
@richtong Thanks

closed 2021-09-09 by Jaeyoung-Lim

mengchaoheng, 2022-04-06:
> Thanks for your reply. I am using Ubuntu 20.04 arm64 on MacBook M1. But QT haven't supported arm64. So I can't use QGC. Maybe I need to build QGC which can run on arm64.

you are right! you need build QGC, but the problem is build Qt on arm64 is very very difficult. I have try to build qgc by docker containers in which have build and installed Qt. But failed. more info can be seem in another [issue](https://github.com/mavlink/qgroundcontrol/issues/10237).

mengchaoheng, 2022-04-06:
> Thanks for your reply. I am using Ubuntu 20.04 arm64 on MacBook M1. But QT haven't supported arm64. So I can't use QGC. Maybe I need to build QGC which can run on arm64.

原来是东南的大佬，失敬失敬。请问您解决了吗，我发现我们这个需求太小众了。暂时没多少人在m1 的虚拟机上玩px4。现在只跑仿真的话是没问题，只是没有qgc，就不完美了。我在docker上跑了几个在arm下编译了qt的容器，但他们都会缺少一些包，qgc需要太多的包了。我自己暂时没搞懂怎么自定义编译一些包。直接从源码编译qt也遇到了不少问题，都是搜不到解决方案的，在qt论坛上咨询了源码编译问题，很多大佬纷纷摇头表示无解。