PX4/PX4-Autopilot #18165: Can not upload PX4 autopilot on Crazyflie 2.1 using command line

shrit, 2021-08-29:
Hello,

I have been trying to put PX4 image on the Crazflie 2.1 quadrotors, I have followed all the steps explained in the tutorials, starting from using dfu to cross-compile the autopilot. However, it stuck automatically at the upload steps.

Using `htop`, I was able to see the command being executed, it seems that it is using python 3 and trying to open `/dev/serial` on my machine which is nowhere to be found. I think that might be the reason for this issue.

I was able eventually to upload the image using Qgroundcontrol. I was wondering if there are any tutorials to explain how to use the python command directly without relying on the `upload` option of the `make` command.

I am using Arch Linux instead of Ubuntu, I am not sure if this might cause the issue in the first place, but I think it might be relevant in order to reproduce the issue.
To reproduce, simply execute the same steps on Arch Linux.
Let me know if this is helpful or if more information is needed for further investigation.

TheLegendaryJedi, 2021-09-08:
Hi @shrit

Do you have the battery unplugged?

Best regards

shrit, 2021-09-09:
Probably yes.

TheLegendaryJedi, 2021-09-09:
Make sure you are doing all the process with battery unplugged. If you can't, ping me again.

closed 2025-11-30 by github-actions[bot]