PX4/PX4-Autopilot #18171: Calibration is required each time after connecting the battery

shrit, 2021-08-30:
Hello,

It seems that the calibration is required each time after I disconnect the battery and reconnect it, this behavior is observed on the Crazyflie 2.1 board.
If I remember right, calibration data should be saved to non-volatile memory, which makes sense as we do not have to calibrate with the original firmware.

Is there any way to overcome this issue or a workaround to save them manually? I am using the master version of PX4 autopilot.

Many thanks,

YHuniv, 2025-01-15:
Same problem ? did you resolve it ?
Thanks

shrit, 2025-01-15:
At that time I do not think that Crazyflie was supported by PX4, so basically, I never completed it and never flied the Crazyflie with PX4 unfortunately

closed 2025-11-30 by github-actions[bot]