PX4/PX4-Autopilot #18177: ERROR [param] not enough arguments in the VTOL gazebo simulation at v1.12.2

Kim-zhi-jiang, 2021-09-01:
From: PX4-Autopilot/src/systemcmds/param/param.cpp
line 292

[image]

INFO  [init] Mixer: etc/mixers-sitl/standard_vtol_sitl.main.mix on /dev/pwm_output0
ERROR [param] not enough arguments.
Try 'param set PWM_AUX_OUT 3 [fail]'

hamishwillee, 2021-09-02:
FWIW I can reproduce this behaviour against tag v1.12.2 using `HEADLESS=1 make px4_sitl gazebo_standard_vtol`. I also see it in master.

The simulated vehicle appears to fly OK. 

@bkueng is this something to worry about? AUX3 is supposed to be an elevator :-)

Kim-zhi-jiang, 2021-10-02:
Still have

closed 2021-10-02 by Kim-zhi-jiang