PX4/PX4-Autopilot #18180: Relogin or reboot computer before attempting to build NuttX targets

SEUZTh, 2021-09-01:
When I use `bash ./PX4-Autopilot/Tools/setup/ubuntu.sh --no-sim-tools`, I met a problem:
```
Installing NuttX dependencies
Reading package lists...
Building dependency tree...
Reading state information...
Package gcc-multilib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package g++-multilib
E: Couldn't find any package by regex 'g++-multilib'
E: Package 'gcc-multilib' has no installation candidate
arm-none-eabi-gcc-9-2020-q2-update found, skipping installation

Relogin or reboot computer before attempting to build NuttX targets
```
I am try to configure PX4 on MacBook M1 Parallels Desktop 17.04 Ubuntu 20.04 arm64 ROS2 foxy.
Who can help me?

dagar, 2021-09-01:
This is arm64 ubuntu? So far `Tools/setup/ubuntu.sh` has really been focused on desktop developer setups (amd64). We'll need to if people are starting to do actual development on arm64.

As a start can you strip out packages like `gcc-multilib` and see how far you get?

SEUZTh, 2021-09-02:
> This is arm64 ubuntu? So far `Tools/setup/ubuntu.sh` has really been focused on desktop developer setups (amd64). We'll need to if people are starting to do actual development on arm64.
> 
> As a start can you strip out packages like `gcc-multilib` and see how far you get?

Thanks for your reply. The following command is valid for me.
```
bash ./PX4-Autopilot/Tools/setup/ubuntu.sh --no-nuttx --no-sim-tools
```

closed 2021-09-02 by SEUZTh

mengchaoheng, 2022-02-27:
The same problem is bothering me. I have tried to configure PX4 on my MacBook M1 pro and it failed. Now I try to compile PX4 with Ubuntu 20.04 arm64 on a MacBook using Parallels Desktop 17.04, same fails. Why is m1 so unpopular.

dagar, 2022-02-27:
M1 support is just new, and the majority of developers are using Linux (amd64) as their primary development environment.

What error do you get within parallels specifically? This is likely a separate issue than your M1 struggles.

mengchaoheng, 2022-02-27:
> M1 support is just new, and the majority of developers are using Linux (amd64) as their primary development environment.
> 
> What error do you get within parallels specifically? This is likely a separate issue than your M1 struggles.

the same like that:
Installing NuttX dependencies
Reading package lists...
Building dependency tree...
Reading state information...
Package gcc-multilib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package g++-multilib
E: Couldn't find any package by regex 'g++-multilib'
E: Package 'gcc-multilib' has no installation candidate
arm-none-eabi-gcc-9-2020-q2-update found, skipping installation

Relogin or reboot computer before attempting to build NuttX targets

nihavend, 2022-04-11:
Having a similar issue :
Installing NuttX dependencies
Reading package lists...
Building dependency tree...
Reading state information...
Package gcc-multilib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package g++-multilib
E: Couldn't find any package by regex 'g++-multilib'
E: Package 'gcc-multilib' has no installation candidate

nihavend, 2022-04-11:
> > Having a similar issue : Installing NuttX dependencies Reading package lists... Building dependency tree... Reading state information... Package gcc-multilib is not available, but is referred to by another package. This may mean that the package is missing, has been obsoleted, or is only available from another source
> > E: Unable to locate package g++-multilib E: Couldn't find any package by regex 'g++-multilib' E: Package 'gcc-multilib' has no installation candidate
> 
> use：
> 
> ```
> bash ubuntu.sh --no-nuttx
> ```

Thanks @mengchaoheng it solved the build issue. what i lose with disabling nuttx ?

mengchaoheng, 2022-04-11:
> Thanks @mengchaoheng it solved the build issue. what i lose with disabling nuttx ?

if you build on arm-based ubuntu, it didn't supported now.

mengchaoheng, 2022-04-11:
> disabling

it mean, you just could use sitl but not nuttx

mengchaoheng, 2022-04-19:
> strip out packages like `gcc-multilib`

@SEUZTh @nihavend @dagar it work now! strip out packages like gcc-multilib and g++-multilib, just delete g++multilib and gcc-multilib in ubuntu.bash file. and then 
```
bash Tools/setup/ubuntu.sh 
```
and then here we go:
``` 
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ make px4_fmu-v5_default
-- PX4 version: v1.13.0-alpha1-4371-gc2c455be0d
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/parallels/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- IO px4_io-v2_default
-- SERIAL_GPS1 /dev/ttyS0
-- SERIAL_TEL1 /dev/ttyS1
-- SERIAL_TEL2 /dev/ttyS2
-- SERIAL_TEL4 /dev/ttyS3
-- UAVCAN_INTERFACES 2
-- UAVCAN_TIMER_OVERRIDE 6
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 9.3.1
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Enabling double FP precision hardware instructions
-- drivers/px4io: ROMFS including px4_io-v2_default
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding platforms/nuttx/init/stm32f7/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_fmu-v5_bootloader.bin -> /etc/extras/px4_fmu-v5_bootloader.bin
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_io-v2_default.bin -> /etc/extras/px4_io-v2_default.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parallels/PX4-Autopilot/build/px4_fmu-v5_default
[0/1124] git submodule src/drivers/gps/devices
[3/1124] git submodule platforms/nuttx/NuttX/nuttx
[9/1124] git submodule src/drivers/uavcan/libuavcan
[15/1124] git submodule src/modules/mavlink/mavlink
[16/1124] Generating Mavlink developme...ssage_definitions/v1.0/development.xml
Validating /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Parsing /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Parsing /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Validating /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Parsing /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Validating /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Parsing /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Merged enum MAV_CMD
Merged enum MAV_CMD
Found 228 MAVLink message types in 4 XML files
Generating C implementation in directory /home/parallels/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/development
Generating C implementation in directory /home/parallels/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/standard
Generating C implementation in directory /home/parallels/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/common
Generating C implementation in directory /home/parallels/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/minimal
Copying fixed headers for protocol 2.0 to /home/parallels/PX4-Autopilot/build/px4_fmu-v5_default/mavlink
[1122/1124] Linking CXX executable px4_fmu-v5_default.elf
Memory region         Used Size  Region Size  %age Used
      FLASH_ITCM:          0 GB      2016 KB      0.00%
      FLASH_AXIM:     1937781 B      2016 KB     93.87%
        ITCM_RAM:          0 GB        16 KB      0.00%
        DTCM_RAM:          0 GB       128 KB      0.00%
           SRAM1:       45992 B       368 KB     12.20%
           SRAM2:          0 GB        16 KB      0.00%
[1124/1124] Creating /home/parallels/PX4-Autopilot/build/px4_fmu-v5_default/px4_fmu-v5_default.px4

```

mengchaoheng, 2022-04-19:
@dagar What is the actual role of g++-multilib and g++-multilib? I mean, if we don't need them, why install by ubuntu.sh