PX4/PX4-Autopilot #18181: Getting started with micrortps

eden-desta, 2021-09-01:
Hi!
Followed this guide: https://docs.px4.io/master/en/middleware/micrortps.html and everything works as expected. I was trying to get micrortps working on my own board. 

```
	SYSTEMCMDS
		param
		perf
		reboot
		top
	MODULES
		micrortps_bridge
```
I have this in the `default.cmake` file. I imagine there are other things I am missing in order to have call `micrortps_client star`t -t UDP`

I get an error saying that micrortps_client is not a known command.

I am absolutely new to this so i might be missing something extremely obvious.

I appreciate your help! Thank you!

TSC21, 2021-09-01:
Adding `micrortps_bridge` as you are doing should be enough. Make sure you make clean and then rebuild. Though there's just a detail: unless your board has Ethernet, you will not be able to use UDP. You should use UART. Make sure you configure the device and baud accordingly to the TELEM/UART port you will be using (UDP is shown on the docs as an example on how to use the bridge with SITL).

eden-desta, 2021-09-01:
You are absolutely right! Haha i should have done a clean before thank you @TSC21!

closed 2021-09-01 by eden-desta