PX4/PX4-Autopilot #18192: Strange behavior of FW pitch rate PID after transition 1.11.3 -> 1.12.1

AndreiShu, 2021-09-04:
After upgrading from 1.11.3 to 1.12.1 our drone (VTOL, tailsitter) became unstable. As we understand now that the main problem was in gyro filters. Somehow during to transition IMU_DGYRO_CUTOFF became equal to 0. (it is perhaps some minor bug in the transition between versions). However, it seems that it is not the only problem. 
During the experiments we found that FW pitch_rate controller with PR_FW_FF = 0 and PR_FF_I = 0 is not equal to its proportional gain. See, for example, the LOG https://logs.px4.io/plot_app?log=d0fd29dc-a149-4233-90a7-5663924d443a

The difference between controller results and proportional gain is similar (but not equal) to -0.2*(pitch rate setpoint). On the plot attached red curve is the difference between actual result of controller and calculated proportional gain. Black is pitch rate setpoint (times -0.2). 

Is it a bug? Or if it is something intentional than what is added to FW control besides usual FF, P and I components? How can we control (and understand) this addition? It seems that even with filters corrected (and the PID parameters re-tuned due to the difference in airspeed scaler between 1.11.3 and 1.12.1) our VTOL is still less stable than it was in 1.11.3.
(The situation is the same in 1.12.1 and 1.12.2)

AndreiShu, 2021-09-04:
[image]

The mentioned plot

AndreiShu, 2021-09-04:
[image]

the mentioned plot

dagar, 2021-09-04:
There's also airspeed gain scheduling.

https://github.com/PX4/PX4-Autopilot/blob/d32f0d6168ffe0e0998b160a07a7cf447cd68b7a/src/modules/fw_att_control/ecl_pitch_controller.cpp#L106-L109

https://github.com/PX4/PX4-Autopilot/blob/d32f0d6168ffe0e0998b160a07a7cf447cd68b7a/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L259-L262

The airspeed is constrained between [FW_AIRSPD_STALL, FW_AIRSPD_MAX] for scaling.

``` Console
FW_AIRSPD_MAX, 40.0
FW_AIRSPD_MIN, 10.899999618530273
FW_AIRSPD_STALL, 9.800000190734863
FW_AIRSPD_TRIM, 11.0
```

AndreiShu, 2021-09-05:
Thank you for the answer. 

We understand that there is a scaler ctl_data.scaler in the expression for the controller. To the best of our knowledge in our case it should be equal to constant 
FW_AIRSPD_TRIM/FW_AIRSPD_STALL =  11/9.8

We took it into account when calculated P-component. 
[image]

On the plot blue line is the output of the controller, yellow line is P-componet with coefficient
FW_PR_P*scaler^2
In any case, the curves are not proportional to one another (as it would be if we miscalculated effective P-coefficient). The difference between this two curves is the red curve on the plot that I showed initially.

AndreiShu, 2021-09-05:
And as long as FW_PR_FF=0 scaler (with any value) should not be able to add FF-component.

moreba1, 2021-12-15:
@AndreiShu 
rate setpoint differs from rate error. you must display kpp*(rate_setpoint - current_rate)*scaler^2 .