PX4/PX4-Autopilot #18209: Simulator doesn't start on using 'make px4_sitl jmavsim'

theflash-17, 2021-09-08:
I was following the steps mentioned in the 'Toolchain Installation' (Windows Setup) section of PX4 User Guide (https://docs.px4.io/master/en/dev_setup/dev_env_windows_cygwin.html).

After following all the 4 steps, when I used the command `make px4_sitl jmavsim` in the 'run-console', the simulation didn't start and the process was stuck at `INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560` .

```
$ make px4_sitl jmavsim                                                                                                                                                        
[0/1] cd /cygdrive/c/PX4/home/PX4-Autopilot/build/px4_sitl_default/tmp && /cygdrive/c/...ive/c/PX4/home/PX4-Autopilot /cygdrive/c/PX4/home/PX4-Autopilot/build/px4_sitl_defaul SITL ARGS
sitl_bin: /cygdrive/c/PX4/home/PX4-Autopilot/build/px4_sitl_default/bin/px4.exe                                                                                                
debugger: none                                                                                                                                                                 
program: jmavsim                                                                                                                                                              
model: none                                                                                                                                                                    
world: none                                                                                                                                                                    
src_path: /cygdrive/c/PX4/home/PX4-Autopilot                                                                                                                                   build_path: /cygdrive/c/PX4/home/PX4-Autopilot/build/px4_sitl_default                                                                                                          empty model, setting iris as default                                                                                                                                           
SITL COMMAND: "/cygdrive/c/PX4/home/PX4-Autopilot/build/px4_sitl_default/bin/px4.exe" "/cygdrive/c/PX4/home/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/cygdrive/c/PX4/home/PX4-Autopilot"/test_data                                                                                                                               
Creating symlink /cygdrive/c/PX4/home/PX4-Autopilot/build/px4_sitl_default/etc -> /cygdrive/c/PX4/home/PX4-Autopilot/bu                                                                                                                                                                                                                                     
_____  __   __    ___                          
| ___ \ \ \ / /   /   |                                                                                                                                                        
| |_/ /  \ V /   / /| |                                                                                                                                                        
|  __/   /   \  / /_| |                                                                                                                                                        
| |     / /^\ \ \___  |                                                                                                                                                        
\_|     \/   \/     |_/                                                                                                                                                                                                                                                                                                                                       

px4 starting.                                                                                                                                                                                                                                                                                                                                                 

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0                                                                                                             
INFO  [init] found model autostart file as SYS_AUTOSTART=10016                                                                                                                 
INFO  [param] selected parameter default file eeprom/parameters_10016                                                                                                          
[param] parameter file not found, creating eeprom/parameters_10016                                                                                                               
   SYS_AUTOCONFIG: curr: 0 -> new: 1                                                                                                                                              
   SYS_AUTOSTART: curr: 0 -> new: 10016                                                                                                                                           
   CAL_ACC0_ID: curr: 0 -> new: 1310988                                                                                                                                           
   CAL_GYRO0_ID: curr: 0 -> new: 1310988                                                                                                                                          
   CAL_ACC1_ID: curr: 0 -> new: 1310996                                                                                                                                           
   CAL_GYRO1_ID: curr: 0 -> new: 1310996                                                                                                                                          
   CAL_ACC2_ID: curr: 0 -> new: 1311004                                                                                                                                           
   CAL_GYRO2_ID: curr: 0 -> new: 1311004                                                                                                                                          
   CAL_MAG0_ID: curr: 0 -> new: 197388                                                                                                                                            
   CAL_MAG1_ID: curr: 0 -> new: 197644                                                                                                                                            
   SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000                                                                                                                                  
   SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010                                                                                                                                  
Buildfile: C:\PX4\home\PX4-Autopilot\Tools\jMAVSim\build.xml                                                                                                                                                                                                                                                                                                  

make_dirs:                                                                                                                                                                                                                                                                                                                                                    

compile:                                                                                                                                                                                                                                                                                                                                                      

create_run_jar:                                                                                                                                                                                                                                   [... cut ...]
  IMU_INTEG_RATE: curr: 200 -> new: 250                                                                                                                                        
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes                                                                                          
PX4 SIM HOST: localhost                                                                                                                                                        
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560    
```

NicolassFelipe, 2021-09-12:
https://github.com/PX4/PX4-Autopilot/issues/18142

theflash-17, 2021-09-13:
@NicolassFelipe Thanks for the comment. I had followed the discussion of #18142 and tried the solution provided by @potaito.
Initially, it didn't work for me. So, I did try different combinations in hope of solving it. 
(**Note:** In all cases, I tried running 'jmavsim' only)

The simulation ran fine when I ran it on 'ubuntu-20.04.3-desktop-amd64' and 'windows 10' virtual machines. For both of them, I followed the method show in this video - https://www.youtube.com/watch?v=OtValQdAdrU (I only cloned the PX4/Firmware repo instead of PX4/PX4-Autopilot) and had launched QGroundControl before using the command - `make px4_sitl jmavsim`.

However, the problem still persists as I am unable to run it using the PX4-Autopilot repo (i.e., by following the method given in https://docs.px4.io/master/en/dev_setup/dev_env_windows_cygwin.html) even after launching QGroundControl at the beginning.

theflash-17, 2021-09-13:
Also, I tried running `make px4_fmu-v5_default` in PX4/Firmware, it didn't work.
Here is the output:
```
$ make px4_fmu-v5_default                                                                                               
[0/21] Performing build step for 'px4io_firmware'                                                                       
[1/3] Linking CXX executable px4_io-v2_default.elf                                                                      
FAILED: px4_io-v2_default.elf                                                                                           
: && /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-g++.exe  -mcpu=cortex-m3 -mthumb -Os -DNDEBUG  --specs=nosys.specs platforms/nuttx/CMakeFiles/px4.dir/__/common/empty.
c.obj  -o px4_io-v2_default.elf -L/cygdrive/c/PX4/home/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware/external/Install/lib -nostartfiles -nodefaultlibs -nostd
lib -nostdinc++ -fno-exceptions -fno-rtti -Wl,--script=C:/PX4/home/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware/NuttX/nuttx-config/scripts/script.ld -Wl,-Ma
p=px4_io-v2_default.map -Wl,--warn-common -Wl,--gc-sections -Wl,--start-group NuttX/apps/libapps.a NuttX/nuttx/arch/arm/src/libarch.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nutt
x/libs/libc/libc.a NuttX/nuttx/boards/libboards.a NuttX/nuttx/libs/libxx/libxx.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sch
ed/libsched.a -Wl,--end-group -lm -lgcc -Wl,--print-memory-usage src/modules/px4iofirmware/libpx4iofirmware.a NuttX/apps/libapps.a platforms/nuttx/src/px4/stm/stm32f1/io_pins/
libarch_io_pins.a platforms/nuttx/src/px4/stm/stm32f1/watchdog/libarch_watchdog_iwdg.a NuttX/nuttx/arch/arm/src/libarch.a boards/px4/io-v2/src/libdrivers_board.a platforms/nut
tx/src/px4/stm/stm32f1/hrt/libarch_hrt.a src/lib/parameters/libparameters.a src/lib/parameters/flashparams/libflashparams.a NuttX/nuttx/arch/arm/src/libarch.a boards/px4/io-v2
/src/libdrivers_board.a platforms/nuttx/src/px4/stm/stm32f1/hrt/libarch_hrt.a src/lib/parameters/libparameters.a src/lib/parameters/flashparams/libflashparams.a src/lib/parame
ters/tinybson/libtinybson.a platforms/common/libpx4_platform.a msg/libuorb_msgs.a src/lib/mixer/libmixer.a src/lib/mixer/AllocatedActuatorMixer/libAllocatedActuatorMixer.a src
/lib/mixer/HelicopterMixer/libHelicopterMixer.a src/lib/mixer/MultirotorMixer/libMultirotorMixer.a src/lib/mixer/NullMixer/libNullMixer.a src/lib/mixer/SimpleMixer/libSimpleMi
xer.a src/lib/mixer/MixerBase/libMixerBase.a src/lib/rc/librc.a src/lib/output_limit/liboutput_limit.a NuttX/nuttx/libs/libxx/libxx.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/libs/l
ibc/libc.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -lm -lgcc && :                                                                      
Memory region         Used Size  Region Size  %age Used                                                                            
           flash:       61896 B        60 KB    100.74%                                                                             
            sram:        3728 B         8 KB     45.51c:/px4/toolchain/gcc-arm/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: px4_io-v2_default.elf s
ection `.data' will not fit in region `flash'                    
c:/px4/toolchain/gcc-arm/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: region `flash' overflowed by 456 bytes                                                                                                        
collect2.exe: error: ld returned 1 exit status                                                                          
%                                                                                                                       
ninja: build stopped: subcommand failed.                                                                                
FAILED: external/Stamp/px4io_firmware/px4io_firmware-build external/Build/px4io_firmware/px4_io-v2_default.elf          
cd /cygdrive/c/PX4/home/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware && /usr/bin/cmake.exe --build . 
ninja: build stopped: subcommand failed.                                                                                
make: *** [Makefile:225: px4_fmu-v5_default] Error 1
```

NicolassFelipe, 2021-09-13:
maybe try with ubuntu 18.04 LTS, at first i tried with 20.4 too but didnt work. then i read on px4 documentation that 18.04 is the last one that you can be sure that it will work

julianoes, 2021-09-14:
@theflash-17 which Java version have you installed?
And does it help to rebuild jMAVSim. You can delete the old build by doing: `rm -rf Tools/jMAVSim/out`.

potaito, 2021-09-20:
@theflash-17 , to add to Julian's response, your jmavsim is crashing. That's why the simulation is not working. Look at the following lines from your output
```
Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0                                                                                               
         at java.lang.ClassLoader.defineClass1(Native Method)                                                                                                                           
         at java.lang.ClassLoader.defineClass(ClassLoader.java:763)                                                                                                                     
         at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)                                                                                                     
         at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)                                                                                                                
         at java.net.URLClassLoader.access$100(URLClassLoader.java:73)                                                                                                                  
         at java.net.URLClassLoader$1.run(URLClassLoader.java:368)                                                                                                                      
         at java.net.URLClassLoader$1.run(URLClassLoader.java:362)                                                                                                                      
         at java.security.AccessController.doPrivileged(Native Method)                                                                                                                  
         at java.net.URLClassLoader.findClass(URLClassLoader.java:361)                                                                                                                  
         at java.lang.ClassLoader.loadClass(ClassLoader.java:424)                                                                                                                       
         at java.lang.ClassLoader.loadClass(ClassLoader.java:357)                                                                                                                       
         at java.lang.Class.forName0(Native Method)                                                                                                                                     
         at java.lang.Class.forName(Class.java:348)                                                                                                                                     
         at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:59) 
``` 
The top line also tells you, that you most likely downgraded your java installation from v55 to v52 after running jmavsim at least once, which is why the compiled binary is now no longer compatible with your currently installed version.

With the virtual machines you don't have this problem since you probably did a fresh install of everything, including Java.

Julian's fix above should solve the problem, since it will force a recompilation with the java version currently installed on your system.

closed 2021-09-20 by potaito