PX4/PX4-Autopilot #18216: Mantis EDU todo list

julianoes, 2021-09-09:
This is to capture work left for the ATL Mantis EDU:

- [ ] Test flow
- [ ] Fix beeps
- [ ] Add beep when switching off
- [ ] Test power shutdown
- [ ] Extract channels from RC_CHANNELS_OVERRIDE
- [ ] Parse buttons from RC_CHANNELS_OVERRIDE
- [ ] Test gimbal using RC
- [ ] Test gimbal from mission
- [ ] Make photo and recording work from PX4 using RC buttons
- [ ] Make firmware upload via http work
- [ ] Add log streaming to SD card or get it to work with QGC
- [ ] Send camera information with camera definition xml for camera settings
- [ ] Send camera video stream information correctly

dagar, 2021-09-14:
Notes from the original PR. https://github.com/PX4/PX4-Autopilot/pull/17910#issuecomment-892266046

julianoes, 2021-09-24:
Docs coming soon: https://github.com/PX4/PX4-user_guide/pull/1560

julianoes, 2021-10-18:
The two issues above should be addressed:
1. Cherry-picked https://github.com/PX4/PX4-Autopilot/pull/18337 which should fix it.
2. Flipped flow orientation, so it should now be correct.