PX4/PX4-Autopilot #18219: PX4's simulation with gazebo compile errors on MAC

kevinuav, 2021-09-10:
I encounter gazebo errors while it is compiling with my Mac.I ran gazebo on my CatalinaOS v10.5.1 very well before. But it can naver compile successfully after that I updated my code from v1.11.0 to v1.12.0. Even I have formatted the whole disk of my Mac.And reinstalled Big Sur v11.5.1.Then I reinstalled everything as the instruction carefully. It still get the very same errors as before.
Why and how to fix it?
The attach is the whole output of the compiling:
gazebo's compiling output.odt

dagar, 2021-09-11:
What happens if you run the macos.sh setup script with current master?

``` Console
 % ./Tools/setup/macos.sh
```

kevinuav, 2021-09-13:
> What happens if you run the macos.sh setup script with current master?
> 
> ```
>  % ./Tools/setup/macos.sh
> ```

% /Users/kevin/PX4-Autopilot/Tools/setup/macos.sh
px4-dev already installed
Installing PX4 Python3 dependencies
Requirement already satisfied: argcomplete in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 1)) (1.12.3)
Collecting argparse>=1.2
  Using cached argparse-1.4.0-py2.py3-none-any.whl (23 kB)
Requirement already satisfied: cerberus in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 3)) (1.3.4)
Requirement already satisfied: coverage in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 4)) (5.5)
Requirement already satisfied: empy>=3.3 in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 5)) (3.3.4)
Requirement already satisfied: jinja2>=2.8 in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 6)) (3.0.1)
Requirement already satisfied: matplotlib>=3.0.* in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 7)) (3.5.0b1)
Requirement already satisfied: numpy>=1.13 in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 8)) (1.21.2)
Requirement already satisfied: nunavut>=1.1.0 in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 9)) (1.4.2)
Requirement already satisfied: packaging in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 10)) (21.0)
Requirement already satisfied: pandas>=0.21 in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 11)) (1.3.2)
Requirement already satisfied: pkgconfig in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 12)) (1.5.5)
Requirement already satisfied: psutil in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 13)) (5.8.0)
Requirement already satisfied: pygments in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 14)) (2.10.0)
Requirement already satisfied: wheel>=0.31.1 in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 15)) (0.37.0)
Requirement already satisfied: pymavlink in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 16)) (2.4.15)
Requirement already satisfied: pyros-genmsg in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 17)) (0.5.8)
Requirement already satisfied: pyserial in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 18)) (3.5)
Requirement already satisfied: pyulog>=0.5.0 in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 19)) (0.9.0)
Requirement already satisfied: pyyaml in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 20)) (5.4.1)
Requirement already satisfied: requests in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 21)) (2.26.0)
Requirement already satisfied: setuptools>=39.2.0 in /usr/local/lib/python3.9/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 22)) (57.4.0)
Requirement already satisfied: six>=1.12.0 in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 23)) (1.16.0)
Requirement already satisfied: toml>=0.9 in ./Library/Python/3.9/lib/python/site-packages (from -r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 24)) (0.10.2)
Requirement already satisfied: MarkupSafe>=2.0 in ./Library/Python/3.9/lib/python/site-packages (from jinja2>=2.8->-r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 6)) (2.0.1)
Requirement already satisfied: cycler>=0.10 in /usr/local/lib/python3.9/site-packages (from matplotlib>=3.0.*->-r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 7)) (0.10.0)
Requirement already satisfied: kiwisolver>=1.0.1 in /usr/local/lib/python3.9/site-packages (from matplotlib>=3.0.*->-r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 7)) (1.3.2)
Requirement already satisfied: pillow>=6.2.0 in /usr/local/lib/python3.9/site-packages (from matplotlib>=3.0.*->-r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 7)) (8.3.2)
Requirement already satisfied: fonttools>=4.22.0 in /usr/local/lib/python3.9/site-packages (from matplotlib>=3.0.*->-r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 7)) (4.26.2)
Requirement already satisfied: setuptools-scm>=4 in /usr/local/lib/python3.9/site-packages (from matplotlib>=3.0.*->-r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 7)) (6.3.1)
[... cut ...]
Requirement already satisfied: charset-normalizer~=2.0.0 in /usr/local/lib/python3.9/site-packages (from requests->-r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 21)) (2.0.4)
Requirement already satisfied: tomli>=1.0.0 in /usr/local/lib/python3.9/site-packages (from setuptools-scm>=4->matplotlib>=3.0.*->-r /Users/kevin/PX4-Autopilot/Tools/setup/requirements.txt (line 7)) (1.2.1)
Installing collected packages: argparse
Successfully installed argparse-1.4.0
All set! PX4 toolchain installed!

julianoes, 2021-09-14:
Have a look through this: https://stackoverflow.com/questions/58628377/catalina-c-using-cmath-headers-yield-error-no-member-named-signbit-in-th

kevinuav, 2021-09-14:
Thanks for your help.
    Dagar and Julianoes. I have fixed the problem after many Time cleaned up and installed everything on my MAC. The problem is caused by the Mac's pythons. After it has been cleaned up. Then the gazebo works!
here's the related issue:  https://gist.github.com/MuhsinFatih/ee0154199803babb449b5bb98d3475f7

closed 2021-09-14 by kevinuav

julianoes, 2021-09-15:
Thanks for the report and link @kevinuav.