PX4/PX4-Autopilot #18221: Add webots support

dzywater, 2021-09-10:


dagar, 2021-09-10:
Could you add any specifics? What would the advantage over Gazebo or Ignition Gazebo be? Are you interested in doing the integration?

dzywater, 2021-09-12:
> Could you add any specifics? What would the advantage over Gazebo or Ignition Gazebo be? Are you interested in doing the integration?

1.Controllers can be written in c, c++, python, and MatLab.
2.A lot of sensors exist.
3.Ability to add custom physics to simulate things such as wind.
4.Ability to add OpenStreetMap and run the simulator in environments very similar to reality
Above is from https://ardupilot.org/dev/docs/sitl-with-webots.html#advantage-of-webots.
Especially, its  class/API definition is brief, understandable and user-friendly.

Beside,
(1)Texture is closer to the real world and it is better to computer vision algorithm development.
(2)Be easier to install on not only Linux but Windows and can attract more users.

I am interested to contribute, but someone can define a framework first.

angel-ayala, 2022-07-19:
I have interest in do this integration too, how can I help?

dagar, 2022-07-19:
> I have interest in do this integration too, how can I help?

@angel-ayala @dzywater should we arrange a short call to discuss? Either join the weekly dev call (https://docs.px4.io/main/en/contribute/dev_call.html) or ping me on PX4 slack.

angel-ayala, 2022-07-19:
> > I have interest in do this integration too, how can I help?
> 
> @angel-ayala @dzywater should we arrange a short call to discuss? Either join the weekly dev call (https://docs.px4.io/main/en/contribute/dev_call.html) or ping me on PX4 slack.

Great! I think I wold star to read some PX4 terms and participate in the weekly call first, when more specific questions appears will be useful that call.

Linus-Goh, 2024-03-09:
Hi, how has the addition of Webots support into PX4 autopilot been?
Recently I've started exploring Webots as an alternative to Gazebo. Having experimented on Gazebo for over a year, Webots seems like a very reliable alternative to Gazebo. I too am interesting in the possibilities of addding Webots support into PX4. In addition, currently im a Undergrad student would seek for opportunities to explore into this area. Thanks!

green-creeper, 2025-06-08:
yeah, I'll play around with it as well tomorrow. It took me a while to make it run on macOS (using either docker or raspberry pi)
I also see there's with hil_state_quterion and wonder if it's mandatory, I don't see it complaining for not having this information and introducing it now means introducing more points of failure.

TinkerYxs, 2025-07-06:
[image]

closed 2025-11-30 by github-actions[bot]