PX4/PX4-Autopilot #18224: VTOL Uncommanded Yaw in Transition

matthtan, 2021-09-11:
Hello,

During a VTOL transition, there is a consistent, uncommanded yaw before the aircraft starts moving.  See this log [here](https://review.px4.io/plot_app?log=51d80790-635e-42df-9f5a-2500002ee080) at the very beginning of the transition around 12:14 to 12:16.  After the yaw (ending at 12:16) the aircraft continues on normally.  Weather vane is disabled.

Thank you!

ThomasRigi, 2021-09-15:
For me this looks like a physical problem on your drone. There is no yaw rate setpoint that would indicate that it's PX4 that wants to yaw. Also, the actuator control looks consistent: the yaw rate setpoint is roughly 0, the yaw rate is negative, so it outputs a positive control output trying to get the negative rate back to the setpoint.
[image]
[image]

matthtan, 2021-09-17:
Thanks for the response!

We ended up switching the propeller directions on the front two motors (tricopter configuration, so swapping the position of the counter-rotating props) and the yaw went away, so it definitely seems like a physical change.

Thanks!

closed 2021-09-17 by matthtan