PX4/PX4-Autopilot #18225: RC input in mission mode

moreba1, 2021-09-12:
Hi

Is it possible that we have input RC during a mission? for example for increase altitude with elevator in mission mode. currently with COM_RC_OVERRIDE enabled with stick movement, mode changes to Position. but I will input RC command during mission mixing with autopilot commands.

julianoes, 2021-09-14:
It's not currently possible. Can you describe the context/use case why/how you would want that? And would it only be altitude or more than that? Is that for fixedwing or multicopter?

moreba1, 2021-09-14:
I will use for fixedwing and VTOL in fixedwing mode. 

1- Sometimes we need to increase altitude or change position to avoid collision with obstacles. Switching mode to position, doing changes and comeback to mission mode is harder and complicates the situation.

2- During simulation (SITL) we can input command and disturbances by RC to see what happen. For example get out of mission track and analyzing behavior. 

currently ardupilot code have this feature on plane and multirotor.

vacoff, 2023-08-28:
Yeah that would be a great feature.