PX4/PX4-Autopilot #18227: EKF2 do not work when SYS_HAS_BARO = 0

aet2230, 2021-09-12:
Tried v1.12.3. On our application we use VIO only, and do not use magnetometer and barometer.
Tried do disable barometer and magnetometer chacks via SYS_HAS_* parameters (to avoid unwanted checks or other issues if unused HW is faulty)
After setting SYS_HAS_MAG = 0, works ok.
When SYS_HAS_BARO = 0, EKF2 does not start (or does not initialize). With SYS_HAS_BARO = 1, works OK, even if not used.

dagar, 2021-09-13:
Currently baro data is required to initialize ekf2. 

Previously I started refactoring things to only require the data for the specific height mode, but that's stalled. https://github.com/PX4/PX4-ECL/pull/996

If you're able to help test and review we could resurrect the effort.

lukegluke, 2021-09-22:
Temporary you could use this workaround for example that use gps height on initialization step if it fits you:
https://github.com/PX4/PX4-ECL/issues/876#issuecomment-901757028

bresch, 2022-10-26:
This works now

closed 2022-10-26 by bresch