PX4/PX4-Autopilot #18237: BMM150 on ARK GPS not starting on master

AlexKlimaj, 2021-09-14:
The BMM150 update to use the new I2CSPI driver looks like it broke the BMM150 starting on ARK GPS. It works on 1.12.3 but not on master. I have seen this issue before on the BMM150 where it will hang the I2C bus.

master
```
NuttShell (NSH) NuttX-10.0.0
nsh> ERROR [PX4_MTD] failed to locate spi bus
ERROR [PX4_MTD] mtd failure: -6 bus 2 address 0 class 1
nsh: sercon: command not found
nsh: ver: command not found
nsh: mft: command not found
nsh: mft: command not found
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
ERROR [param] Parameter EKF2_MULTI_IMU not found.
ERROR [param] Parameter SENS_IMU_MODE not found.
ERROR [param] Parameter EKF2_MULTI_MAG not found.
ERROR [param] Parameter SENS_MAG_MODE not found.
Board defaults: /etc/init.d/rc.board_defaults
nsh: rgbled: command not found
nsh: rgbled_ncp5623c: command not found
Board sensors: /etc/init.d/rc.board_sensors
ERROR [parameters] get: param 65535 invalid
ERROR [parameters] get: param 65535 invalid
icm42688p #0 on SPI bus 1
bmp388 #0 on I2C bus 2 address 0x76
WARN  [SPI_I2C] bmm150: no instance started (no device on bus?)
ERROR [param] Parameter SENS_EN_PX4FLOW not found
INFO  [uavcannode] Node ID 123, bitrate 1000000
```

1.12.3
```
NuttShell (NSH) NuttX-10.0.0
nsh> ERROR [PX4_MTD] failed to locate spi bus
ERROR [PX4_MTD] mtd failure: -6 bus 2 address 0 class 1
nsh: sercon: command not found
nsh: ver: command not found
nsh: mft: command not found
nsh: mft: command not found
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
ERROR [param] Parameter EKF2_MULTI_IMU not found.
ERROR [param] Parameter SENS_IMU_MODE not found.
ERROR [param] Parameter EKF2_MULTI_MAG not found.
ERROR [param] Parameter SENS_MAG_MODE not found.
Board defaults: /etc/init.d/rc.board_defaults
nsh: rgbled: command not found
nsh: rgbled_ncp5623c: command not found
Board sensors: /etc/init.d/rc.board_sensors
ERROR [parameters] get: param 65535 invalid
ERROR [parameters] get: param 65535 invalid
icm42688p #0 on SPI bus 1
bmp388 #0 on I2C bus 2 address 0x76
bmm150 #0 on I2C bus 1
ERROR [param] Parameter SENS_EN_PX4FLOW not found
INFO  [uavcannode] Node ID 123, bitrate 1000000
```

jlaitine, 2021-09-16:
I have also hit this problem of bmm150 not starting properly every time; I fixed it by issuing a soft-reset to the chip in that case. Please ping me if this is of interest, I can present my solution as well.

AlexKlimaj, 2021-09-30:
Fixed with  #18258

closed 2021-09-30 by AlexKlimaj