PX4/PX4-Autopilot #18239: MAV_CMD_NAV_LOITER_TO_ALT with 0 lon/lat coordinates not correctly handled

Hylke-Atmos, 2021-09-15:
**Describe the bug**
According to the MavLink specs a [MAV_CMD_NAV_LOITER_TO_ALT](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_TO_ALT) mission item without lon/lat coordinates (or at least set to 0) should result in a loiter at the current position (no specific location).

Instead Px4 handles the lon/lat 0 values as actual coordinates, resulting in a loiter point in the sea (Lon 0.0 Lat 0.0).

**To Reproduce**
This can very easily reproduced with a SITL device.
Steps to reproduce the behavior:
1. Load the attached QGC mission file or create a mission containing a loiter with Lon/Lat coordinates set to 0.
2. Uploaded mission
3. An error occurs "Distance between waypoints too far: <X>m"! and it won't fly.

**Expected behavior**
A loiter point without Lon Lat coordinates (set to 0) should result in a valid mission plan. According to the mavlink specs a Loiter without coordinates should result in the drone loitering to the designated altitude at the current position (wherever that might be) before continuing.

P.S. QGC is also failing to process this in the correct way..

**Example mission plan**
[Loiter_current_position.zip](https://github.com/PX4/PX4-Autopilot/files/7168457/Loiter_current_position.zip)

**Additional context**
As also can be noticed in the example mission. What I'm trying to achieve is, whenever a Return to Launch (RTL ) is initiated, it would first (from any location) loiter to a target altitude before returning to launch (RTL). I know I can (slightly) manipulate the RTL behavior by using a MAV_CMD_DO_LAND_START marker (and a Mission based RTL).

I know (now) this can also (and probably better) be achieved with the [RTL_RETURN_ALT](http://docs.px4.io/master/en/flight_modes/return.html#RTL_RETURN_ALT) parameter, but nevertheless it is not consistent with the MavLink documentation.

Hylke-Atmos, 2021-09-15:
@hamishwillee and @julianoes As discussed on Slack

hamishwillee, 2021-09-16:
Thanks so much. Hopefully now someone has the bandwidth and interest to fix it.

julianoes, 2026-08-31:
FYI @hamishwillee in case you want to burn some tokens on it.