PX4/PX4-Autopilot #18241: Raspberry Pi not starting when connected with PixHawk

MuhammadBilal1, 2021-09-15:
Hi,

**RPi =** Raspberry Pi 4 Model B
**PixHawk =** PixHawk 4
**Firmware =** PX4 - 1.11.3
**Baudrate =** 921600
**PixHawk Port =** Telemetry 2
**RPi Port =** GPIO14, GPIO15
**RPi OS =** Ubuntu 20.04.04

**Issue:**
I am getting an issue while trying to start the RPi while connected with the PixHawk.
Some invalid characters starts coming up on the screen. Screenshot is attached.

**What I tried?**
1- If I disconnect RPi from PixHawk & start both separately, then after both things bootup everything works fine. I can use mavros/mavlink to arm/disarm the drone (Heartbeat is coming ok). 
2- I tried disconnecting all the peripherals except the telemetry/serial wire. Not worked.

**RPi connection with PixHawk**
Raspberry Pi is connected with PixHawk via serial connection (Telemetry 2 @ 921600).

**Power to RPi**
RPi is powered ON by connecting the power adapter (Micro USB) to the power port.

**The following peripheral are connected with RPi:**
* LCD at HDMI port
* Keyboard
* Mouse
* Ethernet Cable
* Micro USB Cable (Power)

[image]

[image]

[image]

JStoop97, 2021-09-15:
I have had a similar issue before with the pi with a faulty keyboard plugged in. The keyboard would spam information onto the pi USB/uart during start up and the pi would not boot properly. I think this therefore an issue with the pi and you would be better off directing your search there.

MuhammadBilal1, 2021-09-15:
> think this therefore an issue with the pi and you would be better off directing your search there.

The same happened even after removing the Keyboard, Mouse & Ethernet cable.

JStoop97, 2021-09-15:
No what I mean is that sometimes when the uart port is spammed during start up by a device such as a pixhawk or faulty keyboard (as was my case) this could happen. Perhaps try a different baudrate and see if that doesn't solve the error.

MuhammadBilal1, 2021-09-23:
Updating the post so if anyone else is getting the same issue, they can get a fix.
I changed the wiring from GPIO ports to USB and things are now working fine.

I was getting the issue while connecting the PixHawk and Raspberry Pi via GPIO ports.
The working solution is to use the **FTDI cable** to connect the PixHawk and Raspberry Pi.

[image]

closed 2021-09-23 by MuhammadBilal1