PX4/PX4-Autopilot #18254: Getting jinja invalid syntax error on building the px4 catkin workspace

Akashleena, 2021-09-17:
Errors     << px4:make /home/aleena/catkin_ws/logs/px4/build.make.023.log      
  File "libevents/scripts/generate.py", line 112
    def jinja_cpp_namespace(arg_type: str):
                                    ^
SyntaxError: invalid syntax
make[2]: *** [events/events_generated.h] Error 1
make[1]: *** [src/lib/events/CMakeFiles/events_header.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/aleena/catkin_ws/build/px4; catkin build --get-env px4 | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

I have installed Jinja2 3.0.1 and the output of python --version is
Python 3.6.9

Jaeyoung-Lim, 2021-09-17:
@Akashleena It won't work if you put the PX4 repository in the `catkin_ws`. Please follow the documentation http://docs.px4.io/master/en/simulation/ros_interface.html

Jaeyoung-Lim, 2021-10-03:
@Akashleena I am closing this as we didn't get any response. If you think this is still an issue, please reopen the issue with an update.

closed 2021-10-03 by Jaeyoung-Lim