PX4/PX4-Autopilot #18256: micrortps_bridge not publishing specific topics

b0hne, 2021-09-17:
Hi,
I am running the agent and client on a nanopi/armbian device and a pixawk 4(px4_fmu-v5_rtps).
I can subscribe to  "/fmu/sensor_combined/out" and get data, but "/fmu/vehicle_local_position/out" stays silent. I can get a readout via the mavlink_shell. Any idea what the problem is?

TSC21, 2021-09-21:
`vehicle_local_position` is not streamed by default on the bridge. So that topic is not even created in the ROS side. You would have to add the topic on `urtps_bridge_topics.yaml` both on this repo and in px4_ros_com and rebuild both. Make sure to make clean and delete the build and install folders in the colcon ws first before building.

closed 2021-09-22 by b0hne

JohnnyLiu2886, 2021-09-30:
I met the same problem. "micrortps_agent -d /dev/ttyTHS1" did not add new publisher "VehicleGpsPosition" even I add it in /PX4-Autopilot/msg/tools/urtps_bridge_topics.yaml and updating /px4_ros_com/templates/urtps_bridge_topics.yaml using "uorb_to_ros_urtps_topics.py".

I have run colcon build in my workspace after the update. 

The new message "VehicleGpsPosition" is in the same format as "SensorCombined". Both of them are "send: true"

TSC21, 2021-09-30:
> I met the same problem. "micrortps_agent -d /dev/ttyTHS1" did not add new publisher "VehicleGpsPosition" even I add it in /PX4-Autopilot/msg/tools/urtps_bridge_topics.yaml and updating [...]

Did you use the `clean_all.bash` script inside px4_ros_com/scripts? Also did you make clean before rebuilding in the PX4-Autopilot dir?

JohnnyLiu2886, 2021-10-04:
No, I did not.

However, after following your guide to clean the workspace and the PX4-Autopilot dir, the publishers updated successfully. 
Thank you so much for your help!