PX4/PX4-Autopilot #18257: better height estimation for multi-vehicle parameters for better height estimation barometer,lidar, gps

airsimdevdd, 2021-09-17:


airsimdevdd, 2021-09-17:
@Paul Riseborough please help me

garlinplus, 2021-09-18:
Default parameter is enough for your need.

dagar, 2021-09-18:
@airsimdevdd could you please add more context? What problems are you seeing specifically? Can you share any flight logs showing the issue? https://logs.px4.io

airsimdevdd, 2021-09-20:
> Default parameter is enough for your need.
all my 4  vehicles are not maintaining same altitude wat to do

airsimdevdd, 2021-09-20:
> @airsimdevdd could you please add more context? What problems are you seeing specifically? Can you share any flight logs showing the issue? https://logs.px4.io

sir  when i fly my vehicles in offboard for example i gave 10m  altitude some vehicles go to correct height but some go to 15m 13m altitudes

mrpollo, 2021-10-07:
I'm going to close this issue since we don't have any information, next time you run into an issue grab the logs so we can help you.

closed 2021-10-07 by mrpollo