PX4/PX4-Autopilot #18269: Problem with GPS port on Pixhawk 4

FeCLN, 2021-09-20:
Hello, I am from Skyrats, a Brazilian autonomous drone team. Recently we bought a Pixhawk 4 with the intention of using it with some other components in the GPS port, along with the GPS of course. These components are: a normal safety switch, a buzzer, and a SE100 GPS (All of them tested in a Pixhawk 1 and working normally). After making some wiring connections; all of them accordingly to each datasheet and pinout, we tried to use it in a Pixhawk 4, but unfortunately we didn't have good results. The safety switch didn't work, the LED didn't light up, and checking with a voltmeter we didn't even get any signal from these ports. The buzzer makes a sound, but it's completely inaudible. And the worst part is, the GPS didn't work, in QGroundControl we receive only a "0 satellites found with 0 signal strength". Has anyone had this problem? What can we do?

dagar, 2021-09-21:
Does anything at all work with that pixhawk 4? For example on the console can you run `gps status` (through QGC or serial console)? If literally nothing is working (switch, LED, GPS) it points to the wiring or the unit being bad.

On the pixhawk 4 I've mainly used the Holybro GPS/compass/safety/buzzer module with the 10 pin connector. http://www.holybro.com/product/pixhawk-4-gps-module/

FeCLN, 2021-09-21:
The `gps status` returns nothing, the driver is auto-starting, but no info from gps is returned, only some "0". And i've used in other occasion the Here 2 GPS, ant it was working fine

ryanjAA, 2021-09-25:
The pixhawk 4 gps port is essentially pinned out to with with the holybro gps. A few things are shared from the port that if you breakout (button and speaker) you need to splice or tap elsewhere. 

The speaker is not a simple piezo like the Pixhawk 1. It’s its own circuit and the button will need to grab power from elsewhere unless on the same pcb iirc. We have them all split out and ran into this issue and ended up just making pcbs for each component that tapped the necessary power and ground back from the pix 4. 

Take a look at the p4 pinout and you can work it out quite quickly, you’ll just need to steal 3.3v for the switch (iirc) and the speaker is a complete no go unless you make a custom one. 

As for the gps not working, that is strange. Could it be something simple “&@3 the ground wire or rx/tx are reversed if you are making your own plug?

julianoes, 2024-12-06:
No more follow up, closing.

closed 2024-12-06 by julianoes