PX4/PX4-Autopilot #18273: Can I use microRTPS to communicate with crazyflie2.1 ?

chaos4ros2, 2021-09-22:
Hi, I want to control crazyflie2.1 by PX4 within microRTPS for ROS2 app but I can't find a 
RTPS version in PX4-Autopilot.(only [bitcraze_crazyflie21_default](https://github.com/PX4/PX4-Autopilot/blob/master/Makefile#L267), no [_rtps](https://dev.px4.io/v1.11_noredirect/en/middleware/micrortps.html#client_firmware) feature)

Is thre any way to let crazyflie2.1 to operate as a microRTPS client?

benjinne, 2021-09-28:
You need to add the rtps makefile. I did it for another board a while ago. See this PR https://github.com/PX4/PX4-Autopilot/pull/17338

chaos4ros2, 2021-09-29:
Thank you for providing the solution! I am going to try it and If it works fine, I'll create a PR.