PX4/PX4-Autopilot #18285: first px4 firmaware build

jps31, 2021-09-23:
Hello 
I installed the dev environment for PX4 (W10 PC) and succeeded in : 
_make px4_sitl jmavsim_
Then I tried 
_make px4_fmu-v5_
and got an error 
**FAILED: px4_io-v2_default.elf
Memory region         Used Size  Region Size  %age Used
           flash:       61896 B        60 KB    100.74%
            sram:        3728 B         8 KB     45.51c:/px4/toolchain/gcc-arm/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: px4_io-v2_default.elf section `.data' will not fit in region `flash'
c:/px4/toolchain/gcc-arm/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: region `flash' overflowed by 456 bytes**
Can somebody help please ?
BR

jps31, 2021-09-23:
Is there a parameter to set the flash memory size ?

taileron, 2021-09-23:
GCC 9.3.1 is the recommended compiler. 
7.3.1 is not yet able to optimise memory well and could generate code that does not fit into the flash space.

jps31, 2021-09-24:
Ok Thank you i will update
BR

closed 2021-09-24 by jps31