PX4/PX4-Autopilot #18297: Takeoff close to Geofence causes drone to fly away uncontrollably

rizvanovic, 2021-09-25:
**Describe the bug**
Drone flies perpendicular to geofence on takeoff close to geofence. The drone places a setpoint ~7 m away from launch point and tries to reach it aggressively.  This is very dangerous behavior and has caused a crash during a crucial tech demo we conducted earlier this week. 

Today we went out with a cheaper drone to try to trigger the same behavior and managed to do so immediately. We did two flights to ensure repeatability.

**To Reproduce**
Steps to reproduce the behavior:
1. Place drone close to but not outside of geofence border.
2. Enter takeoff mode. 

**Expected behavior**
Takeoff to set altitude.

**Log Files and Screenshots**
[First flight](https://review.px4.io/plot_app?log=3d188929-9d67-4ff6-8a6f-eff40df8d200)
[Second flight](https://review.px4.io/plot_app?log=5f430551-e48f-4486-ae38-35db4489b12d)

**Drone (please complete the following information):**
- Holybro S500
- Pixhawk 4 FC
- CUAV H16 radio module

rizvanovic, 2021-10-03:
This is potentially very dangerous behavior and can cause damage to property and harm if not solved.

julianoes, 2021-10-04:
Thanks for reporting this.

ThomasRigi, 2021-10-15:
I think it's GF_ACTION = 2 (Hold) that messes you up. Probably something in here: https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/navigator/GeofenceBreachAvoidance/geofence_breach_avoidance.cpp which is called from https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/navigator/navigator_main.cpp#L738

RomanBapst, 2021-11-22:
@Katawann @ThomasRigi @dagar @rizvanovic I had a look at this and I can confirm that the main issue is that the autopilot tries to get away from the boundary of the geofence right after takeoff, which is basically very undesired.

There are several options how we can improve this. There is a feature implemented which pushes the drone away from the geofence once it has gotten too close. The reason is was implemented is because otherwise a user who tried continuously to fly again the fence would get closer and closer and eventually he would breach it. Pushing the vehicle away from the fence reduces the chances but also comes with the effect of unpredictable behavior. 

1) Disable the "pushing away" feature by default
2) Disable it if closer than a certain distance above ground
3) Disable it if closer to home than a certain distance.

I'm open for everything, happy to get some feedback.

rizvanovic, 2021-11-28:
@RomanBapst Absolutely think the "push away" feature should be disabled if closer than a certain distance from home during takeoff and landing specifically. Then again why even allow for takeoff if the geofence is considered too close. Maybe a simple message saying "Too close to geofence for take off: _X m_" would work as it would force the operator to find a more suitable take off location which is usually possible.

RomanBapst, 2021-12-09:
@rizvanovic @dagar Please see my proposal for preventing this.
https://github.com/PX4/PX4-Autopilot/pull/18821

NAmmann, 2022-07-07:
I think this post in the PX4 forum is also related to this: https://discuss.px4.io/t/geofence-altitude-premature-trigger-and-overshoot-loop/26180

NAmmann, 2022-07-07:
In addition, i observed a similar behaviour:
The UA acended and approached the vertical geofence in manual mode. The position mode (jerk based) had only low acceleration and jerk, and therefore the predicted breaking distance was computed very long. So the geofence failsafe (hold) was triggered premature. And the setpoints on the braking trajectory let the drone ascend until breaching the virtual geofence.

So the geofence avoidance should be disabled by parameter and only a check if outside the geofence should be used then. This leads to a much clearer behaviour!
Or just set a velocity setpoint without setting the position to stop the uav.