PX4/PX4-Autopilot #18299: Posix Sitl vehicles don't work correctly

eldewany, 2021-09-27:
**Describe the bug**
Hello, whenever i use a launch file that is based on posix_sitl.launch to start a SITL instance an uncalibrated sensors error occur in QGC although the normal make px3_sitl gazebo command launches SITL and everything works fine.

**To Reproduce**
Steps to reproduce the behavior:
1- Source everything needed (PX4, build path and everything)
2- roslaunch px4 mavros_posix_sitl.launch

**Expected behavior**
Everything should work fine but it isn't, and I verified it's not a mavros problem

**Additional context**
I Tried to replicate the same method to spawn the sdf file as in spawn_single_vehicle_sdf.launch but something about the xmlstartlet command didn't work

Jaeyoung-Lim, 2021-09-30:
Unfortunately, I cannot replicate your problem (see the log below)

Could you share exactly which setup (ROS version, system setup) and what exactly you are doing?

```
$ roslaunch px4 mavros_posix_sitl.launch... logging to /home/jaeyoung/.ros/log/0ed483d0-2222-11ec-8325-a4bb6d5c3360/roslaunch-jaeyoung-Precision-3640-Tower-21090.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jaeyoung-Precision-3640-Tower:35533/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
[... 230 lines cut ...]
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2021-09-30/19_10_23.ulg	
INFO  [logger] Opened full log file: ./log/2021-09-30/19_10_23.ulg
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
[ INFO] [1633029024.915576115, 1.528000000]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1633029024.916472384, 1.528000000]: IMU: High resolution IMU detected!
[ INFO] [1633029025.926228219, 2.536000000]: GF: Using MISSION_ITEM_INT
[ INFO] [1633029025.926258023, 2.536000000]: RP: Using MISSION_ITEM_INT
[ INFO] [1633029025.926272803, 2.536000000]: WP: Using MISSION_ITEM_INT
[ INFO] [1633029025.926288402, 2.536000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1633029025.926301576, 2.536000000]: VER: 1.1: Flight software:     010d0040 (86da9893e8000000)
[ INFO] [1633029025.926310491, 2.536000000]: VER: 1.1: Middleware software: 010d0040 (86da9893e8000000)
[ INFO] [1633029025.926321248, 2.536000000]: VER: 1.1: OS software:         050400ff (ca1938fbcccc1a0a)
[ INFO] [1633029025.926331061, 2.536000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1633029025.926343427, 2.536000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1633029025.926353879, 2.536000000]: VER: 1.1: UID:                 4954414c44494e4f
[ INFO] [1633029028.467661909, 5.076000000]: IMU: Attitude quaternion IMU detected!
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative
[ INFO] [1633029039.934814344, 16.528000000]: RP: mission received
[ INFO] [1633029039.934865972, 16.528000000]: WP: mission received
[ INFO] [1633029039.934901095, 16.528000000]: GF: mission received

```

eldewany, 2021-10-03:
@Jaeyoung-Lim sorry for the delay, here's my log

`eldewany@eldewany-Alienware-17-R3:~$ ./launch-2.sh 
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/eldewany/PX4-Autopilot/...y/PX4-Autopilot/build/px4_sitl_default
Not running simulation (DONT_RUN is set).
GAZEBO_PLUGIN_PATH :/home/eldewany/Desktop/PX4-Autopilot/build/px4_sitl_default/build_gazebo:/home/eldewany/Desktop/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH /home/eldewany/Desktop/PX4-Autopilot/Tools/sitl_gazebo/models::/home/eldewany/Desktop/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/eldewany/catkin_ws2/devel/lib:/home/eldewany/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/eldewany/Desktop/PX4-Autopilot/build/px4_sitl_default/build_gazebo
... logging to /home/eldewany/.ros/log/ed5626aa-245b-11ec-be8e-9cb6d00dd7bf/roslaunch-eldewany-Alienware-17-R3-20435.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://eldewany-Alienware-17-R3:34211/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
[... 312 lines cut ...]
[ INFO] [1633273783.084674952, 2.552000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1633273783.084784643, 2.552000000]: VER: 1.1: UID:                 4954414c44494e4f
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify positive
pxh> commander trINFO  [mavlink] Mission download request ignored, already active	
[ INFO] [1633273797.103699627, 16.536000000]: RP: mission received
[ INFO] [1633273797.105367652, 16.540000000]: WP: item #0  F: C:178 p: 1 5 -1 0 x: 0 y: 0 z: 0
[ INFO] [1633273797.106237286, 16.540000000]: WP: item #1  F: C: 22 p: 0 0 0 nan x: 473977419 y: 85455945 z: 3
[ INFO] [1633273797.106930277, 16.540000000]: WP: item #2  F: C: 16 p: 0 0 0 nan x: 473977530 y: 85467099 z: 3
[ INFO] [1633273797.107592057, 16.540000000]: WP: item #3  F: C:178 p: 1 5 -1 0 x: 0 y: 0 z: 0
[ INFO] [1633273797.108672476, 16.544000000]: WP: item #4* F: C: 20 p: 0 0 0 0 x: 0 y: 0 z: 0
[ INFO] [1633273797.108888121, 16.544000000]: WP: mission received
pxh> commander ta[ WARN] [1633273798.103535036, 17.536000000]: GF: timeout, retries left 2
[ INFO] [1633273798.106601411, 17.536000000]: GF: mission received
pxh> commander takeoff
pxh> WARN  [PreFlightCheck] Preflight Fail: Compass 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 2 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro 1 uncalibrated
[ WARN] [1633273799.392594654, 18.824000000]: CMD: Unexpected command 22, result 0
WARN  [PreFlightCheck] Preflight Fail: Gyro 2 uncalibrated
INFO  [tone_alarm] notify negative
`

Jaeyoung-Lim, 2021-10-03:
@eldewany It doesn't seem like you are using the default launch file: `./launch-2.sh `

eldewany, 2021-10-03:
@Jaeyoung-Lim it's just a launch file that has the source lines

#!/bin/bash
```

cd ~/Desktop/PX4-Autopilot/
DONT_RUN=1 make px4_sitl gazebo
source ~/catkin_ws/devel/setup.bash    # (optional)
source ~/catkin_ws2/devel/setup.bash
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 mavros_posix_sitl.launch

```

Jaeyoung-Lim, 2021-10-03:
@eldewany Since I do not have the launchfile you are using, and given that the default launch file works in my environment I am not sure how to help you.

Can you try with the default launchfile and still see if there is a problem?
Also please clarify what environment you are using

eldewany, 2021-10-03:
@Jaeyoung-Lim I'm indeed using the default launch file, the file `./launch-2.launch` only contains the lines that source the PX4 folder and then launches the default command, again this is the contents of `./launch-2.launch` 

```
cd ~/Desktop/PX4-Autopilot/
DONT_RUN=1 make px4_sitl gazebo
source ~/catkin_ws/devel/setup.bash    # (optional)
source ~/catkin_ws2/devel/setup.bash
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 mavros_posix_sitl.launch
```

I'm running an Ubuntu 18.04 , ROS Melodic and a fresh PX4 repo that i cloned just now. I would also like to note that the issue doesn't appear unless you command the vehicle to takeoff or if you open QGC, it always shows Gyro not calibrated

[image]

Jaeyoung-Lim, 2021-10-03:
@eldewany Okay, but could you confirm what the results are with the default launch file? 

Again, the launchfile works fine on my system as well as the [MAVROS CI](https://github.com/PX4/PX4-Autopilot/actions/workflows/mavros_mission_tests.yml) tests that check this automatically.

Therefore the only difference is your launch file, and with the default launch file the results should be the same since I am also testing with ROS melodic Ubuntu 18.04. If this is not the case we can start looking at the logs from there on.

eldewany, 2021-10-03:
@Jaeyoung-Lim It's a typo mistake, i apologize.. i meant `./launch-2.sh` , it's a shell script that sources the PX4 requirements.

Jaeyoung-Lim, 2021-10-03:
@eldewany Ah, right now it makes more sense.

Could you also try removing the second workspace? `source ~/catkin_ws2/devel/setup.bash`

I am still convinced that the problem is not from this repo and something from the script you have created.

Jaeyoung-Lim, 2021-10-08:
@eldewany Glad to hear that you found a way to fix the problem!

AlexWUrobot, 2023-02-07:
Thanks a lot. it is helpful to clean the `~/.ros`

bb6698, 2024-02-03:
The same thing happens to me too, it doesn't work to clean the directory ~/.ros . How can I disable this action? I need to do reinforcement learning and with "battery unhealthy" I can't continue with the training.

The parameters SIM_BAT_MIN_PCT are set 50.0  and SIM_BAT_DRAIN are set 86400