PX4/PX4-Autopilot #18306: Offboard trajectory_setpoint conflict when in auto modes

benjinne, 2021-09-28:
**Describe the bug**
Offboard trajectory_setpoint sent while in an auto mode conflict with flight_tasks sending trajectory_setpoint since they both publish to the same topic. It appears everything is working correctly when sending trajectory_setpoints at the 10hz recommended by the offboard tutorial and in an auto mode, however this is because flight_tasks publishes at 50hz and is fast enough to essentially "override" the setpoints sent by the 10hz offboard publisher. When publishing at 100hz to trajectory_setpoint from offboard, the auto mode is essentially ignored and the offboard setpoint overrides the internal flight_tasks module. 

**To Reproduce**
Steps to reproduce the behavior:
1. Launch simulator and set PX4 to HOLD mode
2. Send only trajectory_setpoints using offboard at 10hz and nothing happens
3. Send only trajectory_setpoints using offboard at 100hz and observe HOLD mode is overriden

**Expected behavior**
There needs a way for trajectory_setpoint to be distinguished if it's coming from offboard or flight_tasks. When in offboard mode, flight_tasks no longer send trajectory_setpoints, but if in any auto mode, a conflict is possible if offboard setpoints are sent faster than the flight_task

dagar, 2021-09-28:
This type of potential conflict needs to be prevented more fundamentally.

Are you using Mavlink or ROS2/FastRTPS?

In the mavlink case this should already be prevented until the system is actually in OFFBOARD mode. 

https://github.com/PX4/PX4-Autopilot/blob/03bd5e0f4841debf5e45d0eadbfb28f131eaaeb1/src/modules/mavlink/mavlink_receiver.cpp#L1004-L1008

benjinne, 2021-09-28:
I'm using ROS2/FastRTPS

dagar, 2021-09-29:
> I'm working on a fix, and got multicopter position controller working, but for the rover, why is there 2 updates for trajectory_setpoint_sub?
> 
> https://github.com/PX4/PX4-Autopilot/blob/360c3781f348b2bf7eb9df424a4ea18f8aed82c1/src/modules/rover_pos_control/RoverPositionControl.cpp#L431
> 
> 
> and
> https://github.com/PX4/PX4-Autopilot/blob/360c3781f348b2bf7eb9df424a4ea18f8aed82c1/src/modules/rover_pos_control/RoverPositionControl.cpp#L478

I don't think it's intentional, it's probably a result of how things evolved to this point with OFFBOARD setpoints previously getting to the controllers through a different topic.

benjinne, 2022-12-07:
Closing since mavlink handles this issue and using ROS2 is experimental and assumes the offboard computer must stop sending setpoints if a pilot takes over.

closed 2022-12-07 by benjinne