PX4/PX4-Autopilot #18309: issues running command 'make px4_sitl jmavsim'. Keep getting 'make: *** [px4_sitl] Error 1' on Macbook Pro

SamuelSHernandez, 2021-09-28:
When running the command above, I get conflicts between the User site and the virtual environment. The command throws the following error:

ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.

I was following the [setup guide for PX4 on mac](https://www.youtube.com/watch?v=tMbMGiMs1cQ&t=1s&ab_channel=PX4Autopilot-OpenSourceFlightControl.) I used these commands on the terminal:

```
brew tap PX4/px4
brew install px4-dev
pip3 install --user pyserial empy toml numpy pandas jinja2 pyyaml pyros-genmsg packaging
brew tap AdoptOpenJDK/openjdk
brew --casks install adoptopenjdk14 
```
brew --casks install adoptopenjdk14 (original 'brew cask install adoptopenjdk14' command is not supported by brew anymore)
```
brew install px4-sim-jmavsim
cd YourDirectory
git clone https://github.com/PX4/Firmware.git --recursive
cd Firmware
make px4_sitl jmavsim
```

The simulator is supposed to start with 'make px4_sitl jmavsim', instead I get the errors above.

Here are some screenshots of the problem.
[image]

[image]

**Additional context**
Problem with running simulator

taileron, 2021-09-29:
there is only a narrow range of Mac Os versions where this works. Not too old and not too new ... it's proven that it definitely won't run on 10.12 or older and I don't think it will run on M1 yet either.

julianoes, 2021-09-30:
You need to figure out your python dependencies and where pip packages are installed. It's a bit tricky on macos especially if you have brew Python and system Python and maybe a virtualenv, etc.

SamuelSHernandez, 2021-09-30:
After working on this issue a bit, installing the python dependencies and using 'make clean', 'make px4_sitl jmavsim' until it didn't give me issues worked for me, but I'd recommend switching to a Linux environment if possible. Thanks for the help! I'll close this issue

closed 2021-09-30 by SamuelSHernandez