PX4/PX4-Autopilot #18310: Multiple iris_obs_avoid drones fails to initialize properly

KrisRadowski, 2021-09-29:
**Describe the bug**

When trying to simulate multiple iris_obs_avoid in Gazebo (using multi_uav_mavros_sitl_sdf.launch file), only the first one is initialized properly; others are spawning but is not possible to connect to them 

**To Reproduce**
1. launch PX4
`roslaunch px4 multi_uav_mavros_sitl_sdf.launch vehicle:=iris_obs_avoid`
2. the drones didn't initialize properly; instead one of px4 processes hogs the computer

**Expected behavior**
there should be possible to connect to all of them

**Drone (please complete the following information):**
Gazebo Simulated iris_obs_avoid (on Ubuntu 18.04 LTS)

**Additional context**
I'm aware that iris_obs_avoid is mostly used with internal avoidance algorithms; I turned them off beforehand

Jaeyoung-Lim, 2021-09-30:
@KrisRadowski `iris_obs_avoid` is not supported for multiple vehicle simulation

KrisRadowski, 2021-10-02:
> @KrisRadowski `iris_obs_avoid` is not supported for multiple vehicle simulation

to be fair, when I attached only one sonar to Iris the same behavior happened (I can only connect to the first drone spawned)

Jaeyoung-Lim, 2021-10-02:
@KrisRadowski Does this also happen to the default iris models? (with no modifications)

KrisRadowski, 2021-10-02:
> @KrisRadowski Does this also happen to the default iris models? (with no modifications)

@Jaeyoung-Lim no, they're visible and steerable (at least in QGC)

Jaeyoung-Lim, 2021-10-02:
> @Jaeyoung-Lim to be fair, when I attached only one sonar to Iris the same behavior happened (I can only connect to the first drone spawned)

@KrisRadowski Okay, that is good to know, this means that the way you were adding the sonar sensor might be the problem - and not a problem of the multi vehicle script itself.

Could you elaborate on how you intended to add the sonar sensor to the vehicles?

KrisRadowski, 2021-10-02:
> Could you elaborate on how you intended to add the sonar sensor to the vehicles?

@Jaeyoung-Lim I'm not sure what did you mean, but i made a modified model (iris with included sonar and joint linking drone and sonar) and on attached
 screenshot is shown that in Gazebo both drones are visible, but in QGC - only one
[image]

also, when I'm launching only one modified drone it's working like charm

Jaeyoung-Lim, 2021-10-02:
> Could you elaborate on how you intended to add the sonar sensor to the vehicles?
@Jaeyoung-Lim I'm not sure what did you mean, but i made a modified model (iris with included sonar and joint linking drone and sonar) and on attached
screenshot is shown that in Gazebo both drones are visible, but in QGC - only one

@KrisRadowski Yes, but which file are you modifying? It means something broke when generating the other models by your modification, and therefore you can't connect

KrisRadowski, 2021-10-02:
@Jaeyoung-Lim 
> Yes, but which files are you modifying and how?

- I modified CMakeLists.txt in `ROMFS/px4fmu_common/init.d-posix/airframes` catalog including my ROMFS file (for drone setup)
- modified sitl_target.cmake in `platforms/posix/cmake' catalog (so I can call my drone as build target)

I'm attaching these files, my drone files from `Tools/sitl_gazebo/models` and script used for launching drones in ZIP archive below

[issue.zip](https://github.com/PX4/PX4-Autopilot/files/7271752/issue.zip)

Jaeyoung-Lim, 2021-10-02:
@KrisRadowski Now I see a few things
- `sitl_target.cmake` is irrelevant, since you are using ROS to launch the vehicle.
- The file you have zipped only contains a sdf file which is for one vehicle. This shows that your multivehicle script was launching the vehicle with the same config (as you defined in your sdf file) and therefore you can only control a single vehicle. 

The reason `iris` works, is because it is templated for the mutlivehicle use case as jinja templates: https://github.com/PX4/PX4-SITL_gazebo/blob/d8366bf2389eae6106d1dbfaac72ebfdf23a5d2d/models/iris/iris.sdf.jinja

Therefore if you want to use your custom model for multivehicle simulations you also need to template it as [jinja](https://jinja.palletsprojects.com/en/3.0.x/) templates

This is the reason the `iris_obs_avoid` model doesn't support multivehicles since it is not templated

KrisRadowski, 2021-10-02:
> Therefore if you want to use your custom model for multivehicle simulations you also need to template it as [jinja](https://jinja.palletsprojects.com/en/3.0.x/) templates

ok, so I will try to make a template and close this issue for now

KrisRadowski, 2021-10-02:
@Jaeyoung-Lim 

ok, I tried a naive approach (making an sdf file copy and suffixing a jinja file extension), but still; it's not working...

Jaeyoung-Lim, 2021-10-02:
@KrisRadowski I dont think that would work. Please use the iris.jinja as an example to understand how thebtemplating works

KrisRadowski, 2021-10-02:
> Please use the iris.jinja as an example to understand how thebtemplating works

I found iris**.sdf**.jinja, but not iris.jinja (but I seem to understand how Jinja works, though)

Jaeyoung-Lim, 2021-10-02:
@KrisRadowski Sorry, I meant `iris.sdf.jinja` which I linked above.

KrisRadowski, 2021-10-03:
ok, I tried to copy content from iris jinja file (inside `<model>` element) to my drone...

it was dirty method but it worked though :)

closed 2021-10-03 by KrisRadowski