PX4/PX4-Autopilot #18321: Add UAVCAN V0 RTCM Correction Transmission

AlexKlimaj, 2021-09-30:
The ARK RTK GPS and Freefly RTK GPS cannodes currently have no way of getting RTCM corrections from the flight controller over UAVCAN.

dakejahl, 2021-09-30:
Looks like we just need to subscribe to the `gps_inject_data` topic in **uavcan/sensors/gnss.cpp** and pack that data into a uavcan `RTCMStream` message. Probably could add it directly to the `broadcast_from_orb` function in **gnss.cpp**

You could probably do this easily, if you want to send me hardware I can do it.

AlexKlimaj, 2021-10-01:
It looks like the publisher is already supported on the autopilot side.
https://github.com/PX4/PX4-Autopilot/blob/master/src/drivers/uavcan/sensors/gnss.cpp#L426-L454

I'll create the subscriber on the uavcannode end.

Maybe I'll change this topic to focus on moving baseline over uavcan for heading estimation.

AlexKlimaj, 2021-10-01:
https://github.com/PX4/PX4-Autopilot/pull/18323

AlexKlimaj, 2021-10-02:
Resolved with https://github.com/PX4/PX4-Autopilot/pull/18323

closed 2021-10-02 by AlexKlimaj