PX4/PX4-Autopilot #18328: Failure during Build ROS 2 Workspace in docker container on Nvidia Jetson AGX Xavier

codingCoffee, 2021-10-01:
**Describe the bug**
1. I've been following [ROS 2 User Guide (PX4-ROS 2 Bridge)](http://docs.px4.io/master/en/ros/ros2_comm.html#build-ros-2-workspace) to setup ROS 2 with PX4 and the last step to build the ROS 2 workspace is failing, particularly the following command.
```
source build_ros2_workspace.bash
```
It fails and gives a non-zero exit code. Hence, it also closes the terminal. So I executed the same command as follows
```
bash build_ros2_workspace.bash --ros_distro foxy --ros_path /opt/ros/foxy/install/setup.bash --verbose
```

**To Reproduce**
Steps to reproduce the behavior:
Follow the official [ROS 2 User Guide (PX4-ROS 2 Bridge)](http://docs.px4.io/master/en/ros/ros2_comm.html#build-ros-2-workspace)

<details><summary>Below is the Dockerfile to reproduce the entire setup</summary>
<p>

```
#
# this dockerfile roughly follows the 'Install ROS From Source' procedures from:
#   https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/
#
ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.5.0
FROM ${BASE_IMAGE}

ARG ROS_PKG=ros_base
ENV ROS_DISTRO=foxy
ENV ROS_ROOT=/opt/ros/${ROS_DISTRO}

ENV DEBIAN_FRONTEND=noninteractive
ENV SHELL /bin/bash
SHELL ["/bin/bash", "-c"]

WORKDIR /tmp

# change the locale from POSIX to UTF-8
RUN locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
ENV LANG=en_US.UTF-8
ENV PYTHONIOENCODING=utf-8

#
# add the ROS deb repo to the apt sources list
#
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
                curl \
                wget \
                gnupg2 \
                lsb-release \
                ca-certificates \
    && rm -rf /var/lib/apt/lists/* \
    && apt-get clean

RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

#
# install development packages
#
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
                build-essential \
                cmake \
                git \
                libbullet-dev \
                libpython3-dev \
                python3-colcon-common-extensions \
                python3-flake8 \
                python3-pip \
                python3-numpy \
[... 271 lines cut ...]
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/micrortps_agent.dir/build.make:496: recipe for target 'CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o' failed
CMakeFiles/Makefile2:173: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
--- stderr: px4_ros_com
c++: error: /root/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp: No such file or directory
c++: fatal error: no input files
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< px4_ros_com [4.05s, exited with code 2]

Summary: 1 package finished [41.2s]
  1 package failed: px4_ros_com
  1 package had stderr output: px4_ros_com
```

**Additional context**
I've been executing all the steps inside a docker container on an Nvidia Jetson AGX Xavier, which is an ARM processor.

Li-Jinjie, 2022-01-12:
I meet the same failure when building ROS 2 Workspace in a docker container on Nvidia Jetson TX2 NX

Li-Jinjie, 2022-01-13:
Hello, I find a solution.
Just install python3-genmsg by inputing 
`apt-get install python3-genmsg`
This works for me.

Besides, I upload my docker image to docker hub: https://hub.docker.com/r/lijinjie/px4-ros2-arm64 , which is verified on TX2 NX.

I hope this method can solve your problem :)

Ray0117, 2022-05-26:
Hi, I think it may be too late to tell the solution. But it may be helpful for the other users.

Actually, the official ROS2 user guide by PX4 is not compatible with Docker users. I found the method below, which can easily solve the compile error of px4_ros_com in ROS2 docker.

`mkdir -p ~/px4_ros_com_ros2/src`
`git clone https://github.com/PX4/px4_ros_com.git ~/px4_ros_com_ros2/src/px4_ros_com`
`git clone https://github.com/PX4/px4_msgs.git ~/px4_ros_com_ros2/src/px4_msgs`
**DIFFERENT BUT USEFUL STEPS**
`cd ~/px4_ros_com_ros2/src`
`source /opt/ros/foxy/setup.bash`
`colcon build`
`source install/setup.bash`

Then, everything will be done. You can `cd` to the PX4 code folder and compile what you want.

closed 2025-11-30 by github-actions[bot]