PX4/PX4-Autopilot #18333: Add UAVCAN U-Blox Moving Baseline

AlexKlimaj, 2021-10-02:
Starting a thread to track adding moving baseline on u-blox modules over UAVCAN. The current implementation requires the use of two UARTs on the flight controller and the second UART on each GPS connected together.

amakurin, 2021-10-06:
Hey, guys! My team is looking forward to implement the same functionality. Actually we've already started. We've branched from stable 12.3. Our current plan was to minimally patch GPS module and, at first, implement RTCM packages transfer using InjectData\GPSDump. 

If it is appropriate, could you please share your understanding of the best option for supporting moving base over UAVCAN? 
Of course we want to implement this as close to your understanding as possible: 
- we're not so experienced with px4 and can easily misunderstand "the big idea", so we consider you architects;
- we don't want to get troubles with migrating to next stable branch;
- we're ready to push our work back to px4 repo.

Thanks in advance!

AlexKlimaj, 2021-10-06:
Here is what I think right now needs to be done. Let me know if I am missing or misunderstanding anything.

- Increase the UART1 baudrate on the ublox module
- Implement a parameter that can be set through uavcan to set each GPS as a rover and moving base
- Based on what is it set to, enable the relevant RTCM messages on each end
- Sending RTCM messages from the flight controller to the uavcannode is already implemented. Since they are on the same bus, the rover and moving base can directly publish and subscribe to each others messages. So publishing the RTCM messages on the uavcannode side will need to be added.

AlexKlimaj, 2021-10-11:
I have a basic proof of concept working.
https://github.com/AlexKlimaj/PX4-Autopilot/tree/pr-uavcan-moving-base

AlexKlimaj, 2021-12-01:
Done via https://github.com/PX4/PX4-Autopilot/pull/18412

closed 2021-12-01 by AlexKlimaj