PX4/PX4-Autopilot #18334: COM_RC_MODE_IN and COM_RCL_EXCEPT not working as expected

Alkxndr, 2021-10-02:
**Describe the bug**
1. COM_RC_MODE_IN when set to 1 still triggers fail safe upon loss of RC signal
2. COM_RCL_EXCEPT the selected bits are not staying assigned to the appropriate "checked" boxes

**To Reproduce**
Steps to reproduce the behavior:
1. Set up drone with COM_RC_MODE_IN set to 1 "Joystick/No RC Checks"
2. Connect RC Radio and Telemetry to QGC 
2. Fly mission
3. Force RC loss by turning off radio
4. Drone enters failsafe mode and does not allow QGC connected via telemetry to enter hold mode only RTL and Land

5. Attempt to circumvent by setting COM_RCL_EXCEPT to select Mission/Hold/Offboard
6. COM_RCL_EXCEPT set to 0 does not enable any ignored modes. COM_RCL_EXCEPT set to 1 enables hold to be ignored. COM_RCL_EXCEPT set to 2 enables Offboard to be ignored. COM_RCL_EXCEPT set to 3 enables hold and off board to be ignored. No way to ignore mission mode in COM_RCL_EXCEPT.

**Expected behavior**
There should be a way to ignore RC loss for BVLOS operations using either COM_RC_MODE_IN or COM_RCL_EXCEPT. If I set COM_RC_MODE_IN to 1, I should be able to lose RC without triggering a failsafe allowing for seamless transition in and out of RC signal with only warning. Due to [the discussion here](https://github.com/PX4/PX4-Autopilot/pull/17785) it seems COM_RCL_EXCEPT was decided as a better way of doing it, however, it seems to be broken in the most recent release. 

**Drone (please complete the following information):**
- Standard s500 v2 with Holybro Pixhawk4

benjinne, 2021-10-11:
I'm not sure about mission mode, but for offboard, it works as correctly when setting COM_RCL_EXCEPT 1<<2 or 4. Here's a recent PR I made to address offboard failsafes https://github.com/PX4/PX4-Autopilot/pull/18160

benjinne, 2021-10-11:
https://github.com/PX4/PX4-Autopilot/blob/bf48057090fa9da54e4113c40af7d4707444feea/src/modules/commander/state_machine_helper.h#L104-L108

MaEtUgR, 2021-11-15:
> 1. COM_RC_MODE_IN when set to 1 still triggers fail safe upon loss of RC signal

@Alkxndr `COM_RC_MODE_IN` is used to indicate what input you're using e.g. Joystick vs RC receiver. It did not change much other than disabling the arm gesture and the checks if the RC receiver is calibrated before #17404. After that it will only accept stick input from the configured source.

> 2. COM_RCL_EXCEPT the selected bits are not staying assigned to the appropriate "checked" boxes

For "There should be a way to ignore RC loss for BVLOS operations" this is exactly the parameter to configure this. Did you by any chance run into this UI issue when testing it? https://github.com/mavlink/qgroundcontrol/issues/9854 , https://github.com/mavlink/qgroundcontrol/issues/9770 which was recently fixed in QGroundcontrol?

I quickly tested on master and the configured first bit allows flying missions without RC.
Since the latest version QGC can currently not be downloaded from the website (daily build is at the moment no uploaded) you can use the stable version which probably doesn't have the issue or set the parameter on the [console](https://docs.qgroundcontrol.com/master/en/analyze_view/mavlink_console.html) directly by typing `param set COM_RCL_EXCEPT 7`.

shikaison, 2023-04-17:
With vio, visual positioning is used to display no parameter set: COM_ACL_EXCEPT, which is not present in qgc

closed 2025-11-05 by MaEtUgR