PX4/PX4-Autopilot #18335: px4flow module gives correct data, but drone can't control its height

LXYYY, 2021-10-03:
**Describe the bug**
I am using pixhawk with px4flow, with firmware 1.12. I  connected the px4flow module to the px4, and set the parameters according to the tutorial (enable optical flow, mounting positioni and ekf2), and the LOCAL_POSITION_NED and OPTICAL_FLOW_RAD gives data that looks fine, but after it takes off, it always suddenly fly up very fast, and hit the ceiling.

Can anyone give me some suggestions?

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

bresch, 2021-10-11:
> Always provide a link to the flight log file

LXYYY, 2021-11-25:
Sorry for late reply. I built another drone using realsense t265, and followed the official guide of using t265 with px4. But I still got the same problem, the drone started to fly up rapidly after a few seconds. the log is here:

https://review.px4.io/plot_app?log=38df69cc-ffc7-4576-bf66-1336a24b50eb

I killed the drone at 0:37 after it started to fly up really fast. so the problem happened between 36 ~ 37 second. It feels like the z data failed to update, but I already set ekf2 height mode to vision, and x and y data from realsense looks fine.

Anyone has some ideas? Thanks.

bresch, 2022-01-12:
Hi @LXYYY ,
Have you tested the quality of the height measurement of the VIO solution (e.g.: without flying, just a handheld test)? From the log it looks that the measurement is quite wrong. The odometry message (VIO) quickly disagree with the EKF in speed and position. Furthermore the distance measurement is telling that the drone flies at 12cm above ground and then 10cm just before you killed the vehicle. This seems quite wrong since you said that the vehicle started to fly upwards (and the baro data shows that the drone is climbing).

My conclusion is then that the VIO measurement is wrong. You should investigate that without flying of when flying in manual mode to avoid crashing your drone again.
[image]

Good luck!

Kihoon-Shin, 2023-03-31:
@LXYYY 
Hello, I'm testing with px4flow but I have similar problem as you.
After take off, I controlled the drone with Altitude mode.
But about 30seconds later, It fly up rapidly..
Do you know what problem is??

LXYYY, 2023-03-31:
It was a long time ago, I don't really remember how this problem disappeared. I suggest you have a look at the height sensors you have on your drone. It happened to me once, after I upgraded or downgraded (don't remember) the px4 version, it lost some drivers for some laser height sensors, as a result they start giving random readings, even the hardwares are not changed.

This problem I mentioned in this issue, even though I am still not sure, but I suspect the power supply is not enough for the realsense, I end up put a separate battery for the raspi and realsense

hamishwillee, 2025-10-29:
I'm going to close this as too old. The original post was 2021 and PX4v1.12. If someone has a similar problem they should post detail against a recent build (and possibly cross link to this)

closed 2025-10-29 by hamishwillee