PX4/PX4-Autopilot #18365: How to add my own CAN bus device

lfeng17, 2021-10-05:
Hello, I noticed that the pixhawk4 has 2 CAN, but I didn't find a clear example like uart in the src. So how can I set CAN rate or mask or ID and read CAN data into PX4?

astekachev, 2022-11-01:
Hello @dagar !
> The primary use case is with UAVCAN (https://github.com/PX4/PX4-Autopilot/tree/master/src/drivers/uavcan), but you can also use the NuttX CAN driver (eg /dev/can0) and write anything custom you want.
Can you please tell me where to find an example of how to use Nuttx CANBus. I want to connect my existing MMW radar NRA24 by Nanoradar. And try to connect it to uORB network with DISTANCE_SENSOR message. 
I think it will be useful for the PX4 project, considering that the ArduPilot already has this radar connected.
If I succeed I will post the source code here.
Best regards

astekachev, 2022-11-03:
Hi all!
I will use the /tattu_can driver as an example.

kjgingerale, 2023-01-18:
Hey @astekachev ! I'm working on something similar. I'm looking to pull in some data from the canbus from a liquid level sensor. I was wondering how everything worked out for you?

DenisPGH, 2023-06-16:
> Hi all! I will use the /tattu_can driver as an example.

Hello @astekachev I also try to implement a new sensor with CAN communication and found "/tattu_can" as example. Was it good example to follow? Could You please share some experience. Thank You very much!

hamishwillee, 2025-10-29:
@dakejahl I propose we add a section in the docs for this and close the issue.

What would you advise here? Above @dagar says:

> The primary use case is with UAVCAN (https://github.com/PX4/PX4-Autopilot/tree/master/src/drivers/uavcan), but you can also use the NuttX CAN driver (eg /dev/can0) and write anything custom you want.

Would be good to provide current links, and list some good examples if you know some.

julianoes, 2025-10-31:
Here is a thread: https://discuss.px4.io/t/nuttx-can-drivers-on-stm32h7x3/29828/39

hamishwillee, 2025-11-06:
Thanks. None of that means anything much to me.

closed 2026-08-27 by github-actions[bot]