PX4/PX4-Autopilot #18369: Bug Found in msgID #332 mavlink protocol BOF

BOB4Drone, 2021-10-06:
Steps to reproduce the behavior:
-----------------------------------
1. Turn the drone on
2. Set the lower one byte to be greater than 0x75 except for the CRC of the MAVLINK PROCOL packet which is msgID 332
3. Send a packet set as a serial port to the drone.
4. BOF crash occurs.

example msgID332 packet
-----------------------------------
fdef0000d2ff004c010003a8ba951e05809b5f1b8c9069167f55c709ad76837e1a585c22264e6a3785ce716070b9ee1954a849e718eb9bb2bdcc21a27d7c75075d3553250d12b7abd7c267c91e2c31ad1e0ce5fc3fd0403c104fb304935a8fa5b399a70fa94b5fcebdafc4cd0dc0c144dd8ddfd7cbea9c124490d4c80671cab01dce514f299b9598f8a193156c981097892bdead25a0b63e7572f73325bd05ff57911407eed28b44f339d09c945812a81661bf8cf633f5e1f4836f4bbea984c8a37e780b53fc9e93df755c224e40326a27e767c3279e3c19fc1af681cf0de9e14b3bbdf67646f2e783a9e2d506eb0354974f14c34c3a03777ad3b9

Describe the bug
-----------------------------------
The lower 1 byte of MAVLINK PROTOCOL with msgID number 332 is "valid_points".
We can set "valid_points" to a maximum of 0xff.
BOF occurs in the source code line below when "valid_points" is set to a large value.
I found it in sitl mode and also checked it on the drone board.

source code
-----------------------------------
https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mavlink/mavlink_receiver.cpp
PX4-Autopilot/src/modules/mavlink/mavlink_receiver.cpp (line 1942 ~ line 1944)

TEST board
-----------------------------------
px4_fmu-v5
nxp_fmuk66-v3

log file page
-----------------------------------
https://logs.px4.io/plot_app?log=1b47a1c3-3afe-4518-9628-f3284d67fb41

[image]

dagar, 2021-10-06:
Thanks for reporting. It appears to be a slight discrepancy with the size difference of the uORB msg and mavlink.

https://github.com/PX4/PX4-Autopilot/blob/e31dd3438887ecff830bb8dd36f20d8a32b3e2cd/src/modules/mavlink/mavlink_receiver.cpp#L1913-L1951

dagar, 2021-10-06:
@BOB4Drone could you try https://github.com/PX4/PX4-Autopilot/pull/18371?

l3u9, 2021-10-06:
Crash did not occur when tested on simulator and drone board.
It seems to have been fixed well.