PX4/PX4-Autopilot #18383: Support for control zero classic in mRobotics

muramura, 2021-10-07:
**Describe problem solved by the proposed feature**
A clear and concise description of the problem, if any, this feature will solve. E.g. I'm always frustrated when ...

mRobotics control zero classic is not recognized.

**Describe your preferred solution**
A clear and concise description of what you want to happen.

I would like to see support for control zero classic in mRobotics.

**Describe possible alternatives**
A clear and concise description of alternative solutions or features you've considered.

None.

**Additional context**
Add any other context or screenshots for the feature request here.

taileron, 2021-10-08:
ardupilots hwdef/hwdef.dat of each fc shows further details 
probably QGC doesn´t already see it´s board id:1022
but flashing with custom file from zero h7 should boot with the sensors but with some IOs missing

TOTON95, 2024-08-09:
@muramura Sorry for the inconvenience, this board is now operational in #22745, backported into stable #22971, and recognized by the latest version of QGC if needed.