PX4/PX4-Autopilot #18395: 3 problems

Kim-zhi-jiang, 2021-10-09:
make -j4 px4_sitl gazebo_standard_vtol

1 parameter file not found
2 not enough arguments. Try 'param set PWM_AUX_OUT 3 [fail]'
3 Too many subscriptions, failed to add: vehicle_trajectory_waypoint_desired 
[image]
[image]

[image]
0

Kim-zhi-jiang, 2021-10-09:
Use v-master of today.

Jaeyoung-Lim, 2021-10-09:
@Kim-zhi-jiang standard_vtol is not supported in jmavsim

Kim-zhi-jiang, 2021-10-09:
@Jaeyoung-Lim    VTOL Testing has been used