PX4/PX4-Autopilot #18396: Service response transfer-ID not matched

Cherish-Gww, 2021-10-09:
In file:
https://github.com/PX4/PX4-Autopilot/blob/4be45229bf074c0c3869ac35d192a39167e3b52a/src/drivers/uavcan_v1/ServiceClients/GetInfo.hpp#L115

transfer-ID defined:
Service response matching – when a server responds to a request, it uses the same transfer-ID for the response transfer as in the request transfer, allowing the client to emit concurrent requests to the same server while being able to match each response with the corresponding local request state.

dagar, 2021-10-09:
@PetervdPerk-NXP @JacobCrabill

PetervdPerk-NXP, 2021-10-09:
Hi @Cherish-Gww 

Thanks for pointing that out, it seems I've completely missed that in all service implementations.
E.g. the service invoker does ignore this as well.
https://github.com/PX4/PX4-Autopilot/blob/1ee5f75aa50f6f21d9d4df85d6aec8e41f6dd01d/src/drivers/uavcan_v1/Services/ServiceRequest.hpp#L86-L93

We've to add these checks and respond with the correct id indeed then.

commit ed394027 (2021-10-14): UAVCANv1 Include Kconfig & Fix #18396
closed 2021-10-14 by dagar