PX4/PX4-Autopilot #18413: ModuleNotFoundError: No module named 'menuconfig' CMake Error at cmake/kconfig.cmake:6 (message)

sanket1211996, 2021-10-12:
I run the command make "px4_sitl jmavsim", it failing and giving below error I tried all solution on the internet nothing worked.
[image]

dagar, 2021-10-12:
The kconfiglib dependency came in last week. You'll need to update the dependencies (look under Tools/setup) or jump back to slightly older PX4.

dagar, 2021-10-12:
You need to make sure it's the same python environment that's being used within cmake.

The kconfig change merged about 5 days ago. Here's the last commit before. https://github.com/PX4/PX4-Autopilot/commit/0cae3c129dd751e92ef5785ad7ba2b86ce14e4f3

ganesh-androidev, 2021-11-29:
I get a Following error on latest checkout, Any help @dagar 
$ make px4_sitl jmavsim

-- PX4 version: v1.13.0-alpha1-3631-gc2cbab1e98
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.5", minimum required is "3")
-- PX4 config file: /Users/gramamoorthy/developer/Firmware/boards/px4/sitl/default.px4board
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'menuconfig'
CMake Error at cmake/kconfig.cmake:6 (message):
  kconfiglib is not installed or not in PATH

  please install using "pip3 install kconfiglib"

Call Stack (most recent call first):
  CMakeLists.txt:168 (include)

-- Configuring incomplete, errors occurred!
Error: /Users/gramamoorthy/developer/Firmware/build/px4_sitl_default is not a directory
make: *** [px4_sitl] Error 1

dagar, 2021-11-29:
Did you try installing kconfiglib (error output)?

``` Console
please install using "pip3 install kconfiglib"
```

Another way to do it is through the build `make px4_sitl_default install_python_requirements` and then it uses the correct python.

ganesh-androidev, 2021-11-29:
Yeah , I did 

$ pip3 install kconfiglib
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: kconfiglib in /Users/gramamoorthy/Library/Python/3.8/lib/python/site-packages (14.1.0)

$ make px4_sitl_default install_python_requirements
-- PX4 version: v1.13.0-alpha1-3631-gc2cbab1e98
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.5", minimum required is "3")
-- PX4 config file: /Users/gramamoorthy/developer/Firmware/boards/px4/sitl/default.px4board
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'menuconfig'
CMake Error at cmake/kconfig.cmake:6 (message):
  kconfiglib is not installed or not in PATH

  please install using "pip3 install kconfiglib"

Call Stack (most recent call first):
  CMakeLists.txt:168 (include)

-- Configuring incomplete, errors occurred!
Error: /Users/gramamoorthy/developer/Firmware/build/px4_sitl_default is not a directory
make: *** [px4_sitl_default] Error 1

dagar, 2021-11-29:
Try with `/usr/local/bin/python3 -m pip install kconfiglib`?

It looks like there's confusion between a system default 3.8 and python 3.7.5 found by cmake.

ganesh-androidev, 2021-11-29:
Thanks Dagar that helped. i used --user to install. 

$ /usr/local/bin/python3 -m pip install kconfiglib --user 

Any idea on below error : 

 Firmware % make px4_sitl jmavsim

[70/521] Generating Mavlink development: src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Traceback (most recent call last):
  File "/Users/gramamoorthy/developer/Firmware/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/Users/gramamoorthy/developer/Firmware/src/modules/mavlink/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ModuleNotFoundError: No module named 'future'
[71/521] Building CXX object src/modules/mc_att_control/CMakeFiles/modules__mc_att_control.dir/mc_att_control_main.cpp.o
FAILED: mavlink/development/development.h /Users/gramamoorthy/developer/Firmware/build/px4_sitl_default/mavlink/development/development.h
cd /Users/gramamoorthy/developer/Firmware/build/px4_sitl_default/src/modules/mavlink && /usr/local/bin/python3 /Users/gramamoorthy/developer/Firmware/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py --lang C --wire-protocol 2.0 --output /Users/gramamoorthy/developer/Firmware/build/px4_sitl_default/mavlink /Users/gramamoorthy/developer/Firmware/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
[89/521] Building CXX object src/modules/mc_att_control/CMakeFiles/modules__mc_att_control.dir/mc_att_control_main.cpp.o
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

ganesh-androidev, 2021-11-29:
This worked

/usr/local/bin/python3 -m pip install future --user 

and i re-run this :
/usr/local/bin/python3 -m pip install numpy --user

mengchaoheng, 2022-02-27:
the same problem,I don't know how to fix it. @dagar

dagar, 2022-02-27:
> the same problem,I don't know how to fix it. @dagar

@mengchaoheng can you paste the full output of what you're trying? A lot of the confusion comes from having multiple versions of python on your system. You need to make sure the one that PX4 cmake finds has all the dependencies installed.

From PX4-Autopilot try...
``` Console
$ python3 -m pip install -r Tools/setup/requirements.txt
```

If that doesn't work take note of the specific python found by PX4 cmake configure and use the full path explicitly.
[image]

In my case that would be `/usr/bin/python3`.

``` Console
$ /usr/bin/python3 -m pip install -r Tools/setup/requirements.txt
```

mengchaoheng, 2022-02-27:
@dagar output as follow:
mengchaoheng@mengchaohengdeMacBook-Pro Downloads % cd PX4-Autopilot 
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % ls
CMakeLists.txt			Jenkinsfile			ROMFS				eclipse.project			posix-configs
CODE_OF_CONDUCT.md		Kconfig				Tools				integrationtests		src
CONTRIBUTING.md			LICENSE				appveyor.yml			launch				test
CTestConfig.cmake		Makefile			boards				msg				test_data
Documentation			PULL_REQUEST_TEMPLATE.md	cmake				package.xml			validation
Firmware.sublime-project	README.md			eclipse.cproject		platforms
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % make px4_sitl jmavsim
-- PX4 version: v1.13.0-alpha1-4371-gc2c455be0d
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.10", minimum required is "3") 
-- PX4 config file: /Users/mengchaoheng/Downloads/PX4-Autopilot/boards/px4/sitl/default.px4board
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'menuconfig'
CMake Error at cmake/kconfig.cmake:6 (message):
  kconfiglib is not installed or not in PATH

  please install using "pip3 install kconfiglib"

Call Stack (most recent call first):
  CMakeLists.txt:174 (include)

-- Configuring incomplete, errors occurred!
Error: /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default is not a directory
make: *** [px4_sitl] Error 1
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % pip3 install kconfiglib
Defaulting to user installation because normal site-packages is not writeable
Collecting kconfiglib
  Using cached kconfiglib-14.1.0-py2.py3-none-any.whl (145 kB)
Installing collected packages: kconfiglib
  WARNING: The scripts alldefconfig, allmodconfig, allnoconfig, allyesconfig, defconfig, genconfig, guiconfig, listnewconfig, menuconfig, oldconfig, olddefconfig, savedefconfig and setconfig are installed in '/Users/mengchaoheng/Library/Python/3.8/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed kconfiglib-14.1.0
WARNING: You are using pip version 20.2.3; however, version 22.0.3 is available.
You should consider upgrading via the '/Library/Developer/CommandLineTools/usr/bin/python3 -m pip install --upgrade pip' command.
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % make px4_sitl jmavsim  
-- PX4 version: v1.13.0-alpha1-4371-gc2c455be0d
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.10", minimum required is "3") 
-- PX4 config file: /Users/mengchaoheng/Downloads/PX4-Autopilot/boards/px4/sitl/default.px4board
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'menuconfig'
CMake Error at cmake/kconfig.cmake:6 (message):
  kconfiglib is not installed or not in PATH

  please install using "pip3 install kconfiglib"

Call Stack (most recent call first):
  CMakeLists.txt:174 (include)

-- Configuring incomplete, errors occurred!
Error: /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default is not a directory
make: *** [px4_sitl] Error 1
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % git status   
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % git submodule update --init --recursive
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % make px4_sitl jmavsim                  
-- PX4 version: v1.13.0-alpha1-4371-gc2c455be0d
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.10", minimum required is "3") 
-- PX4 config file: /Users/mengchaoheng/Downloads/PX4-Autopilot/boards/px4/sitl/default.px4board
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'menuconfig'
CMake Error at cmake/kconfig.cmake:6 (message):
  kconfiglib is not installed or not in PATH
[... 109 lines cut ...]
Entering 'Tools/jsbsim_bridge'
Entering 'Tools/simulation-ignition'
Entering 'Tools/sitl_gazebo'
Entering 'platforms/nuttx/NuttX/apps'
Entering 'platforms/nuttx/NuttX/nuttx'
Entering 'src/drivers/gps/devices'
Entering 'src/drivers/uavcan/libuavcan'
Entering 'src/drivers/uavcan_v1/legacy_data_types'
Entering 'src/drivers/uavcan_v1/libcanard'
Entering 'src/drivers/uavcan_v1/public_regulated_data_types'
Entering 'src/lib/crypto/libtomcrypt'
Entering 'src/lib/crypto/libtommath'
Entering 'src/lib/crypto/monocypher'
Entering 'src/lib/events/libevents'
Entering 'src/modules/mavlink/mavlink'
Entering 'src/modules/micrortps_bridge/micro-CDR'
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % git --version
git version 2.32.0 (Apple Git-132)
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % brew --version
Homebrew 3.3.16

mengchaoheng, 2022-02-27:
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % python3 -m pip install -r Tools/setup/requirements.txt
Defaulting to user installation because normal site-packages is not writeable
Collecting argcomplete
  Downloading argcomplete-2.0.0-py2.py3-none-any.whl (37 kB)
Collecting argparse>=1.2
  Downloading argparse-1.4.0-py2.py3-none-any.whl (23 kB)
Collecting cerberus
  Downloading Cerberus-1.3.4.tar.gz (63 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 63.8/63.8 KB 1.7 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting coverage
  Downloading coverage-6.3.2-cp38-cp38-macosx_10_9_x86_64.whl (183 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 183.1/183.1 KB 1.2 MB/s eta 0:00:00
Collecting empy>=3.3
  Using cached empy-3.3.4-py3-none-any.whl
Collecting jinja2>=2.8
  Using cached Jinja2-3.0.3-py3-none-any.whl (133 kB)
Collecting matplotlib>=3.0.*
  Downloading matplotlib-3.5.1-cp38-cp38-macosx_10_9_x86_64.whl (7.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 7.3/7.3 MB 3.0 MB/s eta 0:00:00
Collecting numpy>=1.13
  Downloading numpy-1.22.2-cp38-cp38-macosx_10_14_x86_64.whl (17.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 17.6/17.6 MB 4.2 MB/s eta 0:00:00
Collecting nunavut>=1.1.0
  Downloading nunavut-1.7.3-1904746659-py3-none-any.whl (311 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 311.1/311.1 KB 646.2 kB/s eta 0:00:00
Collecting packaging
  Using cached packaging-21.3-py3-none-any.whl (40 kB)
Collecting pandas>=0.21
  Downloading pandas-1.4.1-cp38-cp38-macosx_10_9_x86_64.whl (11.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.4/11.4 MB 3.5 MB/s eta 0:00:00
Collecting pkgconfig
  Downloading pkgconfig-1.5.5-py3-none-any.whl (6.7 kB)
Collecting psutil
  Downloading psutil-5.9.0-cp38-cp38-macosx_10_9_x86_64.whl (238 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 238.6/238.6 KB 746.0 kB/s eta 0:00:00
Collecting pygments
  Downloading Pygments-2.11.2-py3-none-any.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 1.5 MB/s eta 0:00:00
Requirement already satisfied: wheel>=0.31.1 in /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/site-packages (from -r Tools/setup/requirements.txt (line 15)) (0.33.1)
Collecting pymavlink
  Downloading pymavlink-2.4.26.tar.gz (5.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.2/5.2 MB 3.0 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting pyros-genmsg
  Using cached pyros_genmsg-0.5.8-py2.py3-none-any.whl (25 kB)
Collecting pyserial
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting pyulog>=0.5.0
  Downloading pyulog-0.9.0-py3-none-any.whl (23 kB)
Collecting pyyaml
  Downloading PyYAML-6.0-cp38-cp38-macosx_10_9_x86_64.whl (192 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 192.2/192.2 KB 286.4 kB/s eta 0:00:00
Collecting requests
  Downloading requests-2.27.1-py2.py3-none-any.whl (63 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 63.1/63.1 KB 181.1 kB/s eta 0:00:00
Requirement already satisfied: setuptools>=39.2.0 in /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/site-packages (from -r Tools/setup/requirements.txt (line 22)) (49.2.1)
Requirement already satisfied: six>=1.12.0 in /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/site-packages (from -r Tools/setup/requirements.txt (line 23)) (1.15.0)
Collecting toml>=0.9
  Using cached toml-0.10.2-py2.py3-none-any.whl (16 kB)
Collecting MarkupSafe>=2.0
  Downloading MarkupSafe-2.1.0-cp38-cp38-macosx_10_9_x86_64.whl (13 kB)
Collecting pillow>=6.2.0
  Downloading Pillow-9.0.1-cp38-cp38-macosx_10_10_x86_64.whl (3.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.0/3.0 MB 2.4 MB/s eta 0:00:00
Collecting fonttools>=4.22.0
  Downloading fonttools-4.29.1-py3-none-any.whl (895 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 895.5/895.5 KB 3.3 MB/s eta 0:00:00
[... 38 lines cut ...]
  Building wheel for pymavlink (setup.py) ... done
  Created wheel for pymavlink: filename=pymavlink-2.4.26-cp38-cp38-macosx_10_14_x86_64.whl size=5374769 sha256=61ff5f45c203d60b1a0c4327c4a14a2bc2a8ad0c7297ad675567ba7b66d167d0
  Stored in directory: /Users/mengchaoheng/Library/Caches/pip/wheels/8e/27/b8/9128b37fe52b856457b3414d125c11e8232860225b68252f36
Successfully built cerberus pymavlink
Installing collected packages: pytz, pyserial, pyros-genmsg, pydsdl, empy, certifi, argparse, zipp, urllib3, toml, pyyaml, python-dateutil, pyparsing, pygments, psutil, pkgconfig, pillow, numpy, MarkupSafe, lxml, kiwisolver, idna, future, fonttools, cycler, coverage, charset-normalizer, cerberus, argcomplete, requests, pyulog, pymavlink, pandas, packaging, jinja2, importlib-resources, nunavut, matplotlib
  WARNING: The scripts pyserial-miniterm and pyserial-ports are installed in '/Users/mengchaoheng/Library/Python/3.8/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script pygmentize is installed in '/Users/mengchaoheng/Library/Python/3.8/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
[... cut ...]
Successfully installed MarkupSafe-2.1.0 argcomplete-2.0.0 argparse-1.4.0 cerberus-1.3.4 certifi-2021.10.8 charset-normalizer-2.0.12 coverage-6.3.2 cycler-0.11.0 empy-3.3.4 fonttools-4.29.1 future-0.18.2 idna-3.3 importlib-resources-5.4.0 jinja2-3.0.3 kiwisolver-1.3.2 lxml-4.8.0 matplotlib-3.5.1 numpy-1.22.2 nunavut-1.7.3 packaging-21.3 pandas-1.4.1 pillow-9.0.1 pkgconfig-1.5.5 psutil-5.9.0 pydsdl-1.13.0 pygments-2.11.2 pymavlink-2.4.26 pyparsing-3.0.7 pyros-genmsg-0.5.8 pyserial-3.5 python-dateutil-2.8.2 pytz-2021.3 pyulog-0.9.0 pyyaml-6.0 requests-2.27.1 toml-0.10.2 urllib3-1.26.8 zipp-3.7.0
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot %

dagar, 2022-02-27:
@mengchaoheng try `/usr/local/bin/python3 -m pip install -r Tools/setup/requirements.txt`

Then delete the build directory (`rm -rf build/`) before trying `make px4_sitl_default` again.

mengchaoheng, 2022-02-27:
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % /usr/local/bin/python3 -m pip install -r Tools/setup/requirements.txt
zsh: command not found: mengchaoheng@mengchaohengdeMacBook-Pro
zsh: command not found: Defaulting
zsh: command not found: Collecting
zsh: unknown file attribute: 3
zsh: 1.2 not found
zsh: unknown file attribute: 2
zsh: command not found: Collecting
zsh: unknown file attribute: 6
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: command not found: Installing
zsh: command not found: Getting
zsh: unknown file attribute: y
zsh: command not found: Collecting
zsh: unknown file attribute: 1
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: 3.3 not found
zsh: command not found: Using
zsh: 2.8 not found
zsh: unknown file attribute: 1
zsh: 3.0.* not found
zsh: unknown file attribute: 7
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: 1.13 not found
zsh: unknown file attribute: 1
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: 1.1.0 not found
zsh: unknown file attribute: 3
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: command not found: Collecting
zsh: unknown file attribute: 4
zsh: 0.21 not found
zsh: unknown file attribute: 1
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: command not found: Collecting
zsh: unknown file attribute: 6
zsh: command not found: Collecting
zsh: unknown file attribute: 2
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: command not found: Collecting
zsh: unknown file attribute: 1
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: bad pattern: (from -r Tools/setup/requirements.txt (line 15))
zsh: command not found: Collecting
zsh: unknown file attribute: 5
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: missing end of string
zsh: command not found: Collecting
zsh: unknown file attribute: 2
zsh: command not found: Collecting
zsh: unknown file attribute: 9
zsh: 0.5.0 not found
zsh: unknown file attribute: 2
zsh: command not found: Collecting
zsh: unknown file attribute: 1
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: command not found: Collecting
zsh: unknown file attribute: 6
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: bad pattern: (from -r Tools/setup/requirements.txt (line 22))
zsh: bad pattern: (from -r Tools/setup/requirements.txt (line 23))
zsh: 0.9 not found
zsh: unknown file attribute: 1
zsh: 2.0 not found
zsh: unknown file attribute: 1
zsh: 6.2.0 not found
zsh: unknown file attribute: 3
zsh: command not found: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
zsh: 4.22.0 not found
zsh: unknown file attribute: 8
[... 115 lines cut ...]
Failed to import toml: No module named 'toml'

You may need to install it using:
    pip3 install --user toml

[9/807] Generating uORB topic headers
FAILED: uORB/topics/uORBTopics.hpp uORB/topics/actuator_armed.h uORB/topics/actuator_controls.h uORB/topics/actuator_outputs.h uORB/topics/adc_report.h uORB/topics/airspeed.h uORB/topics/airspeed_validated.h uORB/topics/airspeed_wind.h uORB/topics/battery_status.h uORB/topics/camera_capture.h uORB/topics/camera_trigger.h uORB/topics/cellular_status.h uORB/topics/collision_constraints.h uORB/topics/collision_report.h uORB/topics/commander_state.h uORB/topics/control_allocator_status.h uORB/topics/cpuload.h uORB/topics/debug_array.h uORB/topics/debug_key_value.h uORB/topics/debug_value.h uORB/topics/debug_vect.h uORB/topics/differential_pressure.h uORB/topics/distance_sensor.h uORB/topics/ekf2_timestamps.h uORB/topics/ekf_gps_drift.h uORB/topics/esc_report.h uORB/topics/esc_status.h uORB/topics/estimator_event_flags.h uORB/topics/estimator_innovations.h uORB/topics/estimator_optical_flow_vel.h uORB/topics/estimator_selector_status.h uORB/topics/estimator_sensor_bias.h uORB/topics/estimator_states.h uORB/topics/estimator_status.h uORB/topics/estimator_status_flags.h uORB/topics/follow_target.h uORB/topics/generator_status.h uORB/topics/geofence_result.h uORB/topics/gimbal_device_attitude_status.h uORB/topics/gimbal_device_information.h uORB/topics/gimbal_device_set_attitude.h uORB/topics/gimbal_manager_information.h uORB/topics/gimbal_manager_set_attitude.h uORB/topics/gimbal_manager_set_manual_control.h uORB/topics/gimbal_manager_status.h uORB/topics/gps_dump.h uORB/topics/gps_inject_data.h uORB/topics/heater_status.h uORB/topics/home_position.h uORB/topics/hover_thrust_estimate.h uORB/topics/input_rc.h uORB/topics/iridiumsbd_status.h uORB/topics/irlock_report.h uORB/topics/landing_gear.h uORB/topics/landing_target_innovations.h uORB/topics/landing_target_pose.h uORB/topics/led_control.h uORB/topics/log_message.h uORB/topics/logger_status.h uORB/topics/mag_worker_data.h uORB/topics/manual_control_setpoint.h uORB/topics/manual_control_switches.h uORB/topics/mavlink_log.h uORB/topics/mission.h uORB/topics/mission_result.h uORB/topics/mount_orientation.h uORB/topics/multirotor_motor_limits.h uORB/topics/navigator_mission_item.h uORB/topics/obstacle_distance.h uORB/topics/offboard_control_mode.h uORB/topics/onboard_computer_status.h uORB/topics/optical_flow.h uORB/topics/orb_test.h uORB/topics/orb_test_large.h uORB/topics/orb_test_medium.h uORB/topics/orbit_status.h uORB/topics/parameter_update.h uORB/topics/ping.h uORB/topics/position_controller_landing_status.h uORB/topics/position_controller_status.h uORB/topics/position_setpoint.h uORB/topics/positio [... cut ...] oheng/Downloads/PX4-Autopilot/src/lib/pwm/pwm_extra_params.yaml /Users/mengchaoheng/Downloads/PX4-Autopilot/src/drivers/gps/module.yaml /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/module.yaml
Failed to import jinja2: No module named 'jinja2'

You may need to install it using:
    pip3 install --user jinja2

[13/807] Generating git version header
ninja: build stopped: subcommand failed.
make: *** [px4_sitl_default] Error 1
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot %

mengchaoheng, 2022-02-27:
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % /usr/bin/python3 -m pip install -r Tools/setup/requirements.txt
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: argcomplete in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 1)) (2.0.0)
Collecting argparse>=1.2
  Using cached argparse-1.4.0-py2.py3-none-any.whl (23 kB)
Requirement already satisfied: cerberus in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 3)) (1.3.4)
Requirement already satisfied: coverage in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 4)) (6.3.2)
Requirement already satisfied: empy>=3.3 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 5)) (3.3.4)
Requirement already satisfied: jinja2>=2.8 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 6)) (3.0.3)
Requirement already satisfied: matplotlib>=3.0.* in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 7)) (3.5.1)
Requirement already satisfied: numpy>=1.13 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 8)) (1.22.2)
Requirement already satisfied: nunavut>=1.1.0 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 9)) (1.7.3)
Requirement already satisfied: packaging in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 10)) (21.3)
Requirement already satisfied: pandas>=0.21 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 11)) (1.4.1)
Requirement already satisfied: pkgconfig in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 12)) (1.5.5)
Requirement already satisfied: psutil in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 13)) (5.9.0)
Requirement already satisfied: pygments in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 14)) (2.11.2)
Requirement already satisfied: wheel>=0.31.1 in /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/site-packages (from -r Tools/setup/requirements.txt (line 15)) (0.33.1)
Requirement already satisfied: pymavlink in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 16)) (2.4.26)
Requirement already satisfied: pyros-genmsg in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 17)) (0.5.8)
Requirement already satisfied: pyserial in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 18)) (3.5)
Requirement already satisfied: pyulog>=0.5.0 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 19)) (0.9.0)
Requirement already satisfied: pyyaml in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 20)) (6.0)
Requirement already satisfied: requests in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 21)) (2.27.1)
Requirement already satisfied: setuptools>=39.2.0 in /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/site-packages (from -r Tools/setup/requirements.txt (line 22)) (49.2.1)
Requirement already satisfied: six>=1.12.0 in /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/site-packages (from -r Tools/setup/requirements.txt (line 23)) (1.15.0)
Requirement already satisfied: toml>=0.9 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from -r Tools/setup/requirements.txt (line 24)) (0.10.2)
Requirement already satisfied: MarkupSafe>=2.0 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from jinja2>=2.8->-r Tools/setup/requirements.txt (line 6)) (2.1.0)
Requirement already satisfied: fonttools>=4.22.0 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from matplotlib>=3.0.*->-r Tools/setup/requirements.txt (line 7)) (4.29.1)
Requirement already satisfied: python-dateutil>=2.7 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from matplotlib>=3.0.*->-r Tools/setup/requirements.txt (line 7)) (2.8.2)
Requirement already satisfied: kiwisolver>=1.0.1 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from matplotlib>=3.0.*->-r Tools/setup/requirements.txt (line 7)) (1.3.2)
Requirement already satisfied: cycler>=0.10 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from matplotlib>=3.0.*->-r Tools/setup/requirements.txt (line 7)) (0.11.0)
Requirement already satisfied: pillow>=6.2.0 in /Users/mengchaoheng/Library/Python/3.8/lib/python/site-packages (from matplotlib>=3.0.*->-r Tools/setup/requirements.txt (line 7)) (9.0.1)
[... cut ...] params.yaml /Users/mengchaoheng/Downloads/PX4-Autopilot/src/lib/pwm/pwm_aux_params.yaml /Users/mengchaoheng/Downloads/PX4-Autopilot/src/lib/pwm/pwm_extra_params.yaml /Users/mengchaoheng/Downloads/PX4-Autopilot/src/drivers/gps/module.yaml /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/module.yaml
Failed to import jinja2: No module named 'jinja2'

You may need to install it using:
    pip3 install --user jinja2

ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

mengchaoheng, 2022-02-27:
built in the macbook pro 2021 which has a m1 pro CPU

dagar, 2022-02-27:
> built in the macbook pro 2021 which has a m1 pro CPU

@mengchaoheng I would try PX4 master instead of v1.12.3. Then try re-installing the full dependencies (`./Tools/setup/macos.sh`), restart, and try again from a clean state.

If it's still failing please message me (@dagar) on PX4 slack and we can try to debug interactively.

mengchaoheng, 2022-02-27:
> @mengchaoheng I would try PX4 master instead of v1.12.3. Then try re-installing the full dependencies (`./Tools/setup/macos.sh`), restart, and try again from a clean state.
>
> If it's still failing [...]

I have checkout to v1.12.3, as above

mengchaoheng, 2022-02-27:
I just bought a new computer today (MacBook pro 2021) and followed the tutorial to the instructions below. I'm sorry it might be a bit long:

mengchaoheng@mengchaohengdeMacBook-Pro Downloads % /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"
==> Checking for `sudo` access (which may request your password)...
Password:
==> This script will install:
/usr/local/bin/brew
/usr/local/share/doc/homebrew
/usr/local/share/man/man1/brew.1
/usr/local/share/zsh/site-functions/_brew
/usr/local/etc/bash_completion.d/brew
/usr/local/Homebrew
==> The following existing directories will be made group writable:
/usr/local/bin
/usr/local/share
/usr/local/share/doc
/usr/local/share/man
/usr/local/share/man/man1
==> The following existing directories will have their owner set to mengchaoheng:
/usr/local/bin
/usr/local/share
/usr/local/share/doc
/usr/local/share/man
/usr/local/share/man/man1
==> The following existing directories will have their group set to admin:
/usr/local/bin
/usr/local/share
/usr/local/share/doc
/usr/local/share/man
/usr/local/share/man/man1
==> The following new directories will be created:
/usr/local/etc
/usr/local/include
/usr/local/lib
/usr/local/sbin
/usr/local/var
/usr/local/opt
/usr/local/share/zsh
/usr/local/share/zsh/site-functions
/usr/local/var/homebrew
/usr/local/var/homebrew/linked
/usr/local/Cellar
/usr/local/Caskroom
/usr/local/Frameworks
==> The Xcode Command Line Tools will be installed.

Press RETURN to continue or any other key to abort:
==> /usr/bin/sudo /bin/chmod u+rwx /usr/local/bin /usr/local/share /usr/local/share/doc /usr/local/share/man /usr/local/share/man/man1
==> /usr/bin/sudo /bin/chmod g+rwx /usr/local/bin /usr/local/share /usr/local/share/doc /usr/local/share/man /usr/local/share/man/man1
==> /usr/bin/sudo /usr/sbin/chown mengchaoheng /usr/local/bin /usr/local/share /usr/local/share/doc /usr/local/share/man /usr/local/share/man/man1
==> /usr/bin/sudo /usr/bin/chgrp admin /usr/local/bin /usr/local/share /usr/local/share/doc /usr/local/share/man /usr/local/share/man/man1
==> /usr/bin/sudo /bin/mkdir -p /usr/local/etc /usr/local/include /usr/local/lib /usr/local/sbin /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Frameworks
==> /usr/bin/sudo /bin/chmod ug=rwx /usr/local/etc /usr/local/include /usr/local/lib /usr/local/sbin /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Frameworks
==> /usr/bin/sudo /bin/chmod go-w /usr/local/share/zsh /usr/local/share/zsh/site-functions
==> /usr/bin/sudo /usr/sbin/chown mengchaoheng /usr/local/etc /usr/local/include /usr/local/lib /usr/local/sbin /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Frameworks
==> /usr/bin/sudo /usr/bin/chgrp admin /usr/local/etc /usr/local/include /usr/local/lib /usr/local/sbin /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Frameworks
==> /usr/bin/sudo /bin/mkdir -p /usr/local/Homebrew
==> /usr/bin/sudo /usr/sbin/chown -R mengchaoheng:admin /usr/local/Homebrew
==> /usr/bin/sudo /bin/mkdir -p /Users/mengchaoheng/Library/Caches/Homebrew
==> /usr/bin/sudo /bin/chmod g+rwx /Users/mengchaoheng/Library/Caches/Homebrew
==> /usr/bin/sudo /usr/sbin/chown -R mengchaoheng /Users/mengchaoheng/Library/Caches/Homebrew
==> Searching online for the Command Line Tools
==> /usr/bin/sudo /usr/bin/touch /tmp/.com.apple.dt.CommandLineTools.installondemand.in-progress
==> Installing Command Line Tools for Xcode-13.2
==> /usr/bin/sudo /usr/sbin/softwareupdate -i Command\ Line\ Tools\ for\ Xcode-13.2
Software Update Tool

Finding available software

Downloading Command Line Tools for Xcode
[... 2117 lines cut ...]
    Extra checks: none
  
  CPU dispatch  :
    Requested   : 'max -xop -fma4'
    Enabled     : none
    Generated   : none
  INFO: CCompilerOpt.cache_flush[817] : write cache to path -> /private/var/folders/x1/3mghbx4d4wg9ctwgtcf36xn40000gn/T/pip-install-fvcy72g3/numpy/build/temp.macosx-10.14-x86_64-3.8/ccompiler_opt_cache_ext.py
  INFO:
  ########### CLIB COMPILER OPTIMIZATION ###########
  INFO: Platform      :
    Architecture: x64
    Compiler    : clang
  
  CPU baseline  :
    Requested   : 'min'
    Enabled     : none
    Flags       : none
    Extra checks: none
  
  CPU dispatch  :
    Requested   : 'max -xop -fma4'
    Enabled     : none
    Generated   : none
  INFO: CCompilerOpt.cache_flush[817] : write cache to path -> /private/var/folders/x1/3mghbx4d4wg9ctwgtcf36xn40000gn/T/pip-install-fvcy72g3/numpy/build/temp.macosx-10.14-x86_64-3.8/ccompiler_opt_cache_clib.py
  error: Command "clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64 -Werror=implicit-function-declaration -ftrapping-math -DNPY_INTERNAL_BUILD=1 -DHAVE_NPY_CONFIG_H=1 -D_ [... cut ...] brary/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -Ibuild/src.macosx-10.14-x86_64-3.8/numpy/core/src/common -Ibuild/src.macosx-10.14-x86_64-3.8/numpy/core/src/npymath -c numpy/core/src/multiarray/dragon4.c -o build/temp.macosx-10.14-x86_64-3.8/numpy/core/src/multiarray/dragon4.o -MMD -MF build/temp.macosx-10.14-x86_64-3.8/numpy/core/src/multiarray/dragon4.o.d -msse3 -I/System/Library/Frameworks/vecLib.framework/Headers" failed with exit status 1

mengchaoheng, 2022-02-27:
----------------------------------------
  ERROR: Failed building wheel for numpy
  Building wheel for pandas (PEP 517) ... /
done
  Created wheel for pandas: filename=pandas-1.4.1-cp38-cp38-macosx_10_14_x86_64.whl size=17108068 sha256=b89f5bd88dd027f0cca7cfd34cc3797bce8c9aeb477d91d77df44ca4404d9b75
  Stored in directory: /Users/mengchaoheng/Library/Caches/pip/wheels/9c/ff/1d/841b2f0737df7a2cb83d13200a407b107998a09ba64201bcbd
  Building wheel for pyyaml (PEP 517) ... done
  Created wheel for pyyaml: filename=PyYAML-6.0-cp38-cp38-macosx_10_14_x86_64.whl size=45317 sha256=9de76e0ef863dac4c81dc5d2a123fbb23e4553917ca80f39ef3bb1f4502fdf50
  Stored in directory: /Users/mengchaoheng/Library/Caches/pip/wheels/52/84/66/50912fd7bf1639a31758e40bd4312602e104a8eca1e0da9645
  Building wheel for future (setup.py) ... done
  Created wheel for future: filename=future-0.18.2-py3-none-any.whl size=491056 sha256=4e04c3be10fcd73a3312352b715648a43c1f4f5426c17a0cabdda249bb9e4275
  Stored in directory: /Users/mengchaoheng/Library/Caches/pip/wheels/8e/70/28/3d6ccd6e315f65f245da085482a2e1c7d14b90b30f239e2cf4
  Building wheel for MarkupSafe (setup.py) ... done
  Created wheel for MarkupSafe: filename=MarkupSafe-2.1.0-cp38-cp38-macosx_10_14_x86_64.whl size=17687 sha256=4964bf3e2e3365330b31b2e15136f48e13893e8c2ba8bc96058ad1bf4e539db6
  Stored in directory: /Users/mengchaoheng/Library/Caches/pip/wheels/3c/db/10/7a50ee1e5c7555d6633e660b63ad8ab97a7bc291c74a4b2d59
  Building wheel for pyrsistent (PEP 517) ... done
  Created wheel for pyrsistent: filename=pyrsistent-0.18.1-cp38-cp38-macosx_10_14_x86_64.whl size=81429 sha256=22573b103b3576a095280885aec9c36fac2cbee4255d5d0ff45230895b1d6b9f
  Stored in directory: /Users/mengchaoheng/Library/Caches/pip/wheels/30/87/4f/6ac908945dc671cfe494bbb1b0374a5f81238e2d67936a397b
Successfully built empy pandas pyyaml future MarkupSafe pyrsistent
Failed to build numpy
ERROR: Could not build wheels for numpy which use PEP 517 and cannot be installed directly
WARNING: You are using pip version 20.2.3; however, version 22.0.3 is available.
You should consider upgrading via the '/Library/Developer/CommandLineTools/usr/bin/python3 -m pip install --upgrade pip' command.
mengchaoheng@mengchaohengdeMacBook-Pro Downloads % 
mengchaoheng@mengchaohengdeMacBook-Pro Downloads % python3 -m pip install --user pyserial empy toml numpy pandas jinja2 pyyaml pyros-genmsg packaging kconfiglib future jsonschema
Collecting pyserial
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Processing /Users/mengchaoheng/Library/Caches/pip/wheels/b0/d7/80/0e8ad4f073e05b2435c3b4c12c230cb219135bae4f59978612/empy-3.3.4-py3-none-any.whl
Collecting toml
  Using cached toml-0.10.2-py2.py3-none-any.whl (16 kB)
Collecting numpy
  Using cached numpy-1.22.2.zip (11.4 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting pandas
  Using cached pandas-1.4.1.tar.gz (4.9 MB)
  Installing build dependencies ... \^[^done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting jinja2
  Using cached Jinja2-3.0.3-py3-none-any.whl (133 kB)
Collecting pyyaml
  Using cached PyYAML-6.0.tar.gz (124 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting pyros-genmsg
  Using cached pyros_genmsg-0.5.8-py2.py3-none-any.whl (25 kB)
Collecting packaging
  Using cached packaging-21.3-py3-none-any.whl (40 kB)
Collecting kconfiglib
  Using cached kconfiglib-14.1.0-py2.py3-none-any.whl (145 kB)
Processing /Users/mengchaoheng/Library/Caches/pip/wheels/8e/70/28/3d6ccd6e315f65f245da085482a2e1c7d14b90b30f239e2cf4/future-0.18.2-py3-none-any.whl
Collecting jsonschema
  Using cached jsonschema-4.4.0-py3-none-any.whl (72 kB)
Collecting python-dateutil>=2.8.1
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting pytz>=2020.1
  Using cached pytz-2021.3-py2.py3-none-any.whl (503 kB)
Collecting MarkupSafe>=2.0
  Using cached MarkupSafe-2.1.0.tar.gz (18 kB)
Collecting pyparsing!=3.0.5,>=2.0.2
  Using cached pyparsing-3.0.7-py3-none-any.whl (98 kB)
Collecting importlib-resources>=1.4.0; python_version < "3.9"
  Using cached importlib_resources-5.4.0-py3-none-any.whl (28 kB)
Collecting attrs>=17.4.0
  Using cached attrs-21.4.0-py2.py3-none-any.whl (60 kB)
Collecting pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0
  Using cached pyrsistent-0.18.1.tar.gz (100 kB)
[... 1749 lines cut ...]
  npy__cpu_baseline_fid(const char *feature)
  ^
  numpy/core/src/common/npy_cpu_features.c.src:138:1: warning: unused function 'npy__cpu_dispatch_fid' [-Wunused-function]
  npy__cpu_dispatch_fid(const char *feature)
  ^
  1 warning generated.
  3 warnings generated.
  numpy/core/src/common/npy_cpu_features.c.src:125:1: warning: unused function 'npy__cpu_baseline_fid' [-Wunused-function]
  npy__cpu_baseline_fid(const char *feature)
  ^
  numpy/core/src/common/npy_cpu_features.c.src:138:1: warning: unused function 'npy__cpu_dispatch_fid' [-Wunused-function]
  npy__cpu_dispatch_fid(const char *feature)
  ^
  INFO: clang: numpy/core/src/common/numpyos.c
  2 warnings generated.
  INFO: clang: numpy/core/src/common/cblasfuncs.c
  warning: overriding currently unsupported use of floating point exceptions on this target [-Wunsupported-floating-point-opt]
  warning: overriding currently unsupported use of floating point exceptions on this target [-Wunsupported-floating-point-opt]
  1 warning generated.
  1 warning generated.
  INFO: clang: numpy/core/src/umath/extobj.c
  warning: overriding currently unsupported use of floating point exceptions on this target [-Wunsupported-floating-point-opt]
  INFO: clang: numpy/core/src/common/python_xerbla.c
  1 warning generated.
  warning: overriding currently unsupported u

mengchaoheng, 2022-02-27:
and then problem come to :

mengchaoheng@mengchaohengdeMacBook-Pro Downloads % brew tap AdoptOpenJDK/openjdk
Running `brew update --preinstall`...
==> Tapping adoptopenjdk/openjdk
Cloning into '/usr/local/Homebrew/Library/Taps/adoptopenjdk/homebrew-openjdk'...
remote: Enumerating objects: 1996, done.
remote: Counting objects: 100% (60/60), done.
remote: Compressing objects: 100% (22/22), done.
remote: Total 1996 (delta 44), reused 49 (delta 38), pack-reused 1936
Receiving objects: 100% (1996/1996), 372.60 KiB | 486.00 KiB/s, done.
Resolving deltas: 100% (1424/1424), done.
Tapped 47 casks (69 files, 522.2KB).
mengchaoheng@mengchaohengdeMacBook-Pro Downloads % brew cask install adoptopenjdk14
Error: `brew cask` is no longer a `brew` command. Use `brew <command> --cask` instead.
mengchaoheng@mengchaohengdeMacBook-Pro Downloads % brew install --cask adoptopenjdk14
==> Downloading https://github.com/AdoptOpenJDK/openjdk14-binaries/releases/download/jdk-14.0.2%2B12/OpenJDK14U-jdk_x64_mac_hotspot_14.0.2_1
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/233878254/76a12100-c74a-11ea-9f22-d3fa20a5
######################################################################## 100.0%
^[==> Installing Cask adoptopenjdk14
==> Running installer for adoptopenjdk14; your password may be necessary.
Package installers may write to any location; options such as `--appdir` are ignored.
Password:
Sorry, try again.
Password:
installer: Package name is AdoptOpenJDK
installer: Installing at base path /
installer: The install was successful.
package-id: net.adoptopenjdk.14.jdk
version: 14.0.2+12
volume: /
location: Library/Java/JavaVirtualMachines/adoptopenjdk-14.jdk
install-time: 1645976954
🍺  adoptopenjdk14 was successfully installed!
mengchaoheng@mengchaohengdeMacBook-Pro Downloads % brew install px4-sim-jmavsim
Running `brew update --preinstall`...
==> Downloading https://raw.githubusercontent.com/PX4/PX4-Autopilot/master/Tools/px4.py
Already downloaded: /Users/mengchaoheng/Library/Caches/Homebrew/downloads/be4aff9b473b8b0c6bae0ff08a6611885de9d56ee3a477c05453bb6bbcaab246--px4.py
==> Installing px4-sim-jmavsim from px4/px4
==> PX4 jMAVSim simulation installed
🍺  /usr/local/Cellar/px4-sim-jmavsim/1.11.0: 3 files, 7.7KB, built in 2 seconds
==> Running `brew cleanup px4-sim-jmavsim`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
mengchaoheng@mengchaohengdeMacBook-Pro Downloads % git clone https://github.com/PX4/PX4-Autopilot.git --recursive
Cloning into 'PX4-Autopilot'...
remote: Enumerating objects: 395840, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 395840 (delta 0), reused 9 (delta 0), pack-reused 395830
Receiving objects: 100% (395840/395840), 172.01 MiB | 2.01 MiB/s, done.
Resolving deltas: 100% (293497/293497), done.
Submodule 'Tools/flightgear_bridge' (https://github.com/PX4/PX4-FlightGear-Bridge.git) registered for path 'Tools/flightgear_bridge'
Submodule 'Tools/jMAVSim' (https://github.com/PX4/jMAVSim.git) registered for path 'Tools/jMAVSim'
Submodule 'Tools/jsbsim_bridge' (https://github.com/PX4/px4-jsbsim-bridge.git) registered for path 'Tools/jsbsim_bridge'
Submodule 'Tools/simulation-ignition' (https://github.com/PX4/px4-simulation-ignition.git) registered for path 'Tools/simulation-ignition'
Submodule 'Tools/sitl_gazebo' (https://github.com/PX4/PX4-SITL_gazebo.git) registered for path 'Tools/sitl_gazebo'
Submodule 'platforms/nuttx/NuttX/apps' (https://github.com/PX4/NuttX-apps.git) registered for path 'platforms/nuttx/NuttX/apps'
Submodule 'platforms/nuttx/NuttX/nuttx' (https://github.com/PX4/NuttX.git) registered for path 'platforms/nuttx/NuttX/nuttx'
Submodule 'src/drivers/gps/devices' (https://github.com/PX4/PX4-GPSDrivers.git) registered for path 'src/drivers/gps/devices'
Submodule 'src/drivers/uavcan/libuavcan' (https://github.com/dronecan/libuavcan.git) registered for path 'src/drivers/uavcan/libuavcan'
Submodule 'src/drivers/uavcan_v1/legacy_data_types' (https://github.com/PX4/public_regulated_data_types.git) registered for path 'src/drivers/uavcan_v1/legacy_data_types'
Submodule 'src/drivers/uavcan_v1/libcanard' (https://github.com/UAVCAN/libcanard.git) registered for path 'src/drivers/uavcan_v1/libcanard'
Submodule 'src/drivers/uavcan_v1/public_regulated_data_types' (https://github.com/UAVCAN/public_regulated_data_types.git) registered for path 'src/drivers/uavcan_v1/public_regulated_data_types'
Submodule 'src/lib/crypto/libtomcrypt' (https://github.com/PX4/libtomcrypt.git) registered for path 'src/lib/crypto/libtomcrypt'
Submodule 'src/lib/crypto/libtommath' (https://github.com/PX4/libtommath.git) registered for path 'src/lib/crypto/libtommath'
Submodule 'src/lib/crypto/monocypher' (https://github.com/PX4/Monocypher.git) registered for path 'src/lib/crypto/monocypher'
Submodule 'src/lib/events/libevents' (https://github.com/mavlink/libevents.git) registered for path 'src/lib/events/libevents'
Submodule 'src/modules/mavlink/mavlink' (https://github.com/mavlink/mavlink.git) registered for path 'src/modules/mavlink/mavlink'
Submodule 'src/modules/micrortps_bridge/micro-CDR' (https://github.com/PX4/Micro-CDR.git) registered for path 'src/modules/micrortps_bridge/micro-CDR'
[... 427 lines cut ...]
     |████████████████████████████████| 17.6 MB 100 kB/s            
Installing collected packages: numpy
[... cut ...]
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'menuconfig'
CMake Error at cmake/kconfig.cmake:6 (message):
  kconfiglib is not installed or not in PATH

  please install using "pip3 install kconfiglib"

Call Stack (most recent call first):
  CMakeLists.txt:174 (include)

-- Configuring incomplete, errors occurred!
Error: /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default is not a directory
make: *** [px4_sitl] Error 1

dagar, 2022-02-27:
> I have checkout to v1.12.3, as above

There were M1 compatibility changes that only went in after v1.12.3. Start with current master to get things working first.

mengchaoheng, 2022-02-27:
now in master:

mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % ./Tools/setup/macos.sh
px4-dev already installed
Installing PX4 Python3 dependencies
Collecting argcomplete
  Using cached argcomplete-2.0.0-py2.py3-none-any.whl (37 kB)
Collecting argparse>=1.2
  Using cached argparse-1.4.0-py2.py3-none-any.whl (23 kB)
Collecting cerberus
  Using cached Cerberus-1.3.4.tar.gz (63 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting coverage
  Downloading coverage-6.3.2-cp39-cp39-macosx_10_9_x86_64.whl (183 kB)
     |████████████████████████████████| 183 kB 238 kB/s            
Collecting empy>=3.3
  Using cached empy-3.3.4.tar.gz (62 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: future in /Users/mengchaoheng/Library/Python/3.9/lib/python/site-packages (from -r /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/setup/requirements.txt (line 6)) (0.18.2)
Collecting jinja2>=2.8
  Using cached Jinja2-3.0.3-py3-none-any.whl (133 kB)
Collecting jsonschema
  Using cached jsonschema-4.4.0-py3-none-any.whl (72 kB)
Collecting kconfiglib
  Using cached kconfiglib-14.1.0-py2.py3-none-any.whl (145 kB)
Collecting lxml
  Downloading lxml-4.8.0-cp39-cp39-macosx_10_15_x86_64.whl (4.6 MB)
     |████████████████████████████████| 4.6 MB 233 kB/s            
Collecting matplotlib>=3.0.*
  Downloading matplotlib-3.5.1-cp39-cp39-macosx_10_9_x86_64.whl (7.3 MB)
     |████████████████████████████████| 7.3 MB 8.2 MB/s            
Requirement already satisfied: numpy>=1.13 in /Users/mengchaoheng/Library/Python/3.9/lib/python/site-packages (from -r /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/setup/requirements.txt (line 12)) (1.22.2)
Collecting nunavut>=1.1.0
  Using cached nunavut-1.7.3-1904746659-py3-none-any.whl (311 kB)
Collecting packaging
  Using cached packaging-21.3-py3-none-any.whl (40 kB)
Collecting pandas>=0.21
  Downloading pandas-1.4.1-cp39-cp39-macosx_10_9_x86_64.whl (11.5 MB)
     |████████████████████████████████| 11.5 MB 5.3 MB/s             
Collecting pkgconfig
  Using cached pkgconfig-1.5.5-py3-none-any.whl (6.7 kB)
Collecting psutil
  Downloading psutil-5.9.0-cp39-cp39-macosx_10_9_x86_64.whl (238 kB)
     |████████████████████████████████| 238 kB 23.5 MB/s            
Collecting pygments
  Using cached Pygments-2.11.2-py3-none-any.whl (1.1 MB)
Requirement already satisfied: wheel>=0.31.1 in /usr/local/lib/python3.9/site-packages (from -r /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/setup/requirements.txt (line 19)) (0.37.1)
Collecting pymavlink
  Using cached pymavlink-2.4.26.tar.gz (5.2 MB)
  Preparing metadata (setup.py) ... done
Collecting pyros-genmsg
  Using cached pyros_genmsg-0.5.8-py2.py3-none-any.whl (25 kB)
Collecting pyserial
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting pyulog>=0.5.0
  Using cached pyulog-0.9.0-py3-none-any.whl (23 kB)
Collecting pyyaml
  Downloading PyYAML-6.0-cp39-cp39-macosx_10_9_x86_64.whl (197 kB)
     |████████████████████████████████| 197 kB 24.1 MB/s            
Collecting requests
  Using cached requests-2.27.1-py2.py3-none-any.whl (63 kB)
Requirement already satisfied: setuptools>=39.2.0 in /usr/local/lib/python3.9/site-packages (from -r /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/setup/requirements.txt (line 26)) (60.5.0)
Collecting six>=1.12.0
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting toml>=0.9
  Using cached toml-0.10.2-py2.py3-none-any.whl (16 kB)
Collecting MarkupSafe>=2.0
  Downloading MarkupSafe-2.1.0-cp39-cp39-macosx_10_9_x86_64.whl (13 kB)
[... 263 lines cut ...]
      [jar] Building jar: /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/jMAVSim/out/production/jmavsim_run.jar

copy_res:
     [copy] Copying 22 files to /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/jMAVSim/out/production

BUILD SUCCESSFUL
Total time: 5 seconds
Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 61.0), this version of the Java Runtime only recognizes class file versions up to 58.0
	at java.base/java.lang.ClassLoader.defineClass1(Native Method)
	at java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1017)
	at java.base/java.security.SecureClassLoader.defineClass(SecureClassLoader.java:151)
	at java.base/java.net.URLClassLoader.defineClass(URLClassLoader.java:514)
	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:422)
	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:416)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:691)
	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:415)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:589)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:427)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:59)
^CERROR [px4] Startup script returned with return value: 2
/Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/sitl_run.sh: line 258: kill: (43944) - No such process
ninja: build stopped: interrupted by user.
make: *** [px4_sitl_default] Interrupt: 2

mengchaoheng, 2022-02-27:
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % brew install --cask adoptopenjdk16
Running `brew update --preinstall`...
==> Homebrew is run entirely by unpaid volunteers. Please consider donating:
  https://github.com/Homebrew/brew#donations
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
Updated 4 formulae.

==> Downloading https://github.com/AdoptOpenJDK/openjdk16-binaries/releases/download/jdk-16.0.1%2B9/OpenJDK16U-jdk_x64_mac_hotspot_16.0.1_9.pkg
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/271810455/6e35ef00-a41c-11eb-935d-d66cca38eed4?X-Amz-Algorithm=AWS4-HMAC
######################################################################## 100.0%
==> Installing Cask adoptopenjdk16
==> Running installer for adoptopenjdk16; your password may be necessary.
Package installers may write to any location; options such as `--appdir` are ignored.
Password:
installer: Package name is AdoptOpenJDK
installer: Installing at base path /
installer: The install was successful.
package-id: net.adoptopenjdk.16.jdk
version: 16.0.1+9
volume: /
location: 
install-time: 1645981733
🍺  adoptopenjdk16 was successfully installed!
mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % make px4_sitl_default jmavsim     
[0/1] cd /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/tmp && ...ds/PX4-Autopilot /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /Users/mengchaoheng/Downloads/PX4-Autopilot
build_path: /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/Users/mengchaoheng/Downloads/PX4-Autopilot"/test_data
Creating symlink /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/etc -> /Users/mengchaoheng/Download

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
* CAL_MAG0_ID: curr: 0 -> new: 197388
* CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
Buildfile: /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/jMAVSim/build.xml
* SYS_AUTOCONFIG: curr: 1 -> new: 0
* IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 61.0), this version of the Java Runtime only recognizes class file versions up to 60.0
	at java.base/java.lang.ClassLoader.defineClass1(Native Method)
	at java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1010)
	at java.base/java.security.SecureClassLoader.defineClass(SecureClassLoader.java:150)
	at java.base/java.net.URLClassLoader.defineClass(URLClassLoader.java:512)
	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:420)
	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:414)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:691)
	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:413)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:586)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:519)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:466)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:59)
^[[A^[[B^[[B^[[B^CERROR [px4] Startup script returned with return value: 2
/Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/sitl_run.sh: line 258: kill: (45814) - No such process
ninja: build stopped: interrupted by user.
make: *** [px4_sitl_default] Interrupt: 2

mengchaoheng, 2022-02-27:
now It can build in "make px4_sitl_defult'', but "make px4_sitl_default jmavsim" fail. @dagar

dagar, 2022-02-27:
> now It can build in "make px4_sitl_defult'', but "make px4_sitl_default jmavsim" fail. @dagar

Sorry, I'm less sure about the core problem there (java issues).

> ```
> Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 61.0), this version of the Java Runtime only recognizes class file versions up to 60.0
> ```

Try `make distclean` and then building again. It looks like you might still have the jmavsim build from before you ran `brew install --cask adoptopenjdk16`.

FYI I'm editing your posts to put the console output in "code blocks" to make them readable. https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

mengchaoheng, 2022-02-27:
"Try make distclean and then building again. "  it doesn't work. @dagar

mengchaoheng, 2022-02-27:
I try gazebo simulation, but:

mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % brew unlink tbb                         
Error: No such keg: /usr/local/Cellar/tbb

dagar, 2022-02-27:
@ThomasDebrunner are you successfully running jmavsim or sitl_gazebo on an M1 macbook?

mengchaoheng, 2022-02-27:
now it work！ As a summary, for a new m1 mac, brew install px4-sim-jmavsim doesn't work, and the video  in 

https://docs.px4.io/master/en/dev_setup/dev_env_mac.html

is out of date. The following directives are important:
1 brew install --cask temurin.
2./Tools/setup/macos.sh.

mengchaoheng, 2022-02-27:
make px4_sitl jmavsim work now, but make px4_fmu-v5 fail, I don't know why:

mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % make px4_fmu-v5                                                                                                                                                                                                            
-- PX4 version: v1.13.0-alpha1-4371-gc2c455be0d
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.10", minimum required is "3") 
-- PX4 config file: /Users/mengchaoheng/Downloads/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- IO px4_io-v2_default
-- SERIAL_GPS1 /dev/ttyS0
-- SERIAL_TEL1 /dev/ttyS1
-- SERIAL_TEL2 /dev/ttyS2
-- SERIAL_TEL4 /dev/ttyS3
-- UAVCAN_INTERFACES 2
-- UAVCAN_TIMER_OVERRIDE 6
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 9.3.1
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Enabling double FP precision hardware instructions
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- drivers/px4io: ROMFS including px4_io-v2_default
CMake Deprecation Warning at src/drivers/uavcan/libuavcan/CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/drivers/uavcan/libuavcan/libuavcan/CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PythonInterp: /usr/local/bin/python3 (found version "3.9.10") 
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding platforms/nuttx/init/stm32f7/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_fmu-v5_bootloader.bin -> /etc/extras/px4_fmu-v5_bootloader.bin
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_io-v2_default.bin -> /etc/extras/px4_io-v2_default.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_fmu-v5_default
[0/1124] git submodule src/drivers/gps/devices
[9/1124] git submodule src/modules/mavlink/mavlink
[13/1124] git submodule src/drivers/uavcan/libuavcan
[15/1124] Generating Mavlink development: src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Parsing /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
[... 23 lines cut ...]
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find src/lib/drivers/device/libdrivers__device.a: Too many open files
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find platforms/nuttx/src/px4/stm/stm32f7/spi/libarch_spi.a: Too many open files
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find src/lib/drivers/led/libdrivers__led.a: Too many open files
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find platforms/nuttx/src/px4/stm/stm32f7/hrt/libarch_hrt.a: Too many open files
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find platforms/nuttx/src/px4/stm/stm32f7/board_hw_info/libarch_board_hw_info.a: Too many open files
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find platforms/nuttx/src/px4/stm/stm32f7/adc/libarch_adc.a: Too many open files
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find NuttX/nuttx/mm/libmm.a: Too many open files
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find platforms/nuttx/src/px4/stm/stm32f7/board_critmon/libarch_board_critmon.a: Too many open files
[... cut ...]
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find NuttX/nuttx/sched/libsched.a: Too many open files
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lm
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lgcc
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make: *** [px4_fmu-v5] Error 1

dagar, 2022-02-27:
"Too many open files" is a Mac specific problem. Try setting a larger value like `ulimit -n 10240`. https://discussions.apple.com/thread/251000125

mengchaoheng, 2022-02-27:
> https://discussions.apple.com/thread/251000125

I don't really understand what it mean, can you tell me how to operator?

mengchaoheng, 2022-02-27:
it work now, by :

brew install --cask temurin

> > https://discussions.apple.com/thread/251000125
> 
> I don't really understand what it mean, can you tell me how to operator?

get it, the step is :
1. in terminal "vi ~/.zshenv"  Create the file and add "ulimit -S -n 2048"
2.in terminal ". ~/.zshenv" to use it, then 
3. in terminal " make px4_fmu-v5"

ThomasDebrunner, 2022-03-01:
> @ThomasDebrunner are you successfully running jmavsim or sitl_gazebo on an M1 macbook?

With x86 emulation (rosetta terminal) everything works fine (emulated). Natively, PX4 with jmavsim works when using this branch for jmavsim: https://github.com/PX4/jMAVSim/pull/132
Not sure about gazebo

mengchaoheng, 2022-03-01:
> 

On m1 macbook, make px4_sitl gazebo fail. jmavsim and fmu-vx work good

mengchaoheng, 2022-03-01:
> > @ThomasDebrunner are you successfully running jmavsim or sitl_gazebo on an M1 macbook?
> 
> With x86 emulation (rosetta terminal) everything works fine (emulated). Natively, PX4 with jmavsim works when using this branch for jmavsim: [PX4/jMAVSim#132](https://github.com/PX4/jMAVSim/pull/132) Not sure about gazebo

PX4 with jmavsim work good but the pages {https://docs.px4.io/master/en/dev_setup/dev_env_mac.html} is out of the date

mengchaoheng, 2022-03-01:
I clean the page, maybe can help a new  m1 Developer. @dagar

mengchaoheng, 2022-03-01:
Issues that remain unresolved on the m1 MacBook are:
1. sitl with gazebo
2. in arm-based ubuntu VM on m1, the Ubuntu Development Environment didn't work.
@ThomasDebrunner @dagar

ThomasDebrunner, 2022-03-01:
I never tried the dev environment on arm ubuntu. I'd expect that at least the compilers for the embedded targets won't work, as these binaries are x86 only. On mac, they (automatically) run on rosetta emulation

mengchaoheng, 2022-03-01:
> I never tried the dev environment on arm ubuntu. I'd expect that at least the compilers for the embedded targets won't work, as these binaries are x86 only. On mac, they (automatically) run on rosetta emulation

Any idea for gazebo sitl on m1 mac?

mengchaoheng, 2022-03-03:
I fix it, by reinstall ffmpeg
<details><summary>CLICK ME</summary>
<p>

```Console
mengchaoheng@MCH-MBP ~ % brew list   
==> Formulae
ampl-mp				libnice
ant				libogg
aom				libpng
assimp				libpsl
astyle				libpthread-stubs
atk				libreplaygain
bash-completion			librist
bdw-gc				librsvg
boost				libsamplerate
brotli				libshout
bullet				libsndfile
ca-certificates			libsodium
cairo				libsoup@2
ccache				libsoxr
ceres-solver			libtar
cjson				libtasn1
cmake				libtiff
cmake@3.21.4			libtool
cmocka				libunistring
console_bridge			libusrsctp
cppzmq				libvidstab
dartsim				libvmaf
dav1d				libvorbis
double-conversion		libvpx
doxygen				libx11
eigen				libxau
exiftool			libxcb
faac				libxdmcp
faad2				libxext
fastcdr				libxrender
fastrtps			libyaml
fcl				libzip
ffmpeg@4			libzzip
flac				little-cms2
flann				lz4
flock				lzo
fontconfig			m4
freeimage			mbedtls
freetype			metis
frei0r				mpdecimal
fribidi				mpfr
gazebo11			musepack
gcc				netcdf
gcc-arm-none-eabi		netpbm
gd				nettle
gdbm				ninja
gdk-pixbuf			nlopt
genromfs			numpy
gettext				octomap
gflags				ode
giflib				ogre1.9
git				open-scene-graph
gl2ps				openblas
glew				opencore-amr
glib				opencv
glib-networking			openexr
glog				openjdk
gmp				openjpeg
gnutls				openssl@1.1
gobject-introspection		opus
graphene			orc
graphite2			ossp-uuid
graphviz			p11-kit
[... 285 lines cut ...]
Error: No available formula with the name "ffmepg". Did you mean ffmpeg?
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
mengchaoheng@MCH-MBP PX4-Autopilot % brew info ffmpeg
ffmpeg: stable 5.0 (bottled), HEAD
Play, record, convert, and stream audio and video
https://ffmpeg.org/
/usr/local/Cellar/ffmpeg/5.0 (272 files, 51.6MB) *
  Poured from bottle on 2022-03-03 at 13:09:30
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/ffmpeg.rb
License: GPL-2.0-or-later
==> Dependencies
Build: nasm ✘, pkg-config ✔
Required: aom ✔, dav1d ✔, fontconfig ✔, freetype ✔, frei0r ✔, gnutls ✔, lame ✔, libass ✔, libbluray ✔, librist ✔, libsoxr ✔, libvidstab ✔, libvmaf ✔, libvorbis ✔, libvpx ✔, opencore-amr ✔, openjpeg ✔, opus ✔, rav1e ✔, rubberband ✔, sdl2 ✔, snappy ✔, speex ✔, srt ✔, tesseract ✔, theora ✔, webp ✔, x264 ✔, x265 ✔, xvid ✔, xz ✔, zeromq ✔, zimg ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 155,418 (30 days), 419,806 (90 days), 1,603,575 (365 days)
install-on-request: 142,699 (30 days), 364,264 (90 days), 1,349,304 (365 days)
build-error: 83 (30 days)
```

</p>
</details>

mengchaoheng, 2022-03-03:
and try again :

<details><summary>CLICK ME</summary>
<p>

```Console

mengchaoheng@MCH-MBP PX4-Autopilot % make distclean
已清除目录 'Tools/flightgear_bridge'
子模组 'Tools/flightgear_bridge'（https://github.com/PX4/PX4-FlightGear-Bridge.git）未对路径 'Tools/flightgear_bridge' 注册
已清除目录 'Tools/jMAVSim'
子模组 'Tools/jMAVSim'（https://github.com/PX4/jMAVSim.git）未对路径 'Tools/jMAVSim' 注册
已清除目录 'Tools/jsbsim_bridge'
子模组 'Tools/jsbsim_bridge'（https://github.com/PX4/px4-jsbsim-bridge.git）未对路径 'Tools/jsbsim_bridge' 注册
已清除目录 'Tools/simulation-ignition'
子模组 'Tools/simulation-ignition'（https://github.com/PX4/px4-simulation-ignition.git）未对路径 'Tools/simulation-ignition' 注册
已清除目录 'Tools/sitl_gazebo'
子模组 'Tools/sitl_gazebo'（https://github.com/PX4/PX4-SITL_gazebo.git）未对路径 'Tools/sitl_gazebo' 注册
已清除目录 'platforms/nuttx/NuttX/apps'
子模组 'platforms/nuttx/NuttX/apps'（https://github.com/PX4/NuttX-apps.git）未对路径 'platforms/nuttx/NuttX/apps' 注册
已清除目录 'platforms/nuttx/NuttX/nuttx'
子模组 'platforms/nuttx/NuttX/nuttx'（https://github.com/PX4/NuttX.git）未对路径 'platforms/nuttx/NuttX/nuttx' 注册
已清除目录 'src/drivers/gps/devices'
子模组 'src/drivers/gps/devices'（https://github.com/PX4/PX4-GPSDrivers.git）未对路径 'src/drivers/gps/devices' 注册
已清除目录 'src/drivers/uavcan/libuavcan'
子模组 'src/drivers/uavcan/libuavcan'（https://github.com/dronecan/libuavcan.git）未对路径 'src/drivers/uavcan/libuavcan' 注册
已清除目录 'src/drivers/uavcan_v1/legacy_data_types'
子模组 'src/drivers/uavcan_v1/legacy_data_types'（https://github.com/PX4/public_regulated_data_types.git）未对路径 'src/drivers/uavcan_v1/legacy_data_types' 注册
已清除目录 'src/drivers/uavcan_v1/libcanard'
子模组 'src/drivers/uavcan_v1/libcanard'（https://github.com/UAVCAN/libcanard.git）未对路径 'src/drivers/uavcan_v1/libcanard' 注册
已清除目录 'src/drivers/uavcan_v1/public_regulated_data_types'
子模组 'src/drivers/uavcan_v1/public_regulated_data_types'（https://github.com/UAVCAN/public_regulated_data_types.git）未对路径 'src/drivers/uavcan_v1/public_regulated_data_types' 注册
已清除目录 'src/lib/crypto/libtomcrypt'
子模组 'src/lib/crypto/libtomcrypt'（https://github.com/PX4/libtomcrypt.git）未对路径 'src/lib/crypto/libtomcrypt' 注册
已清除目录 'src/lib/crypto/libtommath'
子模组 'src/lib/crypto/libtommath'（https://github.com/PX4/libtommath.git）未对路径 'src/lib/crypto/libtommath' 注册
已清除目录 'src/lib/crypto/monocypher'
子模组 'src/lib/crypto/monocypher'（https://github.com/PX4/Monocypher.git）未对路径 'src/lib/crypto/monocypher' 注册
已清除目录 'src/lib/events/libevents'
子模组 'src/lib/events/libevents'（https://github.com/mavlink/libevents.git）未对路径 'src/lib/events/libevents' 注册
已清除目录 'src/modules/mavlink/mavlink'
子模组 'src/modules/mavlink/mavlink'（https://github.com/mavlink/mavlink.git）未对路径 'src/modules/mavlink/mavlink' 注册
已清除目录 'src/modules/micrortps_bridge/micro-CDR'
子模组 'src/modules/micrortps_bridge/micro-CDR'（https://github.com/PX4/Micro-CDR.git）未对路径 'src/modules/micrortps_bridge/micro-CDR' 注册
正删除 Tools/.DS_Store
正删除 Tools/module_config/__pycache__/
正删除 Tools/px4airframes/__pycache__/
正删除 Tools/px4events/__pycache__/
正删除 msg/tools/__pycache__/
正删除 platforms/.DS_Store
正删除 platforms/nuttx/.DS_Store
正删除 platforms/nuttx/NuttX/.DS_Store
正删除 platforms/nuttx/NuttX/tools/__pycache__/
正删除 platforms/nuttx/src/.DS_Store
正删除 platforms/nuttx/src/canbootloader/.DS_Store
正删除 platforms/nuttx/src/canbootloader/arch/.DS_Store
正删除 platforms/nuttx/src/canbootloader/arch/stm/.DS_Store
正删除 platforms/nuttx/src/px4/.DS_Store
正删除 platforms/nuttx/src/px4/nxp/.DS_Store
正删除 src/.DS_Store
正删除 src/drivers/.DS_Store
正删除 src/drivers/imu/.DS_Store
正删除 src/drivers/imu/analog_devices/.DS_Store
正删除 src/drivers/uavcan/.DS_Store
正删除 src/drivers/uavcan_v1/.DS_Store
正删除 src/lib/.DS_Store
正删除 src/lib/crypto/.DS_Store
正删除 src/lib/events/.DS_Store
正删除 src/lib/parameters/px4params/__pycache__/
正删除 src/modules/.DS_Store
[... 421 lines cut ...]
INFO  [navigator] Mission finished, loitering	
INFO  [navigator] RTL HOME activated	
INFO  [navigator] RTL: landing at home position.	
INFO  [navigator] RTL: climb to 499 m (10 m above destination)	
INFO  [navigator] RTL: return at 499 m (10 m above destination)	
INFO  [navigator] RTL: land at destination	
INFO  [commander] Landing detected	
INFO  [commander] Disarmed by landing	
INFO  [vehicle_imu] ACC 2 (1311004) offset committed: [0.000 0.000 0.000]->[-0.064 0.028 -0.042])
INFO  [vehicle_imu] ACC 0 (1310988) offset committed: [0.000 0.000 0.000]->[-0.052 0.034 -0.040])
INFO  [vehicle_imu] ACC 1 (1310996) offset committed: [0.000 0.000 0.000]->[-0.064 0.028 -0.042])
INFO  [tone_alarm] notify positive
INFO  [logger] closed logfile, bytes written: 19556621

PX4 Exiting...
pxh> Exiting NOW.
ninja: build stopped: interrupted by user.
make: *** [px4_sitl] Interrupt: 2

mengchaoheng@MCH-MBP PX4-Autopilot % cd .. 
mengchaoheng@MCH-MBP Downloads % cd ..

````
</p>
</details>

mengchaoheng, 2022-03-03:
I don't know that's this:
```
[Err] [Scene.cc:227] Service call[/shadow_caster_material_name] timed out
[Err] [Scene.cc:249] Service call[/shadow_caster_render_back_faces] timed out

```

but anyway, I unsintall gazebo11 and then install it:

<details><summary>CLICK ME</summary>
<p>

```Console

mengchaoheng@MCH-MBP PX4-Autopilot % brew uninstall gazebo11
Error: Refusing to uninstall /usr/local/Cellar/gazebo11/11.10.1_4
because it is required by px4-sim-gazebo, which is currently installed.
You can override this and force removal with:
  brew uninstall --ignore-dependencies gazebo11
mengchaoheng@MCH-MBP PX4-Autopilot % brew uninstall --ignore-dependencies gazebo11
Uninstalling /usr/local/Cellar/gazebo11/11.10.1_4... (1,294 files, 130.2MB)
mengchaoheng@MCH-MBP PX4-Autopilot % brew info  gazebo11                 
osrf/simulation/gazebo11: stable 11.10.1 (bottled), HEAD
Gazebo robot simulator
https://gazebosim.org
Conflicts with:
  gazebo7 (because differing version of the same formula)
  gazebo9 (because differing version of the same formula)
Not installed
From: https://github.com/osrf/homebrew-simulation/blob/HEAD/Formula/gazebo11.rb
License: Apache-2.0
==> Dependencies
Build: cmake ✔, pkg-config ✔
Required: boost ✔, bullet ✔, cmake@3.21.4 ✔, dartsim ✔, doxygen ✔, ffmpeg@4 ✔, freeimage ✔, graphviz ✔, gts ✔, ignition-common3 ✔, ignition-fuel-tools4 ✔, ignition-math6 ✔, ignition-msgs5 ✔, ignition-transport8 ✔, libtar ✔, ogre1.9 ✔, ossp-uuid ✔, protobuf ✔, protobuf-c ✔, qt@5 ✔, qwt-qt5 ✔, sdformat9 ✔, simbody ✔, tbb@2020_u3 ✔, tinyxml ✔, tinyxml2 ✔, zeromq ✔
==> Options
--HEAD
	Install HEAD version
mengchaoheng@MCH-MBP PX4-Autopilot % brew install gazebo11                      
Running `brew update --preinstall`...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/cask).
==> Updated Casks
Updated 1 cask.

==> Downloading https://osrf-distributions.s3.amazonaws.com/bottles-simulation/gazebo11-11.10.1_4.big_sur.bottl
Already downloaded: /Users/mengchaoheng/Library/Caches/Homebrew/downloads/16afe4a223d5c39119d13f33c93b4980b66772b456a05e96afe7a08d81c65de5--gazebo11-11.10.1_4.big_sur.bottle.tar.gz
==> Installing gazebo11 from osrf/simulation
==> Pouring gazebo11-11.10.1_4.big_sur.bottle.tar.gz
🍺  /usr/local/Cellar/gazebo11/11.10.1_4: 1,294 files, 130.1MB
==> Running `brew cleanup gazebo11`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).

```

mengchaoheng, 2022-03-03:
and anything look good:
<details><summary>CLICK ME</summary>
<p>
```Console

mengchaoheng@MCH-MBP PX4-Autopilot % PX4_NO_FOLLOW_MODE=1 make px4_sitl gazebo
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default...utopilot /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /Users/mengchaoheng/Downloads/PX4-Autopilot
build_path: /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
SITL COMMAND: "/Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/Users/mengchaoheng/Downloads/PX4-Autopilot"/test_data
Creating symlink /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/etc -> /Users/mengchaoheng/Download

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
INFO  [parameters] BSON document size 945 bytes, decoded 945 bytes (INT32:20, FLOAT:24)
[param] Loaded: eeprom/parameters_10016
Gazebo multi-robot simulator, version 11.10.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 127.0.0.1
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [navigator] Mission #3 loaded, 5 WPs
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [init] setting PWM_AUX_OUT none
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
[Wrn] [GuiIface.cc:120] Populating font family aliases took 185 ms. Replace uses of missing font family "Sans" with one that exists to avoid this cost. 
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
[... 31 lines cut ...]
INFO  [navigator] RTL: return at 499 m (10 m above destination)	
INFO  [navigator] RTL: land at destination	
INFO  [commander] Landing detected	
[... cut ...]
INFO  [vehicle_imu] GYRO 1 (1310996) offset committed: [-0.011 -0.014 -0.012]->[-0.004 -0.005 -0.004])
INFO  [tone_alarm] notify positive
INFO  [logger] closed logfile, bytes written: 192607596
INFO  [tone_alarm] notify negative
INFO  [commander] Armed by external command	
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-03-03/07_39_56.ulg	
INFO  [logger] Opened full log file: ./log/2022-03-03/07_39_56.ulg
INFO  [commander] Takeoff detected	
INFO  [tone_alarm] notify positive
INFO  [navigator] Mission finished, loitering

```

</p>
</details>

mengchaoheng, 2022-03-03:
But sometimes the error reappears, but it doesn't seem to affect the normal operation of the simulation. This error is also found on gazebo's issue

https://github.com/osrf/gazebo/issues/3173

ryanjAA, 2022-03-21:
Here is a PR for this. It fixed it for me so I added it to update the Mac build instructions.

aliadeeba98, 2025-06-19:
> Try with `/usr/local/bin/python3 -m pip install kconfiglib`?
> 
> It looks like there's confusion between a system default 3.8 and python 3.7.5 found by cmake.

Thanks a lot!, it solves the same issue for me too.

closed 2025-06-19 by dakejahl