PX4/PX4-Autopilot #18416: Catapult lauch: Motor delay not working

jstrebel, 2021-10-12:
Hi I need to open this ticket again. Initial ticket was: 18354 (it was closed. I can^t change it back via the Web GUI)

PX4 (SW 1.12.3 / HW holibro pixhawk 4) The Motor starts immedialtely after launch is detected. This causes damage to the propeller, because it hits the catapult. Actual behaviour: If Accelerometer threshold > LAUN_CAT_A then Motor starts immediately Expected behaviour: If Accelerometer threshold > LAUN_CAT_A then Motorstart is delayed by value of LAUN_CATMDEL waits then motors starts.
Parameters: LAUN_CAT_MDEL and LAUN_CAT_T have no effect. Motor startup is not delayed as expected.

The log is made with Master
https://review.px4.io/plot_app?log=06b779a9-95b5-4413-a82f-d4f5734ed81d
 This was working in 1.9.1

kevinjhur, 2022-04-05:
Hi guys,
Is it still remaing issue? Thank you.

ryanjAA, 2022-04-29:
Also interested in if this is working.

jstrebel, 2022-05-30:
Hello, I just tried 1.13.beta1. Bug is still present. @dagar . Thank you for takng this up.

Jaeyoung-Lim, 2022-06-15:
@dagar Somehow this seems to be fixed in SITL now

sfuhrer, 2022-10-24:
I also can't reproduce, it looks correct in SITL with current main.
https://user-images.githubusercontent.com/26798987/197636340-16941240-6d50-46c8-80e3-85c99e2d6e76.mp4

jstrebel, 2022-10-27:
Hi here is the promised log. Motor starts immediately powering up when I switch to Mission mode. 
SW vesion main dated 26Oct

https://review.px4.io/plot_app?log=1e9bf8c8-de84-4d00-83b2-1e4cb160e8a2

Description: Launch detection test
Feedback: Launch detect not working
Vehicle type: Fixed Wing
Airframe: Generic Flying Wing, Flying Wing
Hardware: PX4_FMU_V5

sfuhrer, 2022-10-28:
Thanks for the test @jstrebel ! In your log I see that the max acceleration in x measured by the system is only 4, while your set threshold is 8. And are you sure the motors powering up immediately on switching to Mission mode? Because I don't see that in the log, the thrust in x stays 0 the whole time, and also the info messages don't say anything about that the launch is detected and the motor is powering up.

[image]

I actually went trying it out myself yesterday, and for me it worked. With current main on a Pixhawk4 based electroglider. My colleague shook the plane, the motors spooled up 0.5s later, he threw it and it did the climbout and mission. 
My settings are: 
- LAUN_CAT_A 20
- LAUN_CAT_MDEL 0.5
- LAUN_CAT_T 0.01
The motor spins up 0.5s after the acceleration threshold of 20 is breached, see plots 
[image]

https://review.px4.io/plot_app?log=eed31b3c-56f2-4182-af02-167c82d74efd
[image]

https://user-images.githubusercontent.com/26798987/198523647-aefb5af5-f890-4cc3-8769-ac6739bb3ee0.mp4