PX4/PX4-Autopilot #18478: Cannot zoom in the camera in typhoon480

tawfiq1200, 2021-10-20:
I am trying to zoom in and zoom out with the  simulated camera in typhoon480, I have used this:
```

from pymavlink import mavutil

connection = mavutil.mavlink_connection('udpin:0.0.0.0:14550')
connection.wait_heartbeat()
connection.mav.command_long_send(connection.target_system, connection.target_component, mavutil.mavlink.MAV_CMD_DO_DIGICAM_CONTROL, 0, 1, 3, 3, 0, 1, 0, 0)
```

It did not zoom but also did not throw any error, then I tried this in place of `MAV_CMD_DO_DIGICAM_CONTROL,
`
`MAV_CMD_SET_CAMERA_ZOOM`

It threw this error:

`AttributeError: module 'pymavlink.dialects.v20.ardupilotmega' has no attribute 'MAV_CMD_SET_CAMERA_ZOOM'`

Jaeyoung-Lim, 2021-10-20:
@tawfiq1200 Why are you using the ardupilot dialect? PX4 is not compatible with the ardupilot dialect

tawfiq1200, 2021-10-20:
@Jaeyoung-Lim Thanks for letting me know, I used code from [here](https://www.ardusub.com/developers/pymavlink.html) before and it worked so I thought it will be compatible.

If I could get an example of python implementation for camera zooming with PX4, that would be great.

Jaeyoung-Lim, 2022-02-20:
@tawfiq1200 Unfortunately I do not have an python implementation that I can share.

I am closing this issue since it is not a software issue on the firmware side.

Please feel free to reopen if you think this is relevant.

closed 2022-02-20 by Jaeyoung-Lim