PX4/PX4-Autopilot #18556: Dynamic Control Allocation/Mixer

karishmapatnaik, 2021-11-02:
Hi @jlecoeur @sfuhrer 
I have a dynamic morphology and I have been using pixhawk consistently to code our custom controller to generate thrust and torques. However, with this dynamic airframe that can change anytime and is not fixed, I need to now work on the control allocation. At any given instant I can generate the allocation matrix, but the problem is I don't know how to change the mixer dynamically. Can you please point me in the right direction where to add the code for a dynamic mixer or if it is possible to send direct motor commands by disabling the mixer (I cannot seem to figure out how to disable the mixer) ? Thank you!

jlecoeur, 2021-11-02:
Hi @karishmapatnaik 
You need to customize the control_allocator module. This module has three main components:
- ActuatorEffectiveness: describes actuator model (placement and effect on the UAV)
- ControlAllocation: contains the allocation algorithm 
- ControlAllocator: basically glue code

In your case you want to customize the actuatorEffectiveness. Assuming your platform is a multi rotor with dynamic shape, you can implement a child class of ActuatorEffectivenessMultirotor (https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/control_allocator/ActuatorEffectiveness/ActuatorEffectivenessMultirotor.cpp) with your custom dynamic geometry, then instantiate this class in the control_allocator module, here https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/control_allocator/ControlAllocator.cpp#L192

karishmapatnaik, 2021-11-02:
This is perfect! Thank you, I will get started on this

Jawad-RoboLearn, 2022-01-07:
Dear @jlecoeur I am very much interested in this, and I need your help. 

1. What I am doing is to implement the high level control on ROS and using MAVROS topic /mavros/actuator control to send the commands to PX4.
2. What currently I am asking you, how to use your mixing strategy (Actuator Effectiveness and Control Allocator) with ROS. Is it applicable?
3. Also, there is script named rcS or rc that is the startup script. How do we use your strategy with PX4 startup scripts

jlecoeur, 2022-01-07:
1. I believe usually people leave the low level controllers (rate, or rate + attitude) on PX4 for latency reason. Depending on your platform, it may be fine though.
2. Yes.
3. The configuration of control allocation can be done much more easily since https://github.com/PX4/PX4-Autopilot/pull/18776

sfuhrer, 2022-01-07:
I think with https://github.com/PX4/PX4-Autopilot/pull/18776 being merged it's okay to close this issue, re-open if you think otherwise.

closed 2022-01-07 by sfuhrer