PX4/PX4-Autopilot #18562: RTF problem in multi r1_rover vehicle launch

yfaws, 2021-11-03:
Hi! I am using px4-autopilot master branch with commit 8f6fd5f37b5f3f1bfb9a04728be09ee05cddf230 on ubuntu18.04, ros-melodic, gazebo-9 setup.
 
When I try multi-vehicle launch with more than one aion robotics r1_rover, RTF(real time factor) in gazebo goes down into 0.1 even less than. 
But suprisingly, when I launch only one r1_rover, RTF in Gazebo is 1.0. I could not understand actual problem causing that.
Is there any solution for this problem ?

Nanke2012, 2021-11-11:
I also encountered this problem, which has not been solved. There is no such problem in multiple rotorcraft simulations

Nanke2012, 2021-11-11:
If three iris simulators are used, the IMU data frequency is 3 * 250Hz and the RTF is about 1. If three plane or Rover simulators are used, the IMU data frequency is only 250Hz and the RTF is about 0.25. I don't know whether this is the reason

Jaeyoung-Lim, 2022-01-24:
@Nanke2012 @yfaws This was fixed in https://github.com/PX4/PX4-SITL_gazebo/pull/796

closed 2022-01-24 by Jaeyoung-Lim