PX4/PX4-Autopilot #18567: DSHOT support on VTOL tailsitter configuration

galdalali, 2021-11-03:
**Describe the bug**
Using the cube orange configured as a VTOL tailsitter with DSHOT300 configuration and the ESC's are connected via AUX 1-4 output (SYS_USE_IO=0) the ESC'S are not getting signal at all (neither from the main outputs).

When working with PWM/ONESHOT configuration using the AUX 1-4 output (SYS_USE_IO=0) - behavior is OK.

Configure the frame as multi-copter with DSHOT300 configuration and the ESC's are connected via AUX 1-4 output (SYS_USE_IO=0) - behavior is OK.

**Expected behavior**
Working VTOL with DSHOT configuration on the cube orange AUX outputs

dagar, 2021-11-10:
With the new output configuration system (and control allocation) this should all be possible (depending on the physical timers of course). https://github.com/mavlink/qgroundcontrol/pull/9952

taileron, 2021-11-10:
looks very exciting

galdalali, 2021-11-13:
Great! What's holding this PR from merging to master?

Ecstasy-EC, 2021-12-04:
Any updates about this PR? Just came across the same problem. Motors on my tail-sitter did not receive any signals. Many thanks! @dagar

dagar, 2021-12-04:
> Any updates about this PR? Just came across the same problem. Motors on my tail-sitter did not receive any signals. Many thanks! @dagar

Yes the big output configuration/mixing overhaul is coming together for VTOL. https://github.com/PX4/PX4-Autopilot/pull/18776

If you absolutely need dshot for vtol right now we could probably hack something together quickly, but otherwise it would be great to focus on getting the right solution finished.

galdalali, 2021-12-04:
BTW, it appears there is no AUX mixer configured for VTOL tail-sitter configuration, once I configured a new frame and aux mixer by myself everything works OK.

bkueng, 2022-10-25:
Control allocation is now the default (https://github.com/PX4/PX4-Autopilot/pull/19990)

closed 2022-10-25 by bkueng